WebPlatformForEmbedded/meta-wpe #130: Build tries to clone proprietary module

petrosagg, 2017-03-13:
After following the instructions from the README, the yocto build failed because it was unable to clone the `cppsdk` repo which appears to be closed source.

wouterlucas, 2017-03-13:
You can remove these two lines in the package group and it should go through:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-core/packagegroups/packagegroup-rpi-ml-wpe.bb#L15-L16

Sorry about that, cppsdk and webbridge will be removed from meta-wpe in the near future.

petrosagg, 2017-03-13:
Thanks @wouterlucas. That's what I ended up doing as well

closed 2017-03-13 by wouterlucas

HE201250, 2017-04-26:
@wouterlucas Just a question, are you sure about these 2 lines? I'm pretty sure its ccpsdk and webbridge, not ccpdsk and wpelauncher.

wouterlucas, 2017-04-27:
@HE201250 that used to be cppsdk and webbridge indeed, but the file got edited/changed afterwards and now that line reference is a bit messed up. This is how it used to be:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/1df2af5a8cc31553b78bc7868ebf1854ef34fd21/recipes-core/packagegroups/packagegroup-rpi-ml-wpe.bb#L15-L16