WebPlatformForEmbedded/meta-wpe #131: Running WPELaunch segfaults

petrosagg, 2017-03-13:
After creating an SD image using the instructions provided in the README and after making the modifications suggested in #1, the Pi is unable to run WPE.

With the clean image on a Raspberrypi 2 running `WPELauncher` fails with `wpa: could not load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory`.

I tried creating a symlink of `/usr/lib/libWPEBackend-default.so -> libWPEBackend-rdk.so` but that didn't help. What is the correct implementation for the backend?

wouterlucas, 2017-03-13:
Please try with latest bump of wpewebkit.

petrosagg, 2017-03-13:
Hm, I just realised I hadn't done this step from the README:
```
PACKAGECONFIG_append_pn-qemu-native = ""
PACKAGECONFIG_append_pn-nativesdk-qemu = ""
ASSUME_PROVIDED += ""
```
I'm not sure if it's significant. I'll re-run the build.

petrosagg, 2017-03-13:
Rebuilding with those changes in my `local.conf` didn't help. I didn't try building with the new wpe to make sure I change one thing at a time.

But is it correct that I symlink the backend lib? Will the new wpe version you added pick up the correct backend automatically?

wouterlucas, 2017-03-13:
Correct, latest WPEWebKit bump should correct that. There was a typo in the linking part, you cant work around that with ln's. If you pull in the latest and rebuild it should be fine.

petrosagg, 2017-03-14:
I rebuilt the image using the latest commit of this repo with the westeros image and it still segfaults. Here are the precise steps I do:

1. Run `bitbake westeros-wpe-image`. meta-wpe is at commit 84386ccfd2fc13dbd569b69a41afbfae7bf994dc
2. Connect Pi to keyboard, Ethernet, HDMI screen
3. Boot the Pi
4. Login as `root` and run `ln -s /usr/lib/libWPEBackend-rdk.so /usr/lib/libWPEBackend-default.so`
5. Run `WPELauncher`

I can see the yellow mouse cursor of the westeros which works fine. But wpe doesn't seem to start

petrosagg, 2017-03-14:
I'm running all of these on a RaspberryPi 3

wouterlucas, 2017-03-14:
Make sure you tell WPE and Westeros to share the same displayid through exports. The whole launching of westors together with WPE is something we're working on to make it a little easier. 

E.g. by setting the correct exports before launching westeros, if not set al ready:
export XDG_RUNTIME_DIR=/tmp 
export WAYLAND_DISPLAY=wayland-0 
export LD_PRELOAD=/usr/lib/libwayland-egl.so.0.0.0 
export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0 

And same before WPE Launcher:
export XDG_RUNTIME_DIR=/tmp 
export WAYLAND_DISPLAY=wayland-0

kraj, 2017-03-14:
secondly make sure that you use Thumb2 to compile WPE

petrosagg, 2017-03-14:
Hm, my build still segfaults.

The `westeros-wpe-image` runs westeros automatically on startup using a sysv init script that starts westeros by running `westeros-init`. Then I run `WPELaunch` with `XDG_RUNTIME_DIR=/run` and `WAYLAND_DISPLAY=wayland-0`, which are the same as the auto-started westeros instance. 

@kraj Thumb2 is automatically set by the recipe in this repo. I verified that this is the case in my build, wpewebkit is under `work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit`.

I'm currently rebuilding with `GCC_VERSION_forcevariable = "5.4%"` to see if it makes a difference.

Do you guys reproduce this issue on your machines or is it only on my side? I'm trying to follow the README as closely as possible with the only difference being that I remove `cppsdk` and `webbridge` from the packagegroup.

Here are all the commits of the layers I'm using. If you have a known working set of commits I could try with those:

```
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2+snapshot-20170314"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "master:b3d13e058a86cafc33aeab11eddfde77b1fd884f"
meta-raspberrypi  = "master:c768a3d1aed8d16d08afe12fddb359914e0a203a"
meta-metrological = "master:84386ccfd2fc13dbd569b69a41afbfae7bf994dc"
meta-oe           
meta-multimedia   = "master:1dff2351aa6cdafa5a501e8956cb853ab17ed9ae"
```

kraj, 2017-03-14:
I haven't built for sometime now. Let me try it out here as well.

I also have the following in my local.conf

WPE_BACKEND_rpi = "westeros"
ENABLE_UART = "1"
# Enable Turbo mode on Rpi2
ARM_FREQ = "1000"
CORE_FREQ = "500"
SDRAM_FREQ = "500"
OVER_VOLTAGE = "6"

# Set GPU/CPU memory split
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"

kraj, 2017-03-15:
I can confirm the crash as well.

zdobersek, 2017-03-15:
I think the problem is that the wpewebkit recipe is still using the USE_WPE_BACKEND_* configuration options, which were renamed to USE_WPEWEBKIT_BACKEND_*.

https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-wpe/wpewebkit/wpewebkit_0.1.bb#L51
https://github.com/WebPlatformForEmbedded/WPEWebKit/commit/2bc9b8d16447daae15205895fcd982244181b06f

wouterlucas, 2017-03-15:
Thanks for spotting that @zdobersek ! 
I'll have to reflect the Backend-rdk changes anyway so I'll try to adjust it and spin up a build when I get a chance

zdobersek, 2017-03-16:
I don't know about USE_WPE_INPUT_UDEV -- I don't see it in the current code under Source/ThirdParty/WPE-platform/, but it was added recently to the standalone library as USE_INPUT_DEV.

USE_WPE_BUFFER_MANAGEMENT_* options aren't used anymore in libWPEBackend-rdk, so you can remove them.

petrosagg, 2017-03-16:
I built the image after #133 and it seems to be working nicely! Thanks a lot for the quick response :)

he one thing that doesn't seem to be working and might be related to @zdobersek's comment is the mouse. I can move the westeros cursor but any clicks or mouse moves are not registered by WPE.

kraj, 2017-03-16:
USE_INPUT_DEV is set if USE_WPEWEBKIT_INPUT_LIBINPUT is set in the version thats being used with OE, its on 3ba32633268e76631849564c5eb576aa007937a0 of wpewebkit

petrosagg, 2017-03-31:
Thanks for fixing the issue. WPE no longer segfaults

closed 2017-03-31 by petrosagg