WebPlatformForEmbedded/meta-wpe #136: do_fetch: Fetcher failure

qmlsxxxx, 2017-03-23:
cppsdk-1.0+gitrAUTOINC+e3224d996c-r0 do_fetch: Fetcher failure for URL: 'git://git@github.com/Metrological/cppsdk.git;protocol=ssh;branch=master'. Unable to fetch URL from any source.

I can't find cppsdk.git in Metrological github
Does it exist ?
How can I fix this compie error?

Thanks

wouterlucas, 2017-03-23:
You can remove these two lines in the package group and it should go through:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-core/packagegroups/packagegroup-rpi-ml-wpe.bb#L15-L16

Sorry about that, cppsdk and webbridge will be removed from meta-wpe in the near future.

qmlsxxxx, 2017-03-27:
I also remove DEPENDS = "cppsdk" in 
playready_git.bb
libprovision_git.bb
webbridge_git.bb

and finally compile done.

closed 2017-03-27 by qmlsxxxx

kraj, 2017-03-27:
Can we add a global var for this please ? e.g.

set CPPSDK ??= "cppsdk"

in layer.conf

then in all recipes change DEPENDS_append = " ${CPPSDK}"

it will then become one knob to disable in local.conf if we want to disable it set 

CPPSDK = ""

in local.conf