WebPlatformForEmbedded/meta-wpe #138: Video playback regression

petrosagg, 2017-03-31:
There seems to be a regression with video playback. YoutubeTV was working fine at d4e5eff. However, on e1edf16 it displays the spinning loading icon indefinitely with no error logs. The regression happens with both the westeros backend and the rpi backend.

```
root@raspberrypi3:~# WPELauncher 
xkbcommon: ERROR: couldn't find a Compose file for locale "C"
ViewBackendWesteros: running as the nested compositor
Westeros Info: checking repeating composition supported: yes
Westeros Info: checking repeating composition supported: yes
Westeros Debug: calling wl_display_create
Westeros Debug: wl_display=0x6f700470
Westeros Debug: wl_display=0x6f700470 displayName=westeros-843-0
wstRenderGLSetupEGL: eglDisplay=0x1
wstRenderGLSetupEGL: eglInitiialize: major: 1 minor: 4
wstRendererGLSetupEGL: Number of EGL configurations: 28
wstRendererGLSetupEGL: eglChooseConfig: matching configurations: 18
config 0: red: 8 green: 8 blue: 8 alpha: 0 depth: 0
config 1: red: 8 green: 8 blue: 8 alpha: 0 depth: 0
config 2: red: 8 green: 8 blue: 8 alpha: 0 depth: 0
config 3: red: 8 green: 8 blue: 8 alpha: 0 depth: 24
config 4: red: 8 green: 8 blue: 8 alpha: 0 depth: 24
config 5: red: 8 green: 8 blue: 8 alpha: 0 depth: 0
config 6: red: 8 green: 8 blue: 8 alpha: 0 depth: 0
config 7: red: 8 green: 8 blue: 8 alpha: 0 depth: 24
config 8: red: 8 green: 8 blue: 8 alpha: 0 depth: 24
config 9: red: 8 green: 8 blue: 8 alpha: 8 depth: 0
choosing config 9
nativeWindow 0x6f71f8e8
wstRenderGLSetupEGL: eglSurface 0x1
wstRendererGLSetupEGL: eglContext 0x1
eglCreateImageKHR 0x74d2ec18
eglDestroyImageKHR 0x74d2f6d4
glEGLImageTargetTexture2DOES 0x74d30288
wayland-egl support expected, and is advertised by eglQueryString(eglDisplay,EGL_EXTENSIONS): proceeding to use 
eglBindWaylandDisplayWL 0x74d413c8
eglUnbindWaylandDisplayWL 0x74d41488
eglQueryWaylandBufferWL 0x74d41510
calling eglBindWaylandDisplayWL with eglDisplay 0x1 and wayland display 0x6f700470
have wayland-egl: 1
WstRendererCreate: module (libwesteros_render_gl.so.0) loaded and intialized
westeros-sb: WstSBInit: enter: display 0x6f700470
westeros-sb: WstSBInit: exit: display 0x6f700470 sb 0x6f71a1e0
westeros-simpleshell: WstSimpleShellInit: enter: display 0x6f700470
westeros-simpleshell: WstSimpleShellInit: exit: display 0x6f700470 shell 0x6f704f20
Westeros Debug: calling wl_display_run for display: westeros-843-0
[WPEInjectedBundle] Initialized.
didClearWindowObjectForFrame
[WPELauncher] Hello InjectedBundle ...
[WPEInjectedBundle] Hello!
	array item: Test1
	array item: Test2
	array item: Test3
```

petrosagg, 2017-03-31:
It also happens when building the `stable` branch with meta-openembedded and poky checked out to `morty`

wouterlucas, 2017-03-31:
Will push a fix for this in a bit, some of the flags where dropped when we did the library split. Flags that are still needed in the GST pipeline.

petrosagg, 2017-04-01:
@wouterlucas looks like it's fixed with 0d8f99c, thanks!

petrosagg, 2017-04-02:
I think I spoke too fast. It's only fixed for the westeros backend. If I compile with the RPI backend video still doesn't work

petrosagg, 2017-04-02:
turns out that if I manually enable the `gst-gl` PACKAGECONFIG video works for the RPI backend too. Is there any reason the gstreamer isn't selected by default?

wouterlucas, 2017-04-03:
There should be a way to toggle that automatically, thanks for letting me know. I'll add a change for that once I get a chance.

wouterlucas, 2018-01-05:
Fixed with 78d87802c1ef5f9e16faef3941419a0f9db00687

closed 2018-01-05 by wouterlucas