WebPlatformForEmbedded/meta-wpe #140: Is mouse input supported?

petrosagg, 2017-04-01:
This is more of a question rather than an issue, but when running WPE with the default compositor keyboard input works fine but the mouse doesn't.

I can see the yellow pointer on the screen and westeros recognises the pointer, but if I click on the web pages nothing happens. I'm testing by building latest master (0d8f99c0d137b176f98ecbaf49bd9c160612d506)

```
Westeros Info: display wayland-0 client pid 16 connected
Westeros Debug: wstSeatBind: client 0x75802e00 data 0x75800630 version 4 id 6
seat 0x7580afe0 caps: 3
  seat has pointer
  pointer 0x7580af80
```

petrosagg, 2017-04-02:
Some more info. If I use the rpi backend and set the `WPE_BCMRPI_CURSOR` environment variable, mouse input works correctly

markov00, 2018-04-26:
@petrosagg I've the same issue. I can see and move the yellow pointer but clicks doesn't work.

I've set also the WPE_BCMRPI_CURSOR variable together with the touch (I've the 7" official touchscreen)