WebPlatformForEmbedded/meta-wpe #146: cannot find the cppsdk repository

cometzero, 2017-05-12:
Hi,

During build, I cannot find the cppsdk from git://git@github.com/Metrological/cppsdk.git;protocol=ssh;branch=master

https://github.com/Metrological/cppsdk

Is the link obsoluted?

psivasubramanian, 2017-05-12:
@cometzero : No, it's restricted.

wouterlucas, 2017-05-13:
Hi @cometzero, these are still closed. Sorry about that, we should probably add this to a README. We're still upstreaming/open sourcing the CPP and Webbridge parts which should solve this.

You can remove them from the image for now and it should build with the open source WPELauncher until the new framework is ready.

For more info please see:
https://github.com/WebPlatformForEmbedded/meta-wpe/issues/130
https://github.com/WebPlatformForEmbedded/meta-wpe/issues/136

closed 2017-05-13 by wouterlucas

cometzero, 2017-05-15:
@psivasubramanian @wouterlucas Okay. I'll remove it until it's valid. Thanks.