WebPlatformForEmbedded/meta-wpe #149: wpewebkit is trying to install files into a shared area when those files already exist

s-maynard, 2017-07-12:
ERROR: wpewebkit-0.1+gitAUTOINC+6ac1eb13d5-r0 do_populate_sysroot: The recipe wpewebkit is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /home/smaynard/wpe/tmp/sysroots/raspberrypi3/usr/lib/libWPEBackend-default.so
 Matched in b'manifest-raspberrypi3-wpebackend-rdk.populate_sysroot'
Please verify which recipe should provide the above files.

Have you seen this?

Seems to happen after the first build...

wouterlucas, 2017-07-13:
No I haven't seen that one before, let me try to rebuild and see if I can reproduce this on my end.

s-maynard, 2017-07-13:
It happens for me on:
meta-wpe          = "HEAD:7abb159293f80d78ccd6cfdd13b458036b796b4d"
and
meta-wpe          = "HEAD:a571a24ce8d9bcdb01a784f0e2ee0a181bd4cdf7"

Possibly a race condition?  When it happens bb thinks it needs to rebuild webkit!

removing the tmp directory doesn’t help nor does adding the following line to the local.conf:
SSTATE_DUPWHITELIST = "${STAGING_DIR_HOST}${libdir}/libWPEBackend-default.so"

wouterlucas, 2017-07-13:
I tried a571a24ce8d9bcdb01a784f0e2ee0a181bd4cdf7 and 7ad3fbea4576a31af86b986c7be7a52887855f95 (latest) and both go fine for me. Though I haven't synced OE for a while. 

My build config:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta
meta-poky
meta-yocto-bsp    = "HEAD:9ea5a31776440abd6468f003c5e1905f079446d3"
meta-raspberrypi  = "master:11fec461330982ee15b4ac6dca5014fabe59198f"
meta-oe
meta-multimedia   = "HEAD:d3dd8bc3721f3958eb7ddeb026f7221d48daa986"
meta-wpe          = "master:7ad3fbea4576a31af86b986c7be7a52887855f95"

wouterlucas, 2017-08-24:
Fixed with 32326e0827855c39929a897316bb50769acff804 and 00540e8d7d42bfb5befc8e457e2f2531ee7ac6c0

closed 2017-08-24 by wouterlucas