WebPlatformForEmbedded/meta-wpe #151: getting Error while Adding bb layers

hardikharpal, 2017-08-03:
I am following your steps in My Ubuntu 16.04 LTS in virtual box  gettign following error...

`hardik@hardik-VirtualBox:~/rpi-yocto/rpi-ml-build$ bitbake-layers add-layer ../meta-raspberrypi
NOTE: Starting bitbake server...
ERROR: Unable to start bitbake server
ERROR: Last 10 lines of server log /home/hardik/rpi-yocto/rpi-ml-build/bitbake-cookerdaemon.log:
    function()
  File "/home/hardik/rpi-yocto/poky/bitbake/lib/bb/server/process.py", line 403, in _startServer
    self.cooker = bb.cooker.BBCooker(self.configuration, self.featureset, self.readypipein)
  File "/home/hardik/rpi-yocto/poky/bitbake/lib/bb/cooker.py", line 197, in __init__
    self.initConfigurationData()
  File "/home/hardik/rpi-yocto/poky/bitbake/lib/bb/cooker.py", line 348, in initConfigurationData
    self.databuilder.parseBaseConfiguration()
  File "/home/hardik/rpi-yocto/poky/bitbake/lib/bb/cookerdata.py", line 323, in parseBaseConfiguration
    raise bb.BBHandledException
bb.BBHandledException
`

kraj, 2017-08-03:
you should install chrpath makeinfo on your build system.

hardikharpal, 2017-08-04:
@kraj Thank you

closed 2017-08-04 by hardikharpal

rapveen, 2019-05-03:
Yes, it did work for me!!

eariko, 2019-07-11:
Works perfectly