WebPlatformForEmbedded/meta-wpe #164: netflix build issues

karora123, 2017-10-31:
Hi,
Enabling netflix in webkit image throwing build error. Seems like git repository for netflix and dependent components like ccpsdk, libprovision and playready does not exists.
Is the git repository for respective components have been moved and need to be corrected out in the recipe files?
Error as given below:-
-----------------------------Error----------------------------------------------------
ERROR: cppsdk-1.0+gitrAUTOINC+e3224d996c-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/pritam/KUL/hyb/build-raspberrypi-rdk-hybrid/../downloads/git2/github.com.Metrological.cppsdk.git'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

ERROR: cppsdk-1.0+gitrAUTOINC+e3224d996c-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://git@github.com/Metrological/cppsdk.git;protocol=ssh;branch=master'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/pritam/KUL/hyb/build-raspberrypi-rdk-hybrid/tmp/work/cortexa7t2hf-neon-vfpv4-rdk-linux-gnueabi/cppsdk/1.0+gitrAUTOINC+e3224d996c-r0/temp/log.do_fetch.13622
ERROR: libprovision-1.0.gitrAUTOINC+cfa6e55f82-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/pritam/KUL/hyb/build-raspberrypi-rdk-hybrid/../downloads/git2/github.com.Metrological.libprovision.git'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

ERROR: libprovision-1.0.gitrAUTOINC+cfa6e55f82-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://git@github.com/Metrological/libprovision.git;protocol=ssh;branch=master'. Unable to fetch URL from any source.
ERROR: playready-1.0.gitrAUTOINC+3f1ed46727-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/pritam/KUL/hyb/build-raspberrypi-rdk-hybrid/../downloads/git2/github.com.Metrological.playready.git'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

ERROR: playready-1.0.gitrAUTOINC+3f1ed46727-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://git@github.com/Metrological/playready.git;protocol=ssh'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/pritam/KUL/hyb/build-raspberrypi-rdk-hybrid/tmp/work/cortexa7t2hf-neon-vfpv4-rdk-linux-gnueabi/libprovision/1.0.gitrAUTOINC+cfa6e55f82-r0/temp/log.do_fetch.13623
ERROR: Logfile of failure stored in: /home/pritam/KUL/hyb/build-raspberrypi-rdk-hybrid/tmp/work/cortexa7t2hf-neon-vfpv4-rdk-linux-gnueabi/playready/1.0.gitrAUTOINC+3f1ed46727-r0/temp/log.do_fetch.13625
ERROR: Task 661 (/home/pritam/KUL/hyb/meta-metrological/recipes-metrological/cppsdk/cppsdk_git.bb, do_fetch) failed with exit code '1'
ERROR: Task 229 (/home/pritam/KUL/hyb/meta-metrological/recipes-metrological/libprovision/libprovision_git.bb, do_fetch) failed with exit code '1'
---------------------------------------------------------------------------------

regards,
Kulbhushan

wouterlucas, 2017-11-01:
Hi Kulbhushan,

Those are indeed on private repos, since the Netflix port and our integration is not something we can freely share without getting into licensing trouble. Enabling those without having access will throw the above errors. 

Thanks,
Wouter

karora123, 2017-11-01:
Hi Wouter,

Thanks for your update !! So what is the way to enroll buying license with available plan?

regards,
Kulbhushan

wouterlucas, 2017-11-01:
It depends on what you're planning to do with it, whether it is for evaluation or planning to put it in a product. There are various ways we can work together.

Best is to reach out to us directly in one your communication method of choice: https://www.metrological.com/contact.html

hhromic, 2017-12-28:
hi guys!
I've been trying to build a Yocto image uncessfully. I tried the `master` branch during `repo init` and `repo sync` with a patching errors (probably due to bleeding-edge development) so I tought of instead checking out the `rocko` image. This time it stops trying to compile libprovision which as you mentioned is private. However I have the opposite problem, I don't need netflix or libprovision and I would like the image to *not* build those components.
I'm completely new to Yocto and I'm learning, however I have no idea where to look for to disable those packages/components from the image. Can you give me some pointers? Thanks!

karora123, 2017-12-28:
Hi,
You can update build director conf file with removing netflix & libprovision from target variant. Eg. in file <build-directory>/conf/local.conf you can add <Build-target>_remove=" libprovision netflix ". This is one example. There can be multiple way to fix.

hhromic, 2017-12-28:
hi @karora123 ,
thanks for your answer. I tried on adding:

    DISTRO_FEATURES_remove = " x11 libprovision netflix "

Into `conf/local.conf` as you suggested but Yocto again tried to build those components:

    ERROR: libprovision-2.0.gitrAUTOINC+e532b8d115-r0 do_fetch: Fetcher failure for URL: 'git://git@github.com/Metrological/libprovision.git;protocol=ssh;branch=WPEFramework'. Unable to fetch URL from any source.

I also cleaned the build directory completely before attempting the build after the suggested changes.

Can you provide me with a more complete example of your suggestion? I suspect I'm not doing this correctly. Thanks!

karora123, 2017-12-28:
Hi Not for DISTRO_FEATURE, you need to update <target-image>_remove, sorry for types. Even you can find responsible package/target adding these components, you can use same name to remove from conf file.

wouterlucas, 2017-12-28:
Which image are you building? wpe-westeros-image shouldn't include those packages. Only the wpe-framework-image and wpe-rpi-image rely on the WPE Framework package group which will in turn enable those packages.

hhromic, 2017-12-28:
hi @wouterlucas ,
I'm indeed using `wpe-westeros-image`. I started by following the instructions in the Wiki for the Raspberry Pi. The only change I'm doing is using `rocko` instead of `master` as the revision while initialising and syncing the repositories with `repo`.  I first tried the `master` branch but it fails with patching error for the userland package. As for the target machine, I'm using `raspberrypi0-wifi`. This is how my `conf/local.conf` file looks like:

    MACHINE = "raspberrypi0-wifi"
    GCCVERSION="6.%"
    DISTRO_FEATURES_remove = " x11 "
    DISTRO_FEATURES_append = " opengl wayland "
    CONF_VERSION = "1"

I would like to help debugging this. Let me know any kind of information you need. I can try to build again with the `master` revision to get you the exact patching error. Some months ago I tried with the buildroot repository to build a WPE image without success. Hence I'm moving to Yocto, hoping the machine abstraction is easier and I can successfully build an image for the RPI0W.

Thanks a million!

wouterlucas, 2017-12-28:
hi @hhromic seperate issue or this one, doesnt matter to me :)

Can you try reverting d1066f8b74f822a7f88ad85ba095272fd399b323 ? I suspect we need to split out initramfs images.

wouterlucas, 2017-12-28:
And you might want to set
```
PREFERRED_VERSION_harfbuzz = "1.4.%"
``

in your local conf, the netflix backports for harfbuzz are messing up pango. But since you dont need them, make sure bitbake selects the version from poky/oe. I'll probably drop these backports in the rocko/morty branches, since the NRDP 4.2 app is due to be replaced anyway to be more compatible with the newer libraries (as NRDP 4.3/5.0 should be).

hhromic, 2017-12-30:
Hi @wouterlucas,

Sorry for the delay replying to your messages. After your suggestions, I decided to embark on some tests to get the picture clear for me and the current state of building `meta-wpe`.

Firstly, I can confirm the images are compiling successfully for the `master` and `rocko` branches. I guess at that moment I tried them before there was some inconsistency in the repositories that now are gone. Please see below the results of my tests for each. Thanks a lot for your feedback as that definitively helped me understand better what goes on and to make it work.

I hope these results help you on developing the `meta-wpe` repository further. Regarding `freetype` and `harfbuzz` as you mentioned, I didn't need to force a preferred version for the `rocko` branch but I did have for the `master` branch. Also I didn't have to apply the revert you suggested as now the build didn't try to include Netflix nor libprovision.

If `meta-wpe` doesn't need patched or backported versions of `freetype` or `harfbuzz`, probably is better to remove them and leave the originals that come with `poky`. This will avoid warnings and do a cleaner, more up-to-date builds.

The steps I outline below can be used to improve the Wiki page as well for total beginners, as those are the exact steps I needed to do to get a working build using Yocto from scratch.

**Warning:** I just built the images using Yocto, but haven't tried them in real hardware yet. As soon as I regain access to the real hardware I own, I will give those images a try and report back.

**SUMMARY**

* Tests done for: the `master` and `rocko` branches.
* Tests building images for: `raspberrypi3` and `raspberrypi0-wifi` target machines.
* Images build out-of-the box with the `rocko` branch, require `poky` versions of `freetype` and `harfbuzz` for the `master` branch.

# Using the `main` branch

## Setup Steps

    $ mkdir -p rpi-yocto && cd rpi-yocto
    $ repo init -u git://github.com/WebPlatformForEmbedded/meta-wpe.git -b master -m tools/manifests/rpi-yocto.xml
    $ repo sync
    $ source poky/oe-init-build-env
    $ bitbake-layers add-layer ../meta-openembedded/meta-oe
    $ bitbake-layers add-layer ../meta-openembedded/meta-python
    $ bitbake-layers add-layer ../meta-openembedded/meta-multimedia
    $ bitbake-layers add-layer ../meta-openembedded/meta-networking
    $ bitbake-layers add-layer ../meta-raspberrypi
    $ bitbake-layers add-layer ../meta-wpe

## First Attempt

    $ cat <<'__EOF__' >> conf/local.conf
    
    # configuration
    MACHINE = "raspberrypi3"  # or other, e.g.: "raspberrypi0-wifi"
    DISTRO_FEATURES_remove = "x11"
    DISTRO_FEATURES_append = " opengl"
    __EOF__
    $ bitbake wpe-westeros-image

**OUTPUT**

    WARNING: freetype-2.4.6-r1 do_configure: QA Issue: freetype: configure was passed unrecognised options: --with-png --without-harfbuzz [unknown-configure-option]
    ERROR: fontconfig-2.12.6-r0 do_compile: Function failed: do_compile (log file is located at /home/yocto/rpi-yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/temp/log.do_compile.13869)
    | ../../fontconfig-2.12.6/src/fcfreetype.c: In function 'IA__FcFreeTypeQueryFace':
    | ../../fontconfig-2.12.6/src/fcfreetype.c:1273:47: error: 'FT_SFNT_OS2' undeclared (first use in this function); did you mean 'TT_OS2'?
    |      os2 = (TT_OS2 *) FT_Get_Sfnt_Table (face, FT_SFNT_OS2);
    |                                                ^~~~~~~~~~~
    |                                                TT_OS2

**RESULTS**

* `fontconfig` (from `poky`) is not compiling due to an old version of `freetype` from `meta-wpe`.

## Second Attempt

    $ # Add `PREFERRED_VERSION_freetype = "2.8.%"` to `conf/local.conf`
    $ bitbake wpe-westeros-image

**OUTPUT**

    ERROR: pango-1.40.12-r0 do_configure: Function failed: do_configure (log file is located at /home/yocto/rpi-yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/pango/1.40.12-r0/temp/log.do_configure.23936)
    | configure: error: *** Could not enable any backends.
    | *** Must have at least one backend to build Pango.
[... 51 lines cut ...]
    $ bitbake-layers add-layer ../meta-openembedded/meta-oe
    $ bitbake-layers add-layer ../meta-openembedded/meta-python
    $ bitbake-layers add-layer ../meta-openembedded/meta-multimedia
    $ bitbake-layers add-layer ../meta-openembedded/meta-networking
    $ bitbake-layers add-layer ../meta-raspberrypi
    $ bitbake-layers add-layer ../meta-wpe
    $ cat <<'__EOF__' >> conf/local.conf
    
    # configuration
    MACHINE = "raspberrypi3"  # or other, e.g.: "raspberrypi0-wifi"
    DISTRO_FEATURES_remove = "x11"
    DISTRO_FEATURES_append = " opengl"
    __EOF__
    $ bitbake wpe-westeros-image

**OUTPUT**

    WARNING: freetype-2.4.6-r1 do_configure: QA Issue: freetype: configure was passed unrecognised options: --without-harfbuzz --with-png [unknown-configure-option]
    WARNING: harfbuzz-1.4.8-r0 do_configure: QA Issue: harfbuzz: invalid PACKAGECONFIG: graphite [invalid-packageconfig]

**RESULTS**

* `fontconfig` (from `poky`) does compile regardless of old version of `freetype` from `meta-wpe`.
* `pango` (from `poky`) does compile because `harfbuzz` comes from `poky` and not from `meta-wpe`.
* `wpe-westeros-image` **built successfully** for `raspberrypi3`, but doesn't build for `raspberrypi0-wifi`. The package `mesa-gl` fails to compile. I will investigate further why of this.

wouterlucas, 2017-12-31:
Hi @hhromic thanks for the extremely detailed writeup! Much appreciated. 

Master needs to be updated with the rocko branch changes. As for the freetype/harfbuzz/etc backports, right now Netflix requires an older version of freetype. The API changed and is no longer compatible and it will crash if you run with the latest version of freetype. We should be adding a new release to the repo soon which means we can drop those backports, assuming they're no longer required.

hhromic, 2018-01-01:
great to hear @wouterlucas!
What would you recommend me to do? use the `master` or the `rocko` branch? I'm trying to build and run WPE for a `raspberrypi0-wifi`. As I found out in the tests, it does compile fine using `master` but doesn't in `rocko`. If you recommend me to use the latter instead of master, then I will try to find out/investigate why it doesn't compile. Hopefully is some library versioning and not hardware support.

Thanks and happy new year!

wouterlucas, 2018-01-02:
Thanks! Happy NY to you too!

I'm having the opposite actually, Rocko seems fine for me (hence that's what my default setup is poining at) while master continues to fluctuate too much for my likes and i usually hate wasting time to debug an upstream issue while theres still plenty of work to be done over here. Yocto 2.5 (current master) is scheduled to be released in April, so I'm anticipating there could be quite some changes still trickling through.

Is your issue with rpi0 wifi strickly with poky/oe or is it meta-raspberry? Else you could also keep everything else on rocko and put meta-raspberrypi to master for the time being.

hhromic, 2018-01-04:
hi @wouterlucas ,

I found out the `mesa-gl` building problem for `raspberrypi0-wifi` in the `rocko` branch is indeed related to `poky`. See this commit from @jumpnow : https://github.com/jumpnow/meta-rpi/commit/bbbd3aa0ab6d27251b88866b9f35bd5f9ad6f732

As you suggested, I will compile it with `meta-raspberrypi` set to the `master` branch and everything else to the `rocko` branch.

Btw, you should check out @jumpnow 's work about Yocto the raspberrypi: www.jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Yocto.html

Cheers and thanks for all the pointers!
Hugo.

wouterlucas, 2018-01-04:
Glad it worked out! And you're welcome :)
And thanks for the jumpnow's link! looks very comprehensive.

wouterlucas, 2018-01-05:
Fixed with 49ddffb97d64ff5a55f5a5d46b2dd31e6c46ae0c

closed 2018-01-05 by wouterlucas