WebPlatformForEmbedded/meta-wpe #167: Playing video is very jittery

kraj, 2017-11-15:
```
WARNING: from element /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink: A lot of buffers are being dropped.
Additional debug info:
../../../../gstreamer-1.12.3/libs/gst/base/gstbasesink.c(2901): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink:
There may be a timestamping problem, or this computer is too slow.
WARNING: from element /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink: A lot of buffers are being dropped.
Additional debug info:
../../../../gstreamer-1.12.3/libs/gst/base/gstbasesink.c(2901): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink:
There may be a timestamping problem, or this computer is too slow.
WARNING: from element /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink: A lot of buffers are being dropped.
Additional debug info:
../../../../gstreamer-1.12.3/libs/gst/base/gstbasesink.c(2901): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink:

```

its a regression

run the below cmd
```
gst-launch-1.0 souphttpsrc location="http://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_720p_h264.mov" ! typefind ! qtdemux name=demux demux. ! queue ! h264parse ! omxh264dec ! glimagesink demux. ! queue ! faad ! autoaudiosink
```

wouterlucas, 2017-12-04:
I've synced the latest patches for 1.12.x, that should improve it a bit. We're still locked in for 1.10.x and thats where most of the cycles go, hopefully we can get some more love for 1.12.x soon.

hhromic, 2018-05-29:
we are in 1.14.x now :)  is this a stale issue ? :)

wouterlucas, 2018-05-29:
I dont mind to keep it open until we validate its gone on 1.12 or 1.14.

TomMettam, 2018-09-02:
Still getting this issue, unable to play 1080p video smoothly

wouterlucas, 2019-07-25:
Works fine for GST 1.16 and 1.14.

closed 2019-07-25 by wouterlucas