WebPlatformForEmbedded/meta-wpe #168: WARNING: No recipes available for Gstreamer

samouds, 2017-11-27:
Hello,

I am using the latest version of meta-wpe (https://github.com/WebPlatformForEmbedded/meta-wpe/commit/bb6d3bbf21cb071dbec227ebb69954b70b63c3e2)

But I get this error at build : 

```
WARNING: No recipes available for:
  /home/samoud/resin-wpe/base-image/build/../meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.12.3.bbappend
  /home/samoud/resin-wpe/base-image/build/../meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.12.3.bbappend
  /home/samoud/resin-wpe/base-image/build/../meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-good_1.12.3.bbappend
  /home/samoud/resin-wpe/base-image/build/../meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly_1.12.3.bbappend
  /home/samoud/resin-wpe/base-image/build/../meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0_1.12.3.bbappend
```

I checked the path /home/samoud/resin-wpe/base-image/build/../meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.12.3.bbappend and it exists.

Any idea ?

wouterlucas, 2017-11-27:
Which version of oe-core/yocto are you using to generate your build?

samouds, 2017-11-27:
I am using the rocko branch

wouterlucas, 2017-11-27:
And which branch are you using for poky / openembedded-core recipes? Or if you are using a custom distro check the gstreamer1 version, usually this happens if the version of gstreamer is older (or newer) then the bbappends that we provide in meta-wpe. Krogoth OE/Yocto comes with Gstreamer 1.12.x, hence the patches we provide on top of that.

However older releases of Yocto will come with older versions of gstreamer, like morty comes with gst 1.8.x, pyro 1.10.x, etc.

samouds, 2017-11-27:
I'm using the rocko branch for : 

meta-raspberrypi
meta-openembedded
poky

and master for meta-wpe and that's it.

I thought you meant openembedded-core when you said oe-core/yocto that's the same no ?

wouterlucas, 2017-11-27:
Yeah, wasn't sure if you're using poky or some custom distro. So poky rocko has Gstreamer 1.12.2: https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer?h=rocko

Please use our rocko branch, which has the 1.12.2 bbappends: 
https://github.com/WebPlatformForEmbedded/meta-wpe/tree/rocko/recipes-multimedia/gstreamer

Master has been aligned with upstream master poky, which is already on 1.12.3 :-)

samouds, 2017-11-28:
And how about the last fix (https://github.com/WebPlatformForEmbedded/meta-wpe/commit/bb6d3bbf21cb071dbec227ebb69954b70b63c3e2), it's on the rocko branch too ?

wouterlucas, 2017-11-28:
It is now: https://github.com/WebPlatformForEmbedded/meta-wpe/commit/f873ca9a09ea7b6c2960d7706cf0948f6b0dc6cf :)

samouds, 2017-11-28:
It's working ! thanks a lot !

closed 2017-11-28 by samouds