WebPlatformForEmbedded/meta-wpe #169: Building for other target machines "eglplatform.h:146:22: fatal error: X11/Xlib.h: No such file or directory"

ffarhour, 2017-12-04:
Hi
I've been trying to build an image for other currently unsupported target machines (specifically the imx6qsabresd). I can easily create a an image for my target device without the WPE layer. But with the WPE layer I get the following error:
`Building for other target machines "eglplatform.h:146:22: fatal error: X11/Xlib.h: No such file or directory"`
My local.conf file consists of the following additions:
```
MACHINE ??= 'imx6qsabresd'
DISTRO ?= 'fsl-imx-wayland'
... 
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
BB_NUMBER_THREADS = "4" 
PARALLEL_MAKE = "-j 4"
```
I also made certain changes to the westeros source files. I had previously made more changes, but with your new push 3 days ago for the 96boards I reduced it to the following:
*meta-wpe/recipes-graphics/westeros.bb*
```
...
westeros-soc_imx6qsabresd-32 = "westeros-soc-drm"
...
```

*meta-wpe/recipes-graphics/westeros-soc-drm.bb* (Note the added new machine)
```
COMPATIBLE_MACHINE = "(hikey-32|dragonboard-410c-32|dragonboard-820c-32|poplar|imx6qsabresd)"
``` 

However when westeros is compiling I get the error I described above. I'm not sure where I may have gone wrong, but I have a feeling I've missed a step somewhere, and was hoping you could help me with this. Thanks :)

Final note: I also traced where the error was from, and realised that for some reason when EGL is being compiled as a dependency for Westeros, the macro define `WL_EGL_PLATFORM`, and it defaults to using X11 instead.

wouterlucas, 2017-12-04:
What kind of Wayland/EGL instance do you support on the freescale device? Using libdrm/MESA? Because that's the only 'standard' way of running Wayland EGL. The other possibility is to add direct EGL SOC support under Westeros, e.g. like the rpi: https://github.com/rdkcmf/westeros/blob/master/rpi/westeros-gl/westeros-gl.cpp

ffarhour, 2017-12-04:
Thanks for your prompt reply. I'm very new to this all, so apologies if my questions seem very simple.
From my understanding the device supports libdrm/mesa. I have seen them as well in the freescale provided sources for Yocto. 
But I'm confused as to which dependency causes this error.

wouterlucas, 2017-12-04:
No worries, if you have libdrm/MESA it shouldnt need its dedicated SOC renderer. However Westeros will rely on .pc files to see if it has Wayland EGL. That happens here: https://github.com/rdkcmf/westeros/blob/8a6d991a6df0bcf65d81906ac5c1fed707763ee9/configure.ac#L71

However the error message looks to trickle down to eglplatform.h, which should be provided by MESA3d. I think (judging from a distance) your error resides in mesa, did that by any chance compile as x11 instead of with the --enable-egl/--enable-gles1/--enable-gles2 flags?
One way to test that is to not build westeros at this point and take the rootfs + mesa/libdrm/egl stuff and see a standard opengl test app would work (e.g. https://github.com/glmark2/glmark2). 

I'd check the mesa configure flags/log and if that's correctly setup. And check if the westeros package recipe-sysroot contains the right package config files it needs to detect Wayland EGL. In this case the following should be present:
```
-rw-r--r-- 23 wouter wouter 281 Nov 30 22:48 wayland-client.pc
-rw-r--r-- 23 wouter wouter 209 Nov 30 22:48 wayland-cursor.pc
-rw-r--r-- 20 wouter wouter 223 Nov 30 22:53 wayland-egl.pc
-rw-r--r-- 23 wouter wouter 187 Nov 30 22:48 wayland-scanner.pc
-rw-r--r-- 23 wouter wouter 288 Nov 30 22:48 wayland-server.pc
```

ffarhour, 2017-12-12:
Thanks for your help. This part of my project is currently on the back burner, but your suggestions are helpful. Once I get this working I will update everyone with a comment.