WebPlatformForEmbedded/meta-wpe #170: Mouse does not work

mounirkhenfir, 2017-12-12:
Hi, 
I've built wpe following https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/Raspberry-PI.  When running on a pi 3 with an USB mouse connected, the pointer works under westeros but not under the WPE browser. Is there anyway/flag to make it work?

kraj, 2018-06-22:
@jeffwannamaker this seems to be a case that more than one user has seen, Is there some options we need to enable for mouse to work in WPE when launched using westeros compositor.

kraj, 2018-06-22:
This issue is same as #140

jeffwannamaker, 2018-06-22:
Khem,

This is an issue with the WPE Westeros backend that has been known for a long time.

The WPEWebProcess is supposed to connect to the nested wayland display created by the WPE UI process but it doesn’t.  Instead it directly connects to the same upstream wayland display that the UI process is connecting to.  This causes wayland mouse input to not work in WPE.

In westeros/view-backend.cpp in ViewBackend::ViewBackend a nested compositor instance is created, its display name is obtained, and then set to WAYLAND_DISPLAY:

    const char* nestedTargetDisplay = std::getenv("WAYLAND_DISPLAY");
    if (nestedTargetDisplay)
    {
        fprintf(stderr, "ViewBackendWesteros: running as the nested compositor\n");
        WstCompositorSetIsNested(compositor, true);
        WstCompositorSetIsRepeater(compositor, true);
        WstCompositorSetNestedDisplayName(compositor, nestedTargetDisplay);
        //Register for all the necessary callback before starting the compositor
        input_handler->initializeNestedInputHandler(compositor);
        output_handler->initializeNestedOutputHandler(compositor);
        const char * nestedDisplayName = WstCompositorGetDisplayName(compositor);
        setenv("WAYLAND_DISPLAY", nestedDisplayName, 1);
    }

Then in westeros/renderer-backend.cpp in Backend::Backend the WebProcess connects to the default wayland display:

Backend::Backend()
{
    m_display = wl_display_connect(nullptr);

This will connect to the display named by the WAYLAND_DISPLAY env var which is intended to now be the value set in ViewBackend::ViewBackend but it is not.  It still has the same value obtained for nestedTargetDisplay.  This causes mouse input to not work.   It seems like some other mechanism,  IPC perhaps, needs to be used to pass the nested display name from the UI process to the WebProcess to be used by Backend::Backend.

Ihor Ivlev pointed this out back in 2016 and offered a fix.  I don’t recall if the fix was not reliably effective, or if it has been lost:

FROM: Ihor Ivlev <iivlev@productengine.com<mailto:iivlev@productengine.com>>
SENT: Thursday, December 29, 2016 8:11 AM
TO: Wouter-lucas van Boesschoten; zdobersek@igalia.com<mailto:zdobersek@igalia.com>; Sharma,
Sudhanshu; Ravuru, Anil; Borovkov, Igor; Mutavchi, Ievgen; Robinson,
John; Wannamaker, Jeff; Ansari, AzamAyub (Contractor)
SUBJECT: Backend initialization order
Hi Wouter, Zan
When Westeros backend is used for WPE,
There currently is a problem: Web process connects to an incorrect
display:
It connects to Nested instead of Repeater compositor.
This is caused by incorrect initialization order when View is created:
During WKView creation,
1) Web process is started
2) wpe_backend initialization routine initializes Westeros and calls
setenv to set WAYLAND_DISPLAY environment variable but this doesn't
have any effect on the WebProcess which is already started.
Please review
https://github.com/Metrological/WebKitForWayland/pull/154 which
changes the initialization order, thus WAYLAND_DISPLAY will be set to
a correct value in Web Process.
Thank you!