WebPlatformForEmbedded/meta-wpe #171: repo is not working "Permission denied (publickey)."

malouf-erfan, 2017-12-22:
Steps:
repo init -u ssh://git@github.com/WebPlatformForEmbedded/meta-wpe.git -b master -m tools/manifests/rpi-yocto.xml

Result :

Warning: Permanently added the RSA host key for IP address '192.30.255.113' to the list of known hosts.
Permission denied (publickey).
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Permission denied (publickey).
fatal: Could not read from remote repository.

wouterlucas, 2017-12-22:
Please make sure GitHub knows who you are if you are going to check something out through SSH. This can be done by adding your public key to the GitHub user. 

Follow the steps layed out in https://help.github.com/articles/adding-a-new-ssh-key-to-your-github-account/ and it should work fine!

malouf-erfan, 2017-12-23:
Thanks for the quick response

closed 2017-12-23 by malouf-erfan