WebPlatformForEmbedded/meta-wpe #172: wpe, WPELuncher erros out on RPI3

malouf-erfan, 2017-12-23:
Hello,

I build an image using the master branch for raspberry pi 3:

I added below to my local conf:
MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = "opengl"
GCCVERSION = "6.%"

after build everything boots up but when I run the command WPELuncher I get the following error messages:

xkbcommon: ERROR: couldn't find a Compose file for local "C"
ViewBackendWesteros: failed to start the compositor: Error. A renderer module must be supplied
error: XDG_RUNTIME_DIR not set in the environment.

I ried running wpe by itself and I got this error:
/usr/bin/wpe: line 57: cd: /root: No such file or directory
I created the /root directory then got the same error as above.

I appreciate any help, I am hoping to be able to display a fullscreen webpage on the RPI3.

Thanks

Sch1nken, 2017-12-26:
Same Problem here. Same settings as you have (though without the forced GCCVERSION). 

You can fix the "/usr/bin/wpe: line 57: cd: /root: No such file or directory" error by just creating a /root folder with mkdir. Doesn't help much though :<

fellowgeek, 2017-12-27:
@Sch1nken we need a support group so when someone got it working we all could benefit. I am @fellowgeek in Telegram hit me up.

wouterlucas, 2017-12-27:
Sorry guys I havent been able to circle around to this, my current yocto is in a totally different config so I can't cross check. 

However make sure the exports are set if you run westeros and wpelauncher. E.g.:

Westeros:
```
export XDG_RUNTIME_DIR=/tmp
export WAYLAND_DISPLAY=wayland-0
export LD_PRELOAD=/usr/lib/libwayland-egl.so.0.0.0
export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0
```

And same before WPE Launcher:
```
export XDG_RUNTIME_DIR=/tmp
export WAYLAND_DISPLAY=wayland-0
```

That should work just fine. Try stopping /etc/init.d/westeros and running westeros in fore ground with those exports. If that still doesnt work let me know and I'll cook up a Westeros RPI build to cross check the status of master.

malouf-erfan, 2017-12-27:
@wouterlucas thank you so much, stoping and starting westeros did the trick, can you tell me how to hide the big mouse cursor and how to set the output to 1080p instead of 720p

jeffwannamaker, 2017-12-27:
Hi,

If don’t want the westeros reference compositor to display a mouse cursor, remove the “--enableCursor” option from its invocation command line.

By default, the westeros uses a display resolution of 720p.  To run at a different resolution the westeros API’s WstCompositorResolutionChangeBegin and WstCompositorResolutionChangeEnd must be called.  To do this this with the reference compositor a westeros module can be used.  I’m attaching a tarball with the sources for an example module for RPI.  This module just uses the RPI API graphics_get_display_size to determine the current display resolution and then calls WstCompositorResolutionChangeBegin and WstCompositorResolutionChangeEnd to move the compositor to this resolution:

   displayId= DISPMANX_ID_MAIN_LCD;
   int32_t result= graphics_get_display_size( displayId,
                                              &width,
                                              &height );

   if ( result == 0 )
   {
      printf("module: display %dx%d\n", width, height);
      WstCompositorResolutionChangeBegin( compositor );
      WstCompositorResolutionChangeEnd( compositor, width, height );
   }

If you modify the RPI resolution by editing its boot config.txt to, for example, change to 1080p by changing hdmi_mode from 4 to 16 and modify the westeros reference compositor command line to include “--module libresolutionmodule.so.0.0.0” then the compositor will run at 1080p.

-Jeff

jeffwannamaker, 2017-12-28:
[westeros-resolutionmodule-rpi.zip](https://github.com/WebPlatformForEmbedded/meta-wpe/files/1590236/westeros-resolutionmodule-rpi.zip)

fellowgeek, 2017-12-28:
Thanks I will give it a shot tonight

wouterlucas, 2017-12-28:
Tagging @bramoosterhuis, can we add something similar to the WPEFramework implementation of Westeros?

wouterlucas, 2018-01-05:
Fixed with 853366221e1b30505bcbf2a36526b61fcf6c8f37 and ea19b64e753faefdca17ccb968f7eb5cbcb4f929

closed 2018-01-05 by wouterlucas

fellowgeek, 2018-01-06:
@jeffwannamaker Hi, thanks for creating the "libresolutionmodule", I really appreciate it.

I compiled and used the module you provided, but wpe still running in a 720p window at top right corner of my 1080p display. Any help would be appreciated.

please see the output below:

westeros --module /usr/lib/libresolutionmodule.so.0.0.0 --renderer /usr/lib/libwesteros_render_dispmanx.so.0 --enableCursor --framerate 60 --display wayland-0
found /dev/input/event1
opened device /dev/input/event1 : fd 4
found /dev/input/event0
opened device /dev/input/event0 : fd 5
Westeros Debug: calling wl_display_create
Westeros Debug: wl_display=0x756005b8
Westeros Info: removing temp file: /tmp/westeros-943-JIz2bz
Westeros Info: removing temp file: /tmp/westeros-905-zPbpQz
Westeros Debug: wl_display=0x756005b8 displayName=wayland-0
wstRenderDMXCreate: eglInitiialize: major: 1 minor: 4
wstRendererDMXCreate: wayland-egl support expected, and is advertised by eglQueryString(eglDisplay,EGL_EXTENSIONS): proceeding to use 
wstRendererDMXCreate: eglBindWaylandDisplayWL 0x76e77470
wstRendererDMXCreate: eglUnbindWaylandDisplayWL 0x76e77534
wstRendererDMXCreate: eglQueryWaylandBufferWL 0x76e775bc
wstRendererDMXCreate: calling eglBindWaylandDisplayWL with eglDisplay 0x1 and wayland display 0x756005b8
wstRendererDMXCreate: have wayland-egl: 1
WstRendererCreate: module (/usr/lib/libwesteros_render_dispmanx.so.0) loaded and intialized
westeros-sb: WstSBInit: enter: display 0x756005b8
westeros-sb: WstSBInit: exit: display 0x756005b8 sb 0x75602ea0
westeros-simpleshell: WstSimpleShellInit: enter: display 0x756005b8
westeros-simpleshell: WstSimpleShellInit: exit: display 0x756005b8 shell 0x75602ee0
Westeros Debug: calling moduleInit for module (/usr/lib/libresolutionmodule.so.0.0.0)...
module: display 1920x1080
Westeros Debug: WstCompositorResolutionChangeBegin
Westeros Debug: WstCompositorResolutionChangeBegin: exit
Westeros Debug: WstCompositorResolutionChangeEnd: (1920 x 1080)
Westeros Debug: WstCompositorResolutionChangeEnd: exit
Westeros Debug: done calling moduleInit for module (/usr/lib/libresolutionmodule.so.0.0.0) result 1
Westeros Debug: calling wl_display_run for display: wayland-0
Unable to set default cursor: error: (Bad state.  Cannot set default cursor unless the compositor is running)
Westeros Debug: wstCompositorBind: client 0x75602f88 data 0x1c01188 version 1 id 5
Westeros Debug: wstUpdateClientInfo: create clientInfo for client 0x75602f88
Westeros Info: display wayland-0 client pid 1086 connected
Westeros Debug: wstOutputBind: client 0x75602f88 data 0x75600718 version 2 id 6
Westeros Debug: wstSeatBind: client 0x75602f88 data 0x756007a0 version 4 id 7
Westeros Debug: wstGetSurfaceInfo: create surfaceInfo for resource 0x75607828
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x75602f88
Westeros Debug: wstICompositorCreateSurface: client 0x75602f88 resource 0x75607528 id 10 : surface resource 0x75607828
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x75602f88
Westeros Debug: wstKeyboardCheckFocus: no key focus yet
Westeros Debug: wstKeyboardCheckFocus: no key focus yet: client 0x75602f88 has listeners 0
broadcast for surfaceId 1 name noname
Westeros Info: display wayland-0 client pid 1086 disconnected
Westeros Debug: wstDestroySurfaceCallback resource 0x75607828 free surfaceInfo
Westeros Debug: wstSurfaceDestroy: surface 0x75607870 refCount 1

Thanks

jeffwannamaker, 2018-01-06:
Hi Erfan,

From your log output it looks like the compositor resolution is being set to 1080p successfully but WPE is not changing its window size.  Can you check that your westeros is at the latest revision of 44133c6daca01d1126457c5dfa8fdff5c7b4a7d3 (https://github.com/rdkcmf/westeros)?  Your log output looks a bit different from mine.   For me, WPE renders its content occupying the entire 1080p screen:

westeros --module /usr/lib/libresolutionmodule.so.0.0.0 --renderer /usr/lib/libwesteros_render_dispmanx.so.0 --enableCursor --framerate 60 --display wayland-0
found /dev/input/event1
opened device /dev/input/event1 : fd 4
found /dev/input/event0
opened device /dev/input/event0 : fd 5
Westeros Info: waiting for compositor wayland-0 to start...
Westeros Debug: calling wl_display_create
Westeros Debug: wl_display=0x75900468
Westeros Info: removing temp file: /tmp/westeros-1903-0
Westeros Info: removing temp file: /tmp/westeros-1903-0.lock
Westeros Info: removing temp file: /tmp/westeros-1882-X1rbi7
Westeros Debug: wl_display=0x75900468 displayName=wayland-0
wstRenderDMXCreate: eglInitiialize: major: 1 minor: 4
wstRendererDMXCreate: wayland-egl support expected, and is advertised by eglQueryString(eglDisplay,EGL_EXTENSIONS): proceeding to use
wstRendererDMXCreate: eglBindWaylandDisplayWL 0x76f05815
wstRendererDMXCreate: eglUnbindWaylandDisplayWL 0x76f05895
wstRendererDMXCreate: eglQueryWaylandBufferWL 0x76f058ed
wstRendererDMXCreate: calling eglBindWaylandDisplayWL with eglDisplay 0x1 and wayland display 0x75900468
wstRendererDMXCreate: have wayland-egl: 1
WstRendererCreate: module (/usr/lib/libwesteros_render_dispmanx.so.0) loaded and intialized
westeros-sb: WstSBInit: enter: display 0x75900468
westeros-sb: WstSBInit: exit: display 0x75900468 sb 0x75902d28
westeros-simpleshell: WstSimpleShellInit: enter: display 0x75900468
westeros-simpleshell: WstSimpleShellInit: exit: display 0x75900468 shell 0x75902d68
Westeros Debug: calling moduleInit for module (/usr/lib/libresolutionmodule.so.0.0.0)...
module: display 1920x1080
Westeros Debug: WstCompositorResolutionChangeBegin
Westeros Debug: WstCompositorResolutionChangeBegin: exit
Westeros Debug: WstCompositorResolutionChangeEnd: (1920 x 1080)
Westeros Debug: WstCompositorResolutionChangeEnd: exit
Westeros Debug: done calling moduleInit for module (/usr/lib/libresolutionmodule.so.0.0.0) result 1
Westeros Debug: calling wl_display_run for display: wayland-0
Westeros Info: compositor wayland-0 is started
Westeros Debug: default cursor client spawned: pid 2422
Westeros Debug: wstCompositorBind: client 0x75902e10 data 0xa98040 version 1 id 5
Westeros Debug: wstUpdateClientInfo: create clientInfo for client 0x75902e10
Westeros Info: display wayland-0 client pid 2422 connected
Westeros Debug: wstSeatBind: client 0x75902e10 data 0x75900628 version 4 id 6
seat 0xaae988 caps: 3
  seat has pointer
  pointer 0xa9cd00
Westeros Debug: wstGetSurfaceInfo: create surfaceInfo for resource 0x75907228
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x75902e10
Westeros Debug: wstICompositorCreateSurface: client 0x75902e10 resource 0x75906f18 id 3 : surface resource 0x75907228
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x75902e10
Westeros Debug: wstKeyboardCheckFocus: no key focus yet
Westeros Debug: wstKeyboardCheckFocus: no key focus yet: client 0x75902e10 has listeners 0
broadcast for surfaceId 1 name noname
Westeros Debug: wstCompositorBind: client 0x75907008 data 0xa98040 version 1 id 5
Westeros Debug: wstUpdateClientInfo: create clientInfo for client 0x75907008
Westeros Info: display wayland-0 client pid 2463 connected
Westeros Debug: wstOutputBind: client 0x75907008 data 0x759005a0 version 2 id 7
Westeros Debug: wstSeatBind: client 0x75907008 data 0x75900628 version 4 id 9
westeros-sb: wstSBBind: enter: client 0x75907008 data 0x75902d28 version 1 id 11
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x75907008
Westeros Debug: wstUpdateClientInfo: client 0x75907008 update resource 0x7590b688
westeros-sb: wstSBBind: exit: client 0x75907008 id 11
westeros-simpleshell: wstSimpleShellBind: enter: client 0x75907008 data 0x75902d68 version 1 id 12
Westeros Debug: wstISeatGetKeyboard: move focus to client 0x75907008 based on resource 0x7590b6f8
Westeros Debug: wstCompositorBind: client 0x7590b4c8 data 0xa98040 version 1 id 4
Westeros Debug: wstUpdateClientInfo: create clientInfo for client 0x7590b4c8
Westeros Info: display wayland-0 client pid 2470 connected
Westeros Debug: wstGetSurfaceInfo: create surfaceInfo for resource 0x7590fb58
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x7590b4c8
Westeros Debug: wstICompositorCreateSurface: client 0x7590b4c8 resource 0x7590f8b8 id 7 : surface resource 0x7590fb58
Westeros Debug: wstUpdateClientInfo: found clientInfo for client 0x7590b4c8
Westeros Debug: wstKeyboardCheckFocus: no key focus yet
Westeros Debug: wstKeyboardCheckFocus: no key focus yet: client 0x7590b4c8 has listeners 0
broadcast for surfaceId 2 name noname

It could also be a difference in WPE versions.  During startup, WPE will receive a wayland mode event giving the 1920x1080 size and it should set its window size in response to this.

fellowgeek, 2018-01-06:
@jeffwannamaker I am on Rocko branch of meta-wpe, probably that's why, thanks for the help, I will try bumping the westeros to the one you mentioned.

Thanks

fellowgeek, 2018-01-07:
all works great now on videos but sites with javascript /css effects like https://realtimebitcoin.info/ are jittery, I am testing on RPI3, is this normal for the device?

hhromic, 2018-02-02:
hi @jeffwannamaker , thanks for this module!  I was looking exactly how to do this.
@wouterlucas , any chance we can add this to the `meta-wpe` layer so when compiling for `_rpi` it gets installed in the image ?
also, similarly to @jeffwannamaker , I also believe the `--enableCursor` option shouldn't be enforced in the `RENDERER` variable, but instead provided as an option if necessary.

wouterlucas, 2018-02-05:
@hhromic I believe that should be part of Westeros itself. Sounds more fitting then adding another recipe for a module within a package. Or we could add it to files and manually copy it in.

hhromic, 2018-02-05:
hi @wouterlucas ,  indeed I got thinking about this during the weekend and I came to the conclusion that the best shot would be to create a recipe that compiles and install this module, from the source code by @jeffwannamaker .
I was thinking also on submitting it into Westeros itself, but then Westeros is just a low-level compositor and this functionality is specific to the WPE use-case.
Also, more correct would be that the Westeros RPI-based renderer is smart enough to get the current resolution instead of assuming 720p.
I still don't know how this is working currently, but I can give it a try and engineer a more general solution that doesn't affect other boards.
When I have a clearer idea, I will post either here or in the westeros repository with ideas to discuss (and hopefully a PR too).
Thanks!