WebPlatformForEmbedded/meta-wpe #174: Error compiling fontconfig

SteffenDE, 2018-01-04:
I just tried to build an image for the rpi (as described in the wiki), but the build failed when compiling fontconfig:
```
ubuntu@ip:~/poky/meta/recipes-graphics/fontconfig$ cat /home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/temp/log.do_compile.9890
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Executing shell function do_compile
NOTE: make -j 16
make  all-recursive
make[1]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build'
Making all in fontconfig
make[2]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fontconfig'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fontconfig'
Making all in fc-blanks
make[2]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-blanks'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-blanks'
Making all in fc-case
make[2]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-case'
make  all-am
make[3]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-case'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-case'
make[2]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-case'
Making all in fc-lang
make[2]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-lang'
make  all-am
make[3]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-lang'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-lang'
make[2]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-lang'
Making all in fc-glyphname
make[2]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-glyphname'
make  all-am
make[3]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-glyphname'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-glyphname'
make[2]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/fc-glyphname'
Making all in src
make[2]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/src'
make  all-am
make[3]: Entering directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/src'
../arm-poky-linux-gnueabi-libtool  --tag=CC   --mode=compile arm-poky-linux-gnueabi-gcc  -march=armv7ve -marm -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 --sysroot=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../fontconfig-2.12.6/src -I..  -I../../fontconfig-2.12.6 -I../../fontconfig-2.12.6/src -I/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot/usr/include/freetype2    -Wall -Wpointer-arith -Wstrict-prototypes 	-Wmissing-prototypes -Wmissing-declarations 	-Wnested-externs -fno-strict-aliasing -DFC_CACHEDIR='"/var/cache/fontconfig"' -DFONTCONFIG_PATH='"/etc/fonts"'   -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0=/usr/src/debug/fontconfig/2.12.6-r0 -fdebug-prefix-map=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot=  -pthread -c -o fcfreetype.lo ../../fontconfig-2.12.6/src/fcfreetype.c
arm-poky-linux-gnueabi-libtool: compile:  arm-poky-linux-gnueabi-gcc -march=armv7ve -marm -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 --sysroot=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../fontconfig-2.12.6/src -I.. -I../../fontconfig-2.12.6 -I../../fontconfig-2.12.6/src -I/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot/usr/include/freetype2 -Wall -Wpointer-arith -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wnested-externs -fno-strict-aliasing -DFC_CACHEDIR=\"/var/cache/fontconfig\" -DFONTCONFIG_PATH=\"/etc/fonts\" -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0=/usr/src/debug/fontconfig/2.12.6-r0 -fdebug-prefix-map=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/recipe-sysroot= -pthread -c ../../fontconfig-2.12.6/src/fcfreetype.c  -fPIC -DPIC -o .libs/fcfreetype.o
../../fontconfig-2.12.6/src/fcfreetype.c: In function 'IA__FcFreeTypeQueryFace':
../../fontconfig-2.12.6/src/fcfreetype.c:1273:47: error: 'FT_SFNT_OS2' undeclared (first use in this function); did you mean 'TT_OS2'?
     os2 = (TT_OS2 *) FT_Get_Sfnt_Table (face, FT_SFNT_OS2);
                                               ^~~~~~~~~~~
                                               TT_OS2
../../fontconfig-2.12.6/src/fcfreetype.c:1273:47: note: each undeclared identifier is reported only once for each function it appears in
../../fontconfig-2.12.6/src/fcfreetype.c:1208:18: warning: unused variable 'r' [-Wunused-variable]
     FcRange     *r = NULL;
                  ^
Makefile:637: recipe for target 'fcfreetype.lo' failed
make[3]: *** [fcfreetype.lo] Error 1
make[3]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/src'
Makefile:498: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build/src'
Makefile:576: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/build'
Makefile:460: recipe for target 'all' failed
make: *** [all] Error 2
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/ubuntu/rpi-ml-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/fontconfig/2.12.6-r0/temp/log.do_compile.9890)
```

hhromic, 2018-01-04:
hi @SteffenDE ,
Check my post [here](https://github.com/WebPlatformForEmbedded/meta-wpe/issues/164#issuecomment-354562598).
I solved this issue by forcing a preferred version.

Haseenamol, 2018-01-04:
For which package you have set preferred version fontconfig or freetype ? and which version you have configured ?

wouterlucas, 2018-01-04:
@Haseenamol you can set:
`PREFERRED_VERSION_freetype = "2.8.%"`

I'll try to find a higher freetype that still works with the old NRDP and is compatible with latest master bumps on poky/oe.

Haseenamol, 2018-01-04:
Thank you Wouter

SteffenDE, 2018-01-04:
Thanks @hhromic, I'll try to build it again using your configuration!
Off topic: As I've got absolutely no experience at all using Yocto, does anyone know if I can set
e.g. `WEBKITBROWSER_XHRCACHE ?= "true"` (or the UserAgent) from the local.conf?

hhromic, 2018-01-04:
you are welcome @SteffenDE .
I've been trying for a while to get to a stable, easy to follow build of WPE with Yocto for newbies like us.

The latest configuration I'm trying out, that uses the `rocko` branch (which should be more stable than `master`) is this:

**Repositories Cloning**

    git clone -b rocko git://git.yoctoproject.org/poky
    git clone -b master git://git.yoctoproject.org/meta-raspberrypi
    git clone -b rocko git://git.openembedded.org/meta-openembedded
    git clone -b rocko git://github.com/webplatformforembedded/meta-wpe

**Image Building**

    source poky/oe-init-build-env
    bitbake-layers add-layer ../meta-openembedded/meta-oe
    bitbake-layers add-layer ../meta-openembedded/meta-python
    bitbake-layers add-layer ../meta-openembedded/meta-multimedia
    bitbake-layers add-layer ../meta-openembedded/meta-networking
    bitbake-layers add-layer ../meta-raspberrypi
    bitbake-layers add-layer ../meta-wpe
    cat <<'__EOF__' >> conf/local.conf
        
    # configuration
    MACHINE = "raspberrypi0-wifi"
    DISTRO_FEATURES_remove = "x11"
    DISTRO_FEATURES_append = "opengl"
    __EOF__
    bitbake wpe-westeros-image

You can change `MACHINE` to `raspberrypi3` if you have a RPI3. Please note that for `meta-raspberrypi`, I'm using `master` instead due to a `mesa-gl` compilation error for `armv6` (e.g. RPI0/1).

I'll let you know more while I'm learning more about this.

**Warning**: I haven't tried the built images into a real RPI yet.. will do soon and report back too.

SteffenDE, 2018-01-04:
Very nice to hear @hhromic! My goal is to get a stable basis for building a digital signage application using the rpi. Yesterday I tried all day getting Chromium to run (which it does actually very good), but the problem is: either you get a perfectly smooth ui (css animations, webgl) but terrible video performance (using the new OpenGL driver vc4-kms-v3d) OR sluggish animations but (most of the time) very smooth 1080p playback :( ...

WPE seems very promising although I still get some [errors](https://github.com/resin-io-projects/resin-wpe/issues/9), some css animations are not smooth yet (not as good as in Chromium) and videos in iframes seem to be not very performant (currently using a build created via [buildroot](https://github.com/WebPlatformForEmbedded/buildroot)). I'll test `rocko` later to see if that makes any difference, but it takes quite some time to compile even on a 12 core server.
I hope to get these issues fixed and will report back if I get it finally working.

hhromic, 2018-01-04:
very nice to hear too @SteffenDE!
Indeed my goal is very similar to yours. I'm trying to build a digital signage application (html5/css calendar and clock in my case) using an rpi. In my case, the challenge is pushed further because I must use a **Raspberry Pi Zero W** instead of the more beefy rpi3. For this reason, Chromium for me is not an option :(.
I tried Screenly OSE and the browser there is really slow (due to framebuffer backend) and limited.
So I found out about WPE and definitively wanted to give it a try like you. This mostly because of the updated engine and opengl backend.
I started trying to compile images using buildroot but was a total failure (again, for the **raspberrypi0w**). So I decided to jump into the Yocto wagon.
Right now I don't have access to the rpi hardware to try the images, will have from next week, and finally I can test the built images. All this time for me was first to get a successful image built.
I'm very interesting on your findings too, maybe we can exchange tests/knowledge and update the WPE wiki with our findings for others to not be so lost.
Cheers from Ireland!
Hugo.

wouterlucas, 2018-01-04:
I'm not sure if you can write to the wiki, if not let me know and I'll see how to add that (guess I'd have to add you as a contributor Hugo). I just pushed some changes to master to make the westeros + wpelauncher combo a bit more straight forward to use. Also about to commit changes to move to a very recent webkit, I'll push those on master only so if you want to go back to a little older webkit you can pick it off of rocko/pyro for the time being (until we know master + new webkit is stable enough).

Plus there are some bigger changes coming, hopefully I can announce them soon which should make things even easier to run browsers on devices like the rpi :)

SteffenDE, 2018-01-04:
Very glad to hear about these "bigger changes". Currently some things clearly seem to be overly complicated. I'll try the new master version as soon as you update Webkit, but I don't really believe that [my biggest obstacle](https://github.com/WebPlatformForEmbedded/WPEWebKit/issues/431) is fixed there. Maybe you know something about that? I don't know if I have to configure WPE in any way, because there seems to be no real documentation at all and I'm very confused about all these configuration options in [WPEFramework](https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-wpe/wpeframework/wpeframework-plugins_git.bb) (what's that? I didn't even find a git repo for it...), e.g. `WEBKITBROWSER_MEMORYPRESSURE ?= "databaseprocess:50m,networkprocess:100m,webprocess:300m,rpcprocess:50m"` (<- **DATABASE**). It also seems like there can be an integrated webserver?
I believe that I don't really understand all this WPE stuff 😅.

Possibly other issues (should I open a separate one?):
When currently starting from master (after commit ea19b64e753faefdca17ccb968f7eb5cbcb4f929) I still get `INIT: Id "S0" respawning too fast: disabled for 5 minutes`. When logging in, running WPELauncher doesn't work, but running wpe does. I also noticed that as soon as a video starts, the command line quickly flashes on the screen about `gst_westeros_sink_init`.

wouterlucas, 2018-01-05:
I saw the issue and unfortunately I do not know where to configure the quota's or whether or not its an issue with Yocto or WebKit. I'd start by trying to figure out where WPE WebKit IndexedDB / sqlite will try to write the DB file (and if that path is available) and there should be some quota settings somewhere. I'd have to ask around or start digging and I'm currently on the hook for some other things first. So will take me a while.

The WPE Framework is something I'm hoping to share soon, which will make it a lot more convenient to run. That memory pressure line is a linux system memory limit that WPE will try to adhere to, meaning the storage process can take up to 50m of linux system memory. Which should be more then sufficient and I dont think it influences your quota error.

> When logging in, running WPELauncher doesn't work, but running wpe does
The wpe script now sets a bunch of stuff to make sure WPELauncher starts correctly against westeros. I tried it this afternoon and launching normal sites seemed fine for me. I didnt try video to behonest, what where you trying to run? YouTube? Running MSE video on rocko/master can be risky, as WPE WebKit currently more stable on Gstreamer 1.10, rocko and master are using 1.12 already and that may definitely throw a few unknowns.

I know its a lot of moving parts, but we're here to help :)

SteffenDE, 2018-01-05:
I'll try to investigate the quota error, but sadly I know absolutely nothing about Webkit (Node/JS + Python dev).
WPE Frameworks sounds promising!

> I tried it this afternoon and launching normal sites seemed fine for me.

I built the current master (`repo sync` and then `bitbake wpe-westeros-image`), but still get the following error when trying to run `WPELauncher`:
[image]

> I didnt try video to behonest, what where you trying to run? YouTube?

Yes, but it's the same with every HTML5 video. I also noticed that it's flickering when YouTube changes the quality (e.g. `gst_westeros_sink_soc_render: video frame size (1920x1080)`).

wouterlucas, 2018-01-05:
Dont run WPELauncher directly, rely on the wpe script to setup te exports (or do them manually prior to launching WPE Launcher).

SteffenDE, 2018-01-05:
Alright, I just didn't know that this is expected to be this way, because in my buildroot build running `WPELauncher` works and I didn't realize that `wpe` is just a wrapper around `WPELauncher` 👍🏻.

wouterlucas, 2018-01-05:
Yeah WPELauncher on BR works fine since the default RPI image on BR doesn't use Westeros/Wayland for its compositing. But renders directly on top of EGL RPI APIs. Making a Yocto build with Wayland turned off will give you the same behavior :)

If you want to try you can turn off Wayland/Westeros by adding the following to your conf/auto.conf or conf/local.conf:
```
WPE_BACKEND = "rpi"
WPE_PLATFORM = "rpi"
DISTRO_FEATURES_remove = "wayland"
```

If you want to be 100% sure you can BBMASK westeros by doing:
```
BBMASK += "westeros"
```

wouterlucas, 2018-01-05:
By the way latest commit on meta-wpe master should fix the dependencies with master poky. I've dropped all backports for now and we'll deal with it later.

SteffenDE, 2018-01-05:
I'll recompile and test directly rendering on the rpi later (Yocto building is much faster than using buildroot), but is there any benefit of doing this instead of using westeros/wayland? I didn't notice any performance difference (but I also could't change the resolution to 1080p easily using westeros).

wouterlucas, 2018-01-05:
If you want to change the resolution of Westeros check this thread: https://github.com/WebPlatformForEmbedded/meta-wpe/issues/172#issuecomment-354137423 

Westeros definitely has its benefits as it implements Wayland. That gives you a headless super thin window management layer. Its best used when you'd like to use multiple graphical clients and manage them correctly (set focus, w/h and position). If you're only interested in running WebKit alone then running it directly on top of EGL is fine as well.

SteffenDE, 2018-01-05:
I just tried turning off westeros, but then `bitbake wpe-westeros-image` doesn't work anymore since "`Nothing PROVIDES 'wpe-westeros-image'`".
```
steffen@sdrack:/ssd_data/compile/rpi-yocto/build$ bitbake wpe-westeros-image
WARNING: /ssd_data/compile/rpi-yocto/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb: Variable key RDEPENDS_${KERNEL_PACKAGE_NAME}-base (${KERNEL_PACKAGE_NAME}-image) replaces original key RDEPENDS_kernel-base ().
...
Parsing recipes: 100%
Parsing of 2070 .bb files complete (0 cached, 2070 parsed). 2881 targets, 296 skipped, 11 masked, 0 errors.
WARNING: No recipes available for:
  /ssd_data/compile/rpi-yocto/meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.12.3.bbappend
  /ssd_data/compile/rpi-yocto/meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.12.3.bbappend
  /ssd_data/compile/rpi-yocto/meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-good_1.12.3.bbappend
  /ssd_data/compile/rpi-yocto/meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly_1.12.3.bbappend
  /ssd_data/compile/rpi-yocto/meta-wpe/recipes-multimedia/gstreamer/gstreamer1.0_1.12.3.bbappend
Removing 1 recipes from the x86_64 sysroot: 100% |#######################################################################################################| Time: 0:00:00
Removing 1 recipes from the allarch sysroot: 100% |######################################################################################################| Time: 0:00:00
Removing 7 recipes from the cortexa7hf-neon-vfpv4 sysroot: 100% |########################################################################################| Time: 0:00:08
Removing 9 recipes from the raspberrypi3 sysroot: 100% |#################################################################################################| Time: 0:00:02
ERROR: Nothing PROVIDES 'wpe-westeros-image'. Close matches:
  wpe-rpi-image
  wpe-initramfs-image
```

I tried `bitbake wpe-rpi-image` , but that fails because I don't have access to:
libprovision, wpeframework, playready, netflix (is this a drm plugin or for what is that?), wpeframework-dialserver, wpeframework-netflix, wpeframework-plugins, wpeframework-ui.

Seems logical as these are non-public repos, aren't they?

conf/local.conf
```
...
MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = "x11 wayland"
DISTRO_FEATURES_append = " opengl"

BBMASK += "westeros"

WPE_BACKEND = "rpi"
WPE_PLATFORM = "rpi"

#PREFERRED_VERSION_freetype = "2.8.%" // do I still need this?
#PREFERRED_VERSION_harfbuzz = "1.7.%"` // do I still need this?
```

> Its best used when you'd like to use multiple graphical clients and manage them correctly

Alright that makes sense, thank you!

wouterlucas, 2018-01-05:
Ah yes of course, I never built the open stuff with directly RPI. My apologies.

You can remove the wpeframework package group for the time being.
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-core/images/wpe-rpi-image.bb#L21

And try again, I believe it should continue just fine. I guess is should add a wpe-rpi-oss image or something like that.

SteffenDE, 2018-01-05:
Looks ok, currently compiling. If the issue with fontconfig is resolved we could close this issue now.
I also updated the wiki page of the rpi, because the order of adding the bb-layers was wrong and led  to dependency errors.

SteffenDE, 2018-01-06:
Hmm something's wrong. I always compare the builds agains my buildroot build (which succeeded only once) and the wpe-rpi-image performs much worse than the buildroot image when it comes to video (e.g. YouTube) in both 1080p and 720p. What's striking is that, while using the same config.txt, the yocto image reports a clock speed of 600MHz while the buildroot image reports 1350MHz (/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq):
```
# Please note that this is only a sample, we recommend you to change it to fit
# your needs.
# You should override this file using a post-build script.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

kernel=zImage

# To use an external initramfs file
#initramfs rootfs.cpio.gz followkernel

# Disable overscan assuming the display supports displaying the full resolution
# If the text shown on the screen disappears off the edge, comment this out
disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=128
gpu_mem_512=196
gpu_mem_1024=256

# Enable mmc by default
dtoverlay=mmc

# Fixes rpi3 ttyAMA0 serial console
dtoverlay=pi3-miniuart-bt

# Force 720p
#hdmi_group=1
#hdmi_mode=4

# Overclock
arm_freq=1350
gpu_freq=500
sdram_freq=500
over_voltage=5
avoid_warnings=1
```
The SoC temp is also very different while playing the video (1080p):
buildroot about 75°C
yocto about 55°C
It seems like the yocto image ignore the overclocking configuration.

Also: yocto image size: ~490MB whereas buildroot ~137MB.
Something's odd here...

About the image size: It seems like removing westeros didn't make any difference in size (westeros ~494MB, rpi ~490MB). Do I have to completely delete the old build before calling bitbake? When building the rpi image 3677 tasks didn't need to rerun according to yocto, so maybe that's the issue here. Sorry for always bothering you here.

wouterlucas, 2018-01-06:
Yes, well there are a couple of big differences. Indeed we overclock the RPI on Buildroot and we should be able to do the same here. I'll add that to list of things to do for meta-wpe :)

But the main difference is we're building with the poky reference distro, which may include a lot more then buildroot (I've never compared it), but I'd like to try a distroless build with oe-core at some point. Buildroot is very confined to the bare minimum you'd require to run WPE. I'm pretty certain the poky reference distro from Yocto includes more packages and stuff that we may not even need at all. 

Other differences could be WPE WebKit itself, make sure you have the latest commit that includes the switch to WPE WebKit Master instead of stable. WPE WebKit master has quite some performance tweaks over stable. 

Also BR runs with Gtreamer 1.10, which is a lot better supported then Poky master Gstreamer 1.12. That could make up for some differences in playback as well. If you want a more 1 on 1 comparison I think the pyro release of Yocto is closer to what the package version on BR are.

SteffenDE, 2018-01-06:
> make sure you have the latest commit that includes the switch to WPE WebKit Master instead of stable

Yes, I built it using the current master branch. I'll test using pyro tomorrow.
Concerning all this I think I'll try using buildroot again (which failed the last time I tried to compile). When shipping images it's better not to include stuff that isn't needed, but I have to say that Yocto's progress view is way superior to the verbosity of buildroot. With Yocto I always know if there's any progress while buildroot just outputs everything without telling you if it will take days or hours to complete.

I also like this repo more because there is an issue tracker 😄
Where to report errors using buildroot?

By the way do you also work on WPEWebKit itself? I just noticed that many issues aren't getting much attention over there and I'm curios to know why.

closed 2018-05-25 by wouterlucas