WebPlatformForEmbedded/meta-wpe #180: EGL hard dependency

thepian, 2018-01-24:
Did Master change requirements ?

```
| checking for cairo's PNG functions feature...
| checking for png... yes
| checking whether cairo's PNG functions feature could be enabled... yes
| checking for cairo's EGL functions feature...
| checking whether cairo's EGL functions feature could be enabled... no (not required by any backend)
| configure: error:  EGL functions feature could not be enabled
| NOTE: The following config.log files may provide further information.
| NOTE: /home/pokyuser/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/cairo/1.14.10-r0/build/config.log
| ERROR: configure failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/pokyuser/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/cairo/1.14.10-r0/temp/log.do_configure.18229)
ERROR: Task (/baker/poky/meta/recipes-graphics/cairo/cairo_1.14.10.bb:do_configure) failed with exit code '1
```

kraj, 2018-01-24:
you might need opengl in your DISTRO_FEATURES. Can you paste the current DISTRO_FEATUERES list

wouterlucas, 2018-01-24:
Please make sure you have the latest master poky/meta-openembedded as well. I've seen this before if you mix & match older poky/meta-openembedded with latest meta-wpe master or if you mis certain changes upstream. Cairo misses a dependency and does not enable the EGL backend properly but that should be resolved in latest master of the layers.

If it still fails, send me the #'es that you are using for the various layers (meta-wpe/poky/meta-rasp/meta-openembedded). It has been building fine for me so far, but maybe I can reproduce it with the exact same versions.

marcw, 2018-02-28:
I am also affected by this. I followed [these instructions](https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/Raspberry-PI) yesterday. All the repositories were downloaded yesterday (2018/02/27).

Here is my conf/local.conf

```
MACHINE = "raspberrypi3"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"
DISTRO_FEATURES_append = " opengl"
```

kraj, 2018-02-28:
in cairo recipe can you add
```
export png_REQUIRES="libpng"
```

wouterlucas, 2018-02-28:
And make sure you have:
```
DISTRO_FEATURES_remove = "x11"
```

Alongside your append = " opengl". Else it might still select X11 over EGL.

marcw, 2018-03-01:
@wouterlucas Thanks... I missed the line in the wiki. 
@kraj I am not sure what you are talking about. What is the cairo recipe and where should that be located?

closed 2018-05-25 by wouterlucas