WebPlatformForEmbedded/meta-wpe #185: Other platforms then rpi?

woutervanh, 2018-01-31:
This layer looks very dependent of everything RPI related?
I wanted to try it on another embedded platform (imx6), but is completely incompatible with this layer... Is that correct?

wouterlucas, 2018-01-31:
Hi @woutervanh not at all, we just use the RPI a lot for reference and testing so you'll see some tweaks that we apply. You can use imx6, or any other device with a bsp layer of your choice. You may get some warnings on dangling bbappends like: 
```
WARNING: No recipes available for:
  /home/wouter/workspace/yocto/meta-wpe/recipes-bsp/bootfiles/rpi-config_git.bbappend
  /home/wouter/workspace/yocto/meta-wpe/recipes-kernel/linux/linux-raspberrypi_%.bbappend
```

But you can safely ignore that. I'm building for Broadcom STB based devices without issues and 96boards (HiKey) with meta-wpe.

Let us know if you run into any issues and we'll gladly help.

woutervanh, 2018-02-02:
ERROR: Nothing PROVIDES 'virtual/egl-native' (but virtual:native:/opt/yocto/fsl-community-bsp_morty/sources/poky/meta/recipes-graphics/cairo/cairo_1.14.6.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/make-native
  virtual/db-native
  virtual/libx11-native

would you have any idea?
I added my machine to the COMPATIBLE_MACHINES of westeros-drm...

wouterlucas, 2018-02-02:
You'll need to have a driver specific to your board that provides opengl capabilities. Without opengl WPE will not work.

Adding to westeros-drm is not sufficient, you'd need a westeros capable render that is able to render to an EGL surface on your target device. 

Does the imx6 support EGL? If so is the driver correctly setup to provide virtual/egl? Try to see if you can run a OpenGL / EGL sample app to verify if that works before diving into westeros/wpe first.

woutervanh, 2018-02-05:
Yes, it for sure has EGL capabilities. Provided by the Vivante core. It also has egl headers for this.
But now I'm stuck cause I updated all my meta-layers to latest morty state (I was a bit behind) and now it fails compiling gstreamer1.0-plugins-bad. I'll get back to you

markov00, 2018-03-29:
@woutervanh Any news about how is going the compilation of WPE on imx6?
We are also building one but I'm just started learning Yocto and everything around.

wouterlucas, 2018-05-25:
I see that imx.6 support has been added upstream, once it lands in our downstream repo I'll copy over the required flags for imx.6 as well. Closing this issue for now.

closed 2018-05-25 by wouterlucas