WebPlatformForEmbedded/meta-wpe #191: QA Warning: TEXTREL in libWPEWebKit.so for armv6

hhromic, 2018-02-19:
Hi all,
I've been receiving the following warning when compiling for `armv6`, e.g. RPI0:

    WARNING: wpewebkit-20170728+gitAUTOINC+804a923c1d-r0 do_package_qa: QA Issue: ELF binary '/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/20170728+gitAUTOINC+804a923c1d-r0/packages-split/wpewebkit/usr/lib/libWPEWebKit.so.0.0.20170728' has relocations in .text [textrel]

This does not happen when compiling for `armv7`, e.g. RPI3.

According to the documentation in [1], I tried forcing the `-fpic` compiler flags in the recipe in the following manner:

    CFLAGS_append = " -fPIC -fpic"

However this didn't help. The resulting image works fine, however this problem is supposed to bring performance penalties when starting up WPE and waste of active memory aswell [2].

I checked and the CMakes of WPE seems to enable position-independent code when building. Perhaps there is an obscure condition that is not triggering for `armv6`.

Any ideas?

[1] http://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#qa-issue-textrel
[2] https://blog.flameeyes.eu/2016/01/textrels-text-relocations-and-their-impact-on-hardening-techniques/

kraj, 2018-02-19:
usually bad asm code is the underlying cause for such errors. you might have to hunt for it

hhromic, 2018-02-20:
Thanks for your answer, @kraj !
I tried a bit to hunt it down now to see how hard would it be, and I think I found the offending asm code.

However I'm not experienced at all and I might break it further. The problem seems to be located in `Source/JavaScriptCore/llint/LowLevelInterpreter.asm`, symbol `llint_entry`.
Because is not related to the `meta-wpe` layer itself, maybe this is more appropiate to be further discussed in the WPEWebKit repository instead?

hhromic, 2018-02-20:
Closing as I migrated the issue to mainstream WPEWebKit. Thanks!

closed 2018-02-20 by hhromic