WebPlatformForEmbedded/meta-wpe #193: WPE is not building in the master branch for EGLFS

hhromic, 2018-03-17:
Hello!

In case you haven't noticed, I wanted to report that `meta-wpe` is not building in `master` since a while now when using the **EGLFS backend**. It builds fine when building for Westeros.

Below is my build configuration and the results when building for EGLFS. It was building fine from `master` before, so I assume there was a small mistake somewhere recently (I suspect it has to do with the `rpi` packageconfig being not found).

Let me know how can I help debugging too.

* The `local.conf` File

```
MACHINE = "raspberrypi0-wifi"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
WPE_BACKEND = "rpi"
WPE_PLATFORM = "rpi"
DISTRO_FEATURES_remove = "wayland"
yocto@srvgal93:~/wpe$
```

* Build Configuration

```
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-9"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4+snapshot-20180316"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "master:674cdda1b973f87f5adb3f98c24f3b088e21ae0f"
meta-oe
meta-python
meta-multimedia
meta-networking      = "master:790ddac5a9f2d4a40012551eb3b7a0743c2975e6"
meta-raspberrypi     = "master:79ea44b997a6eb6e2f0d36b1d583c85fa9e37f36"
meta-wpe             = "master:555d900b9a8b0d04188e6bc83f647c1bd2ea7033"
```

When compiling for **EGLFS**, there is a warning during the build:

```
WARNING: wpewebkit-20170728+gitAUTOINC+8daad8b139-r0 do_configure: QA Issue: wpewebkit: invalid PACKAGECONFIG: rpi [invalid-packageconfig]
```

And then the build fails with:

```
ERROR: wpebackend-rdk-0.1-r0 do_compile: Function failed: do_compile (log file is located at /home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpebackend-rdk/0.1-
r0/temp/log.do_compile.3805)

| In file included from /home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpebackend-rdk/0.1-r0/git/src/bcm-rpi/view-backend.cpp:30:0:
| /home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpebackend-rdk/0.1-r0/git/src/input/Libinput/LibinputServer.h:37:10: fatal error: libinput.h: No such file or
directory
|  #include <libinput.h>
|           ^~~~~~~~~~~~
| compilation terminated.

ERROR: Task (/home/yocto/meta-wpe/recipes-wpe/wpebackend-rdk/wpebackend-rdk_0.1.bb:do_compile) failed with exit code '1'
```

---

In an unrelated issue, when I was testing for **Westeros**, there was the following warning. However the build succeeds anyway.

```
WARNING: gstreamer1.0-omx-1.12.4-r0 do_patch:
Some of the context lines in patches were ignored. This can lead to incorrectly applied patches.
The context lines in the patches can be updated with devtool:

    devtool modify <recipe>
    devtool finish --force-patch-refresh <recipe> <layer_path>

Then the updated patches and the source tree (in devtool's workspace)
should be reviewed to make sure the patches apply in the correct place
and don't introduce duplicate lines (which can, and does happen
when some of the context is ignored). Further information:
http://lists.openembedded.org/pipermail/openembedded-core/2018-March/148675.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=10450
Details:
Applying patch 0001-config-files-path.patch
patching file config/bellagio/gstomx.conf
Hunk #1 succeeded at 1 with fuzz 2.
Hunk #2 succeeded at 9 with fuzz 2.
patching file config/rpi/gstomx.conf
Hunk #8 succeeded at 63 with fuzz 2.
Hunk #10 succeeded at 82 with fuzz 2.
Hunk #11 succeeded at 92 with fuzz 2.

Now at patch 0001-config-files-path.patch
```

wouterlucas, 2018-03-18:
Thanks for reporting, I've seen that input error before with the Westeros build as well and thought we fixed that with the recent bumps. I'll circle back to it and see if I can run into it as well with the eglfs build.

As for the Gstreamer patches, the 1.12 patches need a refresh. Something that we've unfortunately did not have enough time for, we'll start looking more into Gst 1.12 soon.

hhromic, 2018-03-18:
I'm constantly building using eglfs with master (mostly for testing the build itself), so I can debug and report back when you apply fixes to master.
Thanks for replying!

hhromic, 2018-03-19:
Hi again @wouterlucas , I just realised the Gst warning actually belongs to the `meta-raspberrypi` layer (where the patch file actually is) and not `meta-wpe` directly.
Should I move that issue there instead?

wouterlucas, 2018-03-19:
Oh its an omx patch :) That would probably make more sense, sounds like its pretty easy to rebase and fix.

wouterlucas, 2018-03-19:
Ah just ran into the same, its because libinput is turned off by default now. 

You can get around this by doing:
WPE_BACKEND = "rpi libinput"

I'll add it to the wiki.

kraj, 2018-03-19:
thats a workaround but we need to ensure that packageconfig is added when rpi is chosen backend

wouterlucas, 2018-03-19:
Well, I initially thought that too. But there may be scenarios where you do not want libinput on an rpi device (graphics), e.g. when relying on xkb for input. 

The proper solution would be to enable libinput when xkb is not part of the image.

kraj, 2018-03-19:
I would rather make it default and let users of xkbcommon override it if needed.

hhromic, 2018-03-19:
In my humble opinion, libinput should be added as default.
My reasoning is that it seems to be a hard dependency for the `wpe-eglfs-image` recipe, e.g. it doesn't build properly without it. The user should not need to explicitly add libinput for the build to succeed.
I agree with @kraj that users not wanting libinput can override it under their own responsability/understanding of what they are doing.
Thanks for the discussion!

@wouterlucas , regarding the gst-omx patch, yes it just seems to be a simply rebasing. Tomorrow I will try to give it a try and if I succeed will send a PR upstream to `meta-raspberrypi` =).

wouterlucas, 2018-03-20:
Okay, well as long as we realize that we're hardcoding something that's not a hard dependency in the recipe I'm okay.

hhromic, 2018-03-20:
I got thinking that some of the `local.conf` settings that are actually *required* to build certain images, e.g. `wpe-eglfs-image` should be indeed in the image recipe.
For example, currently it doesn't work to try to build the eglfs image without the `WPE_BACKEND="rpi"` variable set.

It is also recommended to set `DISTRO_FEATURES_remove = "wayland"` for eglfs builds.
This one in particular was added recently to the recipe of `wpe-eglfs-image`, which I assume renders unnecessary to add it to the `local.conf` file.
Should this be the same case for the `WPE_BACKEND/PLATFORM` variables?

I'm not familiar with what kind of devices you are supporting, hence my confusions. Is the "eglfs" image recipe more general than just RPI? In other words, eglfs implies RPI-backend or is the other way around, RPI-backend implies eglfs build? Sorry for all the blah-blah guys :)

Perhaps the image recipes should be "smarter" in this aspect?
It is clear that `wpe-westeros-image` should build a Westeros-based image for any device that supports it.
The `wpe-eglfs-image` should then have the same logic, but if we are building for RPI, then should set variables accordingly to accomplish the build, e.g. set the `WPE_BACKEND/PLATFORM` variables.

So in summary, if you want to use the `wpe-westeros-image` recipe, it should add on its own:

    DISTRO_FEATURES_remove = "x11"
    DISTRO_FEATURES_append = " opengl"

to the environment. Likewise, if you want the `wpe-eglfs-image` recipe, it should add:

    WPE_BACKEND = "rpi"  # only if we are building for rpi?
    WPE_PLATFORM = "rpi"  # only if we are building for rpi?
    DISTRO_FEATURES_remove = "wayland"  # already done in the recipe

Those variables are described in the Wiki to be necessary for the respective builds. Then why not just add them in the recipe so they are automatic? Is there any reason to not use them in some cases?

Does all this make any sense to you? :)

closed 2018-05-25 by wouterlucas