WebPlatformForEmbedded/meta-wpe #195: wpewebkit 20170728+gitAUTOINC+c6f5733838-r1 contains a compilation error

mrstinky, 2018-03-27:
referenced in https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-wpe/wpewebkit/wpewebkit_20170728.bb 

wpewebkit/20170728+gitAUTOINC+c6f5733838-r1 fails to build

error is:
/wpe-build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/wpewebkit/20170728+gitAUTOINC+c6f5733838-r1/git/Source/WebKit/WebProcess/WebPage/wpe/AcceleratedSurfaceWPE.cpp:82:99: error: invalid cast from type 'EGLNativeWindowType {aka long unsigned int}' to type 'uint64_t {aka long long unsigned int}'
	      return reinterpret_cast<uint64_t>(wpe_renderer_backend_egl_target_get_native_window(m_backend));

psivasubramanian, 2018-03-27:
> wpe/blob/master/recipes-wpe/wpewebkit/wpewebkit_20170728.bb

@mrstinky this error disappears when changed to static_cast instead of
reinterpret_cast.

mrstinky, 2018-03-27:
Yes, That's how I fixed the error for my own build. :)

Would this normally be fixed by a patching in the layer, or should we update the recipe to point to the fixed version of the original source?

wouterlucas, 2018-05-25:
If this is still a problem a fix should be pushed towards https://github.com/WebPlatformForEmbedded/WPEWebKit

closed 2018-05-25 by wouterlucas