WebPlatformForEmbedded/meta-wpe #196: building for Dragonboard 410c

mrstinky, 2018-03-27:
Hello,

I've been able to get an image built for Dragonboard 410c.

I added meta-qcom to my .repo/manifest.xml
`<project remote="github" name="ndechesne/meta-qcom" path="meta-qcom"/>`

I added meta-qcom to my conf/bblayers.conf file
` /local/mnt/workspace/webkit/96boards-westeros-wpe/meta-qcom \`

The build worked, and I flashed the board.

I can see that there is a westeros daemon started by default, but I can not see or switch to the screen. The alt-Fn keys don't do anything.

If I run westeros-init, it gets to this point and stalls:
```
found /dev/input/event2`
opened device /dev/input/event2 : fd 4
found /dev/input/event1
opened device /dev/input/event1 : fd 5
found /dev/input/event0
opened device /dev/input/event0 : fd 6
Westeros Info: waiting for compositor wayland-2 to start...
Westeros Debug: calling wl_display_create
Westeros Debug: wl_display=0xf5b005b8
Westeros Debug: wl_display=0xf5b005b8 displayName=wayland-2
westeros-gl: wstGLInit: realEGLGetDisplay=(nil)
westeros-gl: wstGLInit: unable to resolve eglGetDisplay
wstRendererGLSetupEGL: eglDisplay=0xf5b00f80
```
<stalls here>

If I run westeros manually, it starts, and outputs:
```
# XDG_RUNTIME_DIR=/tmp westeros --renderer /usr/lib/libwesteros_render_gl.so --display wayland-2
found /dev/input/event2
opened device /dev/input/event2 : fd 4
found /dev/input/event1
opened device /dev/input/event1 : fd 5
found /dev/input/event0
opened device /dev/input/event0 : fd 6
Westeros Info: waiting for compositor wayland-2 to start...
Westeros Debug: calling wl_display_create
Westeros Debug: wl_display=0xf5b005b8
Westeros Debug: wl_display=0xf5b005b8 displayName=wayland-2
westeros-gl: wstGLInit: realEGLGetDisplay=(nil)
westeros-gl: wstGLInit: unable to resolve eglGetDisplay
wstRendererGLSetupEGL: eglDisplay=0xf5b00f80
wstRendererGLSetupEGL: unable to initialize EGL display
eglCreateImageKHR 0xf71517dc
eglDestroyImageKHR 0xf7151a3c
glEGLImageTargetTexture2DOES 0xf6c89770
have wayland-egl: 0
WstRendererCreate: module (/usr/lib/libwesteros_render_gl.so) loaded and intialized
westeros-sb: WstSBInit: enter: display 0xf5b005b8
westeros-sb: WstSBInit: exit: display 0xf5b005b8 sb 0xf5b01858
westeros-simpleshell: WstSimpleShellInit: enter: display 0xf5b005b8
westeros-simpleshell: WstSimpleShellInit: exit: display 0xf5b005b8 shell 0xf5b01898
Westeros Debug: calling wl_display_run for display: wayland-2
Westeros Info: compositor wayland-2 is started
```
But I cannot see any display.

If I type "shutdown now", my monitor sometimes flashes briefly to a graphical screen which says, "Yocto" and shows a progress bar. After about 1 second the screen disappears, and the system never shuts down.

How do I proceed?

psivasubramanian, 2018-03-27:
For DRM platforms such as DB410c, westeros should be launched like below with LD_PRELOAD

$ LD_PRELOAD=/usr/lib/libwesteros_gl.so.0 westeros --renderer /usr/lib/libwesteros_render_gl.so.0 --enableCursor --display westeros-0

mrstinky, 2018-03-27:
I did try that as well, it provides more debug. 

I've attached the debug files for running westeros-init and westeros_test. 

Not shown on the westeros-init debug is this line which occurs right at the end:
`westeros_gl: WstGLSwapBuffers: drmModeAddFB error: -22 errno 22`

What should happen when I run westeros? Should the screen change from the terminal to a graphics screen?

[westeros_test.debug.txt](https://github.com/WebPlatformForEmbedded/meta-wpe/files/1852745/westeros_test.debug.txt)

[westeros-init.debug.txt](https://github.com/WebPlatformForEmbedded/meta-wpe/files/1852746/westeros-init.debug.txt)

psivasubramanian, 2018-03-28:
@mrstinky Yes, it should with big mouse pointer if enable cursor. Did you tried any DRM test application to make sure it's not DRM/EGL issue?

mrstinky, 2018-04-11:
I'm still stuck on this.

The invalid argument was the stride value being passed to drmModeAddFB().

The value being passed was 5120 (1284 x 4), when I think it should have been 6400 (1600 x 4).

The kernel accepted this 6400 stride value, but the kernel still fails to create the framebuffer at a later stage.

`[drm:internal_framebuffer_create] could not create framebuffer`

I've traced this to:

```
	fb = dev->mode_config.funcs->fb_create(dev, file_priv, r);
	if (IS_ERR(fb)) {
		DRM_DEBUG_KMS("could not create framebuffer\n");
		return fb;
	}
```
I think that fb_create calls `msm_framebuffer_create()` which calls `msm_framebuffer_init()`

The final return value is EINVAL, which occurs in two places in `msm_framebuffer_init()`
```	
if (!format) {
		dev_err(dev->dev, "unsupported pixel format: %4.4s\n",
				(char *)&mode_cmd->pixel_format);
		ret = -EINVAL;
		goto fail;
	}
```
I don't see this error, so the second place is:
```
	if (n > ARRAY_SIZE(msm_fb->planes)) {
		ret = -EINVAL;
		goto fail;
	}
```

Any thoughts on what this could be?

I'm currently trying to add some debug to the kernel, but having a lot of trouble doing this in Yocto, and getting the kernel to recompile with my changes.

mrstinky, 2018-04-11:
7680000 = 1600 * 1200 * 4  <-- that's the resolution I'm trying to achieve

4194304 = 1024 * 1024 * 4 ...  what's this number?

mrstinky, 2018-04-30:
Working now.  

Problem was caused by a bug in the Westeros debug printing which indicated that it was choosing the resolution of 1600x1200, but instead it was failing to find any matching resolution, and it was choosing the zeroth item in the list - because that was the default value of the mode match variable. 

So I thought that it was setting this resolution, but it actually wasn't. The required resolution was 720p

There are several locations in several files where the resolution 1280x720 is hardcoded.

I "fixed" it by editing all these files to the resolution of my screen, and then it worked. 

The problem would have been avoided in the first instance if I'd used a television with 1280x720 capabilities. :)

Oh, well.

closed 2018-04-30 by mrstinky

kraj, 2018-05-01:
Can you attach the westeros patch here for future references ?

mrstinky, 2018-05-01:
I didn't fix anything, I just became aware of the limitations and assumptions made by the Westeros compositor.

To change the resolution, I grep'd for "1280" in the files, and replaced with "1600". Same for "720" -> "1200". It's possible I made changes in files that didn't need to be changed.

Fixing the resolution issue in Westeros would require an extensive rewrite in order to pass the resolution around between several files.