WebPlatformForEmbedded/meta-wpe #197: libepoxy patch fails

DWagenk, 2018-03-29:
My yocto build failed, because it couldnt apply this patch: https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-graphics/libepoxy/libepoxy/0001_make_graphic_libs_configureble.patch

Seems that libepoxy source for this file has changed and I couldn't find where the recipies selected wich version to pick.

I got it working in my copy manually changing this patch file and recompiling, but am not to firm in yocto yet, so I won't submit it as a fix

estobbart, 2018-03-29:
@DWagenk Same issue. What were your steps to resolve it?

kraj, 2018-03-30:
libepoxy has been updated in master  see
http://git.openembedded.org/openembedded-core/commit/meta/recipes-graphics/libepoxy?id=b72154eccfbcc178a2c09c7c7d4cd0264d4cf0f6

this patch needed to be forward ported.

PR #198 (2018-03-30): libepoxy: refresh patch, fix naming (fixes #197)
This PR fixes #197
commit 8f2d0db2 (2018-03-30): Merge pull request #198 from hhromic/libepoxy-refresh-patch

libepoxy: refresh patch, fix naming (fixes #197)
closed 2018-03-30 by wouterlucas