WebPlatformForEmbedded/meta-wpe #200: broadcom-refsw: recipe not found in RPI build

hhromic, 2018-04-03:
Hi guys,
I've been receiving this warning since a while for RPI builds:

```
WARNING: No recipes available for:
  /home/yocto/meta-wpe/recipes-bsp/broadcom-refsw/broadcom-refsw_%.bbappend
```

There is no `broadcom-refsw` recipe anywhere in the chain to build for RPI, e.g. `poky`, `meta-openembedded` and `meta-raspberrypi`.

I suspect the recipe is for another machine that I'm not aware of, however the build for RPI is anyway trying to use it. This warning was introduced in commit 3637dae8cd7aed7b5a9e244e772eaf94716d4e67.

Thanks!

hhromic, 2018-04-04:
I just happened to discover that actually there is a design pattern for doing exactly this: **dynamic-layers**.

For example `meta-raspberrypi` uses it: https://github.com/agherzan/meta-raspberrypi/blob/292e00099bae1e4a822a56969221e5fee6b5db8d/conf/layer.conf#L15

In summary, we to just need to do the following:
1. Create a `dynamic-layers` directory in `meta-wpe`.
2. Add that piece of code from `meta-raspberrypi` to `conf/layer.conf` here.
3. Move `recipes-bsp/broadcom-refsw` to `dynamic-layers/<broadcom_layer>/recipes-bsp/broadcom-refsw`.

Then the build automatically will use the bbappends if the `<broadcom_layer>` is actually included in the build, or leave it alone if not (for example for RPI builds).

I believe this is the best way to approach the warning, is future-proof and allows for more layer-specific recipes to be added without interfering with different types of builds. Also this approach supports the goal of `meta-wpe` to be more independent of the underlying BSP.

If you tell me the name for `<broadcom_layer>` that you are using, I can create and a test a PR.

wouterlucas, 2018-04-04:
I still have to try to understand it, thanks for the input and I'll start a new educational journey of yet uncharted parts of Yocto for me :-)

That said the same needs to happen for any raspberry pi specific changes that we have in meta-wpe. We shouldnt be device specific in this repo and that would work across all sorts of device specific changes we maintain.

kraj, 2018-04-04:
yes using dynamic layers is the right approach here. @hhromic if you are upto it cook up a patch., here is a simple example 

https://github.com/cbrake/meta-bec/commit/ed5c7d847a9b41f141efff5d441925ee69979273

hhromic, 2018-04-05:
@wouterlucas : I'm no yocto gurú myself, learning a lot thanks to this project actually!
I agree that device-specific stuff should be minimised, and I belive using dynamic layers will help a lot in this regard, specially on keeping device-specific recipes/bbappends well organised and separated.

@kraj : thanks for the pointer, looks very elegant, more than the snippet I found :)
One question: you mean `collection_name` in `BBFILES_DYNAMIC` should accept wildcards, hence using `brcm*` would match any layer name that starts with `brcm`?
The documentation you linked in your commit (https://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#var-BBFILES_DYNAMIC) doesn't say anything explicit about this.

I don't know the `BBFILE_COLLECTIONS` name you have for the layer that contains `broadcom-refsw`, can you confirm me this please from `conf/layer.conf` there? This way any change I attempt won't break your builds.

I will investigate further and see if I can come with a solid approach to make a PR.
My idea is to put all RPI specific things in its own set of recipes in dynamic layers, and so people should do the same for other board-spefic things.

hhromic, 2018-04-05:
I already worked out a solution that works nicely for easy to identify raspberrypi-specific recipes, using `BBFILES_DYNAMIC` :).
However, looks like using wildcards for the collection name to be matched is not supported:
https://github.com/openembedded/bitbake/blob/256990943075e89cb9aee2bc6488344b6783e07b/lib/bb/cookerdata.py#L388

This is not a big deal, but I will need to know the collection name of your Broadcom BSP layer to be able to produce a PR that doesn't break your build. Looking fwd!

kraj, 2018-04-05:
Just consider  "brcm" thats where main recipes for broadcom-refsw are in.

closed 2018-04-09 by hhromic