WebPlatformForEmbedded/meta-wpe #206: webkit remote inspector wont load

malouf-erfan, 2018-04-24:
Hi,

I am trying to use remote inspector, I set the WEBKIT_INSPECTOR_SERVER variable and I can see that the port is opened on network.

export WEBKIT_INSPECTOR_SERVER=0.0.0.0:9222

I opened http://ipaddress:port in browser but I get no response 
"IP didn’t send any data.
ERR_EMPTY_RESPONSE

I appreciate any help.

wouterlucas, 2018-04-25:
Try setting WEBKIT_LEGACY_INSPECTOR_SERVER for the old web inspector and make sure you have the latest commit that includes the shared-mime-info rdep.

T94T, 2018-05-22:
I'm having the exact same problem. Using WEBKIT_LEGACY_INSPECTOR_SERVER did some improvements. But after selecting a web view there are some errors showing up in the console. No page is displayed.
```
- parseURL is not defined
- invalid regexp group
- parseLocationQueryParameters is not defined
```
I'm currently using your buildroot implementation.

wouterlucas, 2018-05-22:
A recent version? Which browser are you using on your desktop to open the webinspector?

T94T, 2018-05-22:
Thank you for the hint. In Google Chrome (66) and Safari (11.1) it works like expected.
(btw: We are using the most recent master branch)