WebPlatformForEmbedded/meta-wpe #209: Expected output of meta-wpe on RaspberryPi3

Pixelfahnder, 2018-05-28:
Good day everyone,

I followed along the guide at https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/Raspberry-PI and wonder what the expected output of the whole process should be. I see some kind of console window with a big yellow arrow which can be controlled by an attached mouse but not by a connected HID conform touch pad.

Two questions rise out of this:

1) Is that the expected output I should see? I thought WPE will serve a "browser" kind of view directly on wayland, so shouldn't I see something like a webpage that is served fullscreen?

2) Should HID conformant touch work or do I need to include another (YOCTO) package for that?

Best regards.

Pixelfahnder, 2018-05-28:
I already found issue #172 and tried setting the exports. This worked and helped me to get a white screen. How to set this export settings during the build of a RaspberryPi3 image?

wouterlucas, 2018-05-28:
Hi @Pixelfahnder, I'm not sure how to get the touch screen to work. I'd imagine you'd need to enable some kernel flags to get the touch screen enabled. 

Did you go with westeros or the EGLFS image?

Pixelfahnder, 2018-05-29:
Dear @wouterlucas thanks for your answer. I used westeos.

Also do you have hint how I can get the environment variables mentioned in Issue #172 into the build process? I still have to set them manually after the build and also only experience a white screen whatever local file I try to pass as parameter to the WPELauncher.

Pixelfahnder, 2018-06-03:
@wouterlucas is it possible to inject touch events to the platform in the EGLFS version? On Wayland I could send touch events to the Wayland protocol or something like custom keyboard events via third party hardware that should be served to the WPE seamlessly.

wouterlucas, 2018-06-03:
This can be enabled in the config of the launcher. Go into /etc/WPEFramework/plugins/WebKitBrowser.json and set the touch boolean to true. 

Note this only works if the kernel recognised the touch interface. But assuming on your remark you already had that working with Wayland.

Note2: If you’re still on the old WPELauncher you have to comment out an export. Edit the launcher script and you can’t miss it.

Pixelfahnder, 2018-06-03:
Thanks, I will try that in the office tomorrow. Just to make sure I have the right version:Which branch do Ihave to checkout?

wouterlucas, 2018-06-03:
Depends on which version of poky/oe-core you are using. Branches align with the poky releases.

Pixelfahnder, 2018-06-04:
SO I check if I have sumo everywhere.

wouterlucas, 2018-06-04:
Yeah sumo should be fine, though that comes with 2.20 and may have some issues down the road. If you're running into stability issues or if that is very important I'd suggest to use Rocko for now.

Pixelfahnder, 2018-06-04:
I tried it agaiin (even in a new directory) did a sumo checkout and a repo sync but now I'm stuck with GSTREAMER compile errors when following the abovementioned guide for RPI3 with the EGL configuration. Any hints? I'm totally lost now.

wouterlucas, 2018-06-04:
Can you share the error with me?

Pixelfahnder, 2018-06-04:
This is the output I get btw:

```
developer@devNook:~/rrr-yocto/meta-wpe$ git checkout sumo
Zu Branch 'sumo' gewechselt
Ihr Branch ist auf demselben Stand wie 'webplatformforembedded/master'.
developer@devNook:~/rrr-yocto/meta-wpe$ repo sync
Fetching project meta-openembedded
Fetching project meta-wpe
Fetching projects:  50% (2/4)  Fetching project poky
Fetching projects:  75% (3/4)  Fetching project meta-raspberrypi
Fetching projects: 100% (4/4), done.  

developer@devNook:~/rrr-yocto/meta-wpe$ cd ..
developer@devNook:~/rrr-yocto$ cd rpi-ml-build/
developer@devNook:~/rrr-yocto/rpi-ml-build$ cd ..
developer@devNook:~/rrr-yocto$ source poky/oe-init-build-env rpi-ml-build/

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'
developer@devNook:~/rrr-yocto/rpi-ml-build$ bitbake wpe-eglfs-image
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#############################################################################################################################################################################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 2974 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5+snapshot-20180604"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:90a29096f47981e21e86594ca2310b659844c9c1"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "HEAD:bb57bac845f3cd1634862fa9868bc8e294ba74a9"
meta-raspberrypi     = "HEAD:b2da4618b0ac2ad7dbc26387cad4c03796f8a06e"
meta-wpe             = "sumo:d749c9af544f00b6635f531d07a4c166ed895fc0"

Initialising tasks: 100% |########################################################################################################################################################################################################################################################################################################################################################################################| Time: 0:00:01
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: wpewebkit-2.20+gitAUTOINC+498ecbc875-r1 do_compile: Function failed: do_compile (log file is located at /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/temp/log.do_compile.23595)
ERROR: Logfile of failure stored in: /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/temp/log.do_compile.23595
Log data follows:
| DEBUG: Executing shell function do_compile
| ninja: Entering directory `/home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/build'
| [1/1284] Generate bindings (WebCoreBindings)
| [2/1281] cd /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/build/Source/WebKit && /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/recipe-sysroot-native/usr/bin/perl-native/perl /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebKit/Scripts/generate-forwarding-headers.pl --include-path /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebKit --output /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/build/DerivedSources/ForwardingHeaders --platform wpe --platform soup
| [3/1281] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/audio/gstreamer/AudioSourceProviderGStreamer.cpp.o
| [4/1281] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/audio/gstreamer/AudioDestinationGStreamer.cpp.o
| [5/1281] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/audio/gstreamer/FFTFrameGStreamer.cpp.o
| [6/1281] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/audio/gstreamer/AudioFileReaderGStreamer.cpp.o
| [7/1281] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/gstreamer/mse/SourceBufferPrivateGStreamer.cpp.o
[... 285 lines cut ...]
[... cut ...]
| ERROR: Function failed: do_compile (log file is located at /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/temp/log.do_compile.23595)
ERROR: Task (/home/developer/rrr-yocto/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_2.20.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4023 tasks of which 4020 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/developer/rrr-yocto/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_2.20.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```

Pixelfahnder, 2018-06-04:
This is the output when I try to compile the **sumo** branch. Compiling the master gives the following output:

```
developer@devNook:~/rrr-yocto/rpi-ml-build$ cd ..
developer@devNook:~/rrr-yocto$ cd meta-wpe/
developer@devNook:~/rrr-yocto/meta-wpe$ git checkout master
Zu Branch 'master' gewechselt
Ihr Branch ist auf demselben Stand wie 'webplatformforembedded/master'.
developer@devNook:~/rrr-yocto/meta-wpe$ repo sync
Fetching project meta-openembedded
Fetching project meta-wpe
Fetching projects:  50% (2/4)  Fetching project poky
Fetching projects:  75% (3/4)  Fetching project meta-raspberrypi
Fetching projects: 100% (4/4), done.  

developer@devNook:~/rrr-yocto/meta-wpe$ cd ..
developer@devNook:~/rrr-yocto$ source rpi-ml-build/
bash: source: rpi-ml-build/: ist ein Verzeichnis.
developer@devNook:~/rrr-yocto$ bitbake wpe-eglfs-image
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#############################################################################################################################################################################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 2974 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5+snapshot-20180604"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:90a29096f47981e21e86594ca2310b659844c9c1"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "HEAD:bb57bac845f3cd1634862fa9868bc8e294ba74a9"
meta-raspberrypi     = "HEAD:b2da4618b0ac2ad7dbc26387cad4c03796f8a06e"
meta-wpe             = "master:d749c9af544f00b6635f531d07a4c166ed895fc0"

Initialising tasks: 100% |########################################################################################################################################################################################################################################################################################################################################################################################| Time: 0:00:02
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: wpewebkit-2.20+gitAUTOINC+498ecbc875-r1 do_compile: Function failed: do_compile (log file is located at /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/temp/log.do_compile.23881)
ERROR: Logfile of failure stored in: /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/temp/log.do_compile.23881
Log data follows:
| DEBUG: Executing shell function do_compile
| ninja: Entering directory `/home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/build'
| [1/1268] Generate bindings (WebCoreBindings)
| [2/1265] cd /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/build/Source/WebKit && /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/recipe-sysroot-native/usr/bin/perl-native/perl /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebKit/Scripts/generate-forwarding-headers.pl --include-path /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebKit --output /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/build/DerivedSources/ForwardingHeaders --platform wpe --platform soup
| [3/1265] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/image-decoders/cairo/ImageBackingStoreCairo.cpp.o
| In file included from DerivedSources/ForwardingHeaders/wtf/Gigacage.h:127:0,
|                  from DerivedSources/ForwardingHeaders/wtf/CagedPtr.h:29,
|                  from DerivedSources/ForwardingHeaders/JavaScriptCore/ArrayBuffer.h:31,
|                  from /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebCore/platform/SharedBuffer.h:30,
|                  from /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebCore/platform/graphics/ImageBackingStore.h:32,
|                  from /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/git/Source/WebCore/platform/image-decoders/cairo/ImageBack [... cut ...]
| ERROR: Function failed: do_compile (log file is located at /home/developer/rrr-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+498ecbc875-r1/temp/log.do_compile.23881)
ERROR: Task (/home/developer/rrr-yocto/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_2.20.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4023 tasks of which 4022 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/developer/rrr-yocto/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_2.20.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```

wouterlucas, 2018-06-04:
I've seen that error too but only when I use wpe-westeros-image. Can you check what WPE WebKit is configured for? I'm suspecting its configured for Westeros/Wayland.

Pixelfahnder, 2018-06-04:
SO it seems to be something with cairo this time, I believe the gstreamer error comes when retrying the westeros. Not sure about this right now. But this was the output from a couple of minutes ago. I also tried to it with deleting the tmp folder.

Pixelfahnder, 2018-06-04:
@wouterlucas Where to check this? (Asking just to make sure I check the right localtion)

Pixelfahnder, 2018-06-04:
The **local.conf** file has the following settings:

```
...
MACHINE = "raspberrypi3"
...
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
```

wouterlucas, 2018-06-04:
Can you add this to your local.conf:

```
DISTRO_FEATURES_remove = "wayland"
```

That would avoid building wayland and thus configuring WPE for Westeros. With that DISTRO remove you'll take the EGL route with wpebackend-rdk directly onto the dispmanx EGL surface.

The configure output is in `tmp/work/<tuple>/wpewebkit/<version>/temp/log.do_configure`. I'm pretty convinced its because it's still trying to go the Wayland/Westeros route.

Pixelfahnder, 2018-06-04:
OK that seems to work. So I can take care of the original problem. I have booted up the generated SD Card Image with EGL and have entered the WPEFramework command. Nothing really happens. How to show a webpage? Actually I also don't see a mouse cursor like I have before when using wayland.

Pixelfahnder, 2018-06-04:
OK I believe something went wrong during the compilation as I even don't have some basic linux commands like top or find.

steveford-obs, 2018-06-06:
@Pixelfahnder, if you want to build the wayland image, I had the same issue as you building cairo and it can be fixed, temporarily, by adding a "#include <GraphicsContext3D.h>" in MediaPlayerPrivateGStreamerBase.cpp.

However, I also have the most recent problem you have, which is that the image created doesn't work. Our Pi3 boots to a shell that is missing process control, so there's no 'ps' command, along with lots of others. I've noticed that 'busybox' is missing, though I'm not sure whether this has been replaced by coreutils, but none of the kernel modules are present either. I believe the wpe image should include 'core-image-minimal' and building this directly produces an image that boots and includes everything you'd expect, so the issue seems to be in the build process of wpe-westeros-image, but yocto/bitbake/etc are all new to me so I haven't been able to find out where it's going wrong.

Is anyone able to help with the problem creating a bootable system image?

Pixelfahnder, 2018-06-07:
@steveford-obs I was able to compile a bootable westeros image on the first hand but I'm not able to reproduce it; this is so depressing.

steveford-obs, 2018-06-07:
@Pixelfahnder, same problem here. I've tried master and sumo branches, as recommended above, but sumo doesn't build and master creates an image that doesn't boot correctly. However, both of these branches seem to have the same changes to their recipe files, though I don't know that this is the problem, for master at least.

Pixelfahnder, 2018-06-08:
@steveford-obs Should we do a voice call session and try to master it together? Maybe worth a try maybe we can assist each other and both learn something out of this.

gil0109, 2018-06-21:
I confirmed building from Master & Sumo for EGLFS  is successfully building but the image is still broken at startup.  See #212 for screen capture.  Any updates?

steveford-obs, 2018-06-22:
@gil0109, I had the same problem shown in your screen capture, but didn't get anywhere with it. Sorry.

gil0109, 2018-06-22:
@steveford-obs thanks for the update.  Since I tested the meta-raspberrypi layer and it successfully built, I am assuming it's something in meta-wpe.  I am currently building with commit 509dc2a815b8f4afd1f2c9a2da61e8ca9230adf9.  If that works, I will work way up to find the commit that caused the issue.  I am hoping it's not something upstream. 

And hopefully @wouterlucas will find something as he is testing as well.

BatslyAdams, 2018-06-22:
Just wanted to say that I'm experiencing the same issue when targeting rpi3/eglfs on sumo / ubuntu 16.04. Successful build with a hang on boot when run off the card (identical to screenshot posted earlier). I've got a fairly fast machine here if testing a different build will help and happy to assist in any other way, appreciate the effort.

gil0109, 2018-06-22:
@BatslyAdams Could you try a few builds and report your what is in your layer.conf as well?  My machine is really slow at this.

BatslyAdams, 2018-06-22:
Sure!

MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
WPE_BACKEND_rpi  = "rpi"
DISTRO_FEATURES_remove = "wayland"

I made sure each layer was the same branch through the repo tool (rocko/sumo)

Rocko shows Pi boot screen, but nothing gets output to log and underscore cursor stops flashing after a few seconds
[wpe-eglfs-image-raspberrypi3.rpi-sdimg.rocko](https://drive.google.com/open?id=1-LSxGwCc3fCsLLrskO0nMEQz53KfxG5T)

Sumo loads for about 3 seconds, then hangs as shown in previous screenshot
[wpe-eglfs-image-raspberrypi3.rpi-sdimg.sumo](https://drive.google.com/open?id=1__BoCiZhs6IXCGwDkfQ9tqBr3K8xQXyY)

Sorry that I didn't take screencaps, I'll do it on next boot, hope that helps.

gil0109, 2018-06-22:
@BatslyAdams Can you try with previous commits to meta-wpe repo as well?  Say the last commit on Jun 1, May 15, May 1?

BatslyAdams, 2018-06-22:
Sure, just to clarify: I'll just checkout the last commit from those dates in meta-wpe, then run 'bitbake wpe-eglfs-image' correct? Do I need clean anything first or will it rebuild automatically?

gil0109, 2018-06-22:
Great question, i am not sure.  Try it and let me know.

I am learning too..

gil0109, 2018-06-23:
@BatslyAdams Any luck?

To go back, I used this command:
git reset --hard e9f0560783b6f39081a24d0b531fddc409f0fc3b

but had some gstreamer errors so I think (still learning) i just added the latest commit to fix that by using this command: git  cherry-pick 05f5525  -m 1

so running git status shows this:
Your branch and 'webplatformforembedded/master' have diverged,
and have 1 and 29 different commits each, respectively.

Did you have the same issue?  My builds take forever, so I am looking forward to hearing your results.

BatslyAdams, 2018-06-23:
Sorry, no luck here. I'm new to yocto so I'm just reading up a bit more to make sure that I'm building properly.

FYI, I've been doing `repo forall -c git checkout <branch>` then `repo sync` before the bitbake command.

gil0109, 2018-06-23:
Same issue when going back to e9f0560: wpebackend-rdk: enable virtual input committed on May 24

@wouterlucas Any other suggestions?

gil0109, 2018-06-23:
@BatslyAdams Any chance you tried to build wpe-initramfs-image?  If not, can you ?

gil0109, 2018-06-23:
@BatslyAdams and @wouterlucas Confirmed the following commit worked:  5920963af33db8ca32f3c16d6b10375a742ae468

I need help finding the exact commit that broke the image boot incorrectly issue.

gil0109, 2018-06-24:
I believe  the issue is with recipes-core/images/wpe-image.inc & this line:
   IMAGE_INSTALL_rpi += "wpebackend-rdk"

If I remove that line from the file and build, I get the a login in screen.  Not sure if that causes another problem though.  The display shows an IP, when I browse to it from another machine, connection refused.

BatslyAdams, 2018-06-24:
@gil0109 I've built the intiramfs image but haven't tested it on my device yet.

Are there any builds/branches you'd like me to verify as working here? Learned a whole lot about how the process works this weekend.

gil0109, 2018-06-24:
I think we are good, I am hoping someone can help figure out why having "wpebackend-rdk" in wpe-image.inc breaks the rp3 image. And find a solution.

Maybe @BatslyAdams, you could build the master and exit the bb to use wpebackend instead of wpebackend-rdk.  I am currently building a chromium image instead but sadly the image is 30% larger than webkit and is overkill for my needs.

wouterlucas, 2018-06-24:
We need wpebackend-rdk for graphics on RPI. Its weird that the image fails with that included, it has always been part of the image and I don’t see how including a package breaks the boot on RPI. Makes no sense. 

I’ll be home tomorrow, I’ll be able to help then.

gil0109, 2018-06-25:
@wouterlucas Another newby question :)

In the docs it says to add this line: DISTRO_FEATURES_remove = "x11"
But isn't this needed to run WPEFramework without Westeros/Wayland support on RPI (fullscreen EGL)?

The reason I asked it that WPEFramework fails with this message:
Activation of plugin [WebKitBrowser]:[UX] blocked
Activation of plugin [WifiControl]:[WifiControl] blocked

Thanks

wouterlucas, 2018-06-25:
Removal of X11 is required to make sure certain packags do not fall back to X11 when disabling Wayland.

The messages:
```
Activation of plugin [WebKitBrowser]:[UX] blocked
Activation of plugin [WifiControl]:[WifiControl] blocked
```

Are unrelated to X11, these are due to our internal dependency / autostart management. You can press `s` to get the events and see if all the events are available or if certain are unavailable.

Not starting UX is normal, that by default is not enabled for autostart in the configuration. Not starting WifiControl can be because of lacking wpa-supplicant. Are you building for and RPI2 or RPI3? 

I'll fire up a fresh build today looking deeper into that line you've found. Much appreciated for all the due diligence.

gil0109, 2018-06-25:
I am using RPI3.

gil0109, 2018-06-25:
And if up meant "PS", here is the output:
root@raspberrypi3:~# ps
  PID USER       VSZ STAT COMMAND
    1 root      1740 S    init [5]
    2 root         0 SW   [kthreadd]
    4 root         0 IW<  [kworker/0:0H]
    5 root         0 RW   [kworker/u8:0]
    6 root         0 IW<  [mm_percpu_wq]
    7 root         0 SW   [ksoftirqd/0]
    8 root         0 IW   [rcu_sched]
    9 root         0 IW   [rcu_bh]
   10 root         0 SW   [migration/0]
   11 root         0 SW   [cpuhp/0]
   12 root         0 SW   [cpuhp/1]
   13 root         0 SW   [migration/1]
   14 root         0 SW   [ksoftirqd/1]
   16 root         0 IW<  [kworker/1:0H]
   17 root         0 SW   [cpuhp/2]
   18 root         0 SW   [migration/2]
   19 root         0 SW   [ksoftirqd/2]
   20 root         0 IW   [kworker/2:0]
   21 root         0 IW<  [kworker/2:0H]
   22 root         0 SW   [cpuhp/3]
   23 root         0 SW   [migration/3]
   24 root         0 SW   [ksoftirqd/3]
   25 root         0 IW   [kworker/3:0]
   26 root         0 IW<  [kworker/3:0H]
   27 root         0 SW   [kdevtmpfs]
   28 root         0 IW<  [netns]
   31 root         0 IW   [kworker/2:1]
   32 root         0 IW   [kworker/3:1]
   33 root         0 SW   [khungtaskd]
   34 root         0 SW   [oom_reaper]
   35 root         0 IW<  [writeback]
   36 root         0 SW   [kcompactd0]
   37 root         0 IW<  [crypto]
   38 root         0 IW<  [kblockd]
   39 root         0 IW<  [watchdogd]
   40 root         0 IW<  [rpciod]
   41 root         0 IW<  [xprtiod]
   42 root         0 SW   [kswapd0]
   43 root         0 IW<  [nfsiod]
   53 root         0 IW<  [kthrotld]
   54 root         0 IW<  [iscsi_eh]
   55 root         0 IW<  [dwc_otg]
   56 root         0 IW<  [DWC Notificatio]
   57 root         0 SW<  [vchiq-slot/0]
   58 root         0 SW<  [vchiq-recy/0]
   59 root         0 SW<  [vchiq-sync/0]
   60 root         0 SW   [vchiq-keep/0]
   61 root         0 SW<  [SMIO]
   62 root         0 SW   [irq/92-mmc1]
   66 root         0 SW   [mmcqd/0]
   67 root         0 IW<  [kworker/3:1H]
   68 root         0 SW   [jbd2/mmcblk0p2-]
   69 root         0 IW<  [ext4-rsv-conver]
   70 root         0 IW<  [kworker/0:1H]
   73 root         0 IW<  [kworker/2:1H]
  103 root      3996 S    /sbin/udevd -d
  107 root         0 IW<  [kworker/1:1H]
  129 root         0 IW<  [cfg80211]
  132 root         0 IW   [kworker/u8:1]
  244 root      2788 S    udhcpc -R -b -p /var/run/udhcpc.eth0.pid -i eth0
  251 messageb  2876 S    /usr/bin/dbus-daemon --system
  257 root         0 IW<  [ipv6_addrconf]
  258 root      2404 S    /usr/sbin/dropbear -r /etc/dropbear/dropbear_rsa_host
  264 root      4452 S    /usr/libexec/bluetooth/bluetoothd
  268 root      2788 S    /sbin/syslogd -n -O /var/log/messages
  271 root      2788 S    /sbin/klogd -n
  289 root      3084 S    -sh
 2046 root         0 IW   [kworker/0:2]
 2440 root         0 IW   [kworker/0:1]
 2702 root         0 IW   [kworker/1:2]
 2837 root         0 IW   [kworker/1:0]
 2855 root      2496 R    /usr/sbin/dropbear -r /etc/dropbear/dropbear_rsa_host
 2856 root      3068 S    -sh
 3007 root         0 IW   [kworker/1:1]
 3012 root      2880 R    ps

wouterlucas, 2018-07-12:
I think this thread has gotten a bit too large and has been (partially) been answered in other issue threads. Closing it off.

closed 2018-07-12 by wouterlucas

skanagaraj1, 2018-11-09:
Hi, I am seeing the same error. Could you please help me to find out what could have caused these error?

[PluginServer.cpp:336]: Startup: Activated plugin [Controller]:[Controller]
[PluginServer.cpp:586]: Startup: Activation of plugin [Bluetooth]:[Bluetooth] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [Dictionary]:[Dictionary] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [WebKitBrowser]:[UX] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [WifiControl]:[WifiControl] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [WebKitBrowser]:[YouTube] blocked
[PluginHost.cpp:403]: Startup: WPEFramework actively listening.
[PluginServer.cpp:336]: Startup: Activated plugin [LocationSync]:[LocationSync]
[PluginServer.cpp:279]: Startup: Activation of plugin [TimeSync]:[TimeSync], postponed, preconditions have not b.
[PluginServer.cpp:336]: Startup: Activated plugin [TraceControl]:[Tracing]
[PluginServer.cpp:336]: Startup: Activated plugin [Snapshot]:[Snapshot]
[PluginServer.cpp:336]: Startup: Activated plugin [DeviceInfo]:[DeviceInfo]
[PluginServer.cpp:336]: Startup: Activated plugin [Monitor]:[Monitor]
[PluginServer.cpp:336]: Startup: Activated plugin [RemoteControl]:[RemoteControl]
WPEFramework[22158]: open failed
WPEFramework[22158]: 18:02:19:760 06 Assert 124!../../RefTarget_ZRC_MSO_GP712_
kill[22201]: kill: you need to specify whom to kill
systemd[1]: wpeframework.service: Control process exited, code=exited status=1

wouterlucas, 2018-11-09:
You issue is being tracked here https://github.com/WebPlatformForEmbedded/meta-wpe/issues/250, please keep things separated. That's likely a RF4CE issue.