WebPlatformForEmbedded/meta-wpe #210: wpewebkit is not compiling in latest master

hhromic, 2018-05-28:
Hi guys,
I went back to my habit of building `meta-wpe` regularly and I noticed that the `wpe-westeros-image` is not building anymore, failing to compile `wpewebkit`. It looks like it's an upstream problem.
I'm raising this issue in case you haven't noticed yet.  The details below:

```
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "master:d886fa118c930d0e551f2a0ed02b35d08617f746"
meta-oe
meta-python
meta-multimedia
meta-networking      = "master:d59b9806f743cea0168cddf942a31dcf446839e6"
meta-raspberrypi     = "master:de597abc3b9447611951469ad1ad18d5068ff074"
meta-wpe             = "master:f9c5f17c5acf835fd3166f5b6ca069a4b1e01ac5"
```

The `local.conf` file:

```
MACHINE = "raspberrypi0-wifi"
SERIAL_CONSOLE = ""
DISABLE_RPI_BOOT_LOGO = "1"
DISABLE_OVERSCAN = "1"
DISABLE_SPLASH = "1"
BOOT_DELAY = "0"
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = ""
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
PACKAGE_CLASSES = "package_ipk"
IMAGE_INSTALL_append = " linux-firmware-bcm43430 wpa-supplicant"
IMAGE_FSTYPES = "tar.bz2 ext4 rpi-sdimg"
SDIMG_ROOTFS_TYPE = "ext4"
```

And the compilation warnings/error:

```
/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+7059b7c7b0-r1/git/Source/bmalloc/bmalloc/Gigacage.h:136:59: warning: cast from 'char*'
 to 'Gigacage::BasePtrs*' increases required alignment of target type [-Wcast-align]
     return *reinterpret_cast<BasePtrs*>(g_gigacageBasePtrs);
                                                           ^
{standard input}: Assembler messages:
{standard input}:328: Error: invalid literal constant: pool needs to be closer
{standard input}:329: Error: invalid literal constant: pool needs to be closer
{standard input}:334: Error: invalid literal constant: pool needs to be closer
{standard input}:335: Error: invalid literal constant: pool needs to be closer
{standard input}:340: Error: invalid literal constant: pool needs to be closer
{standard input}:341: Error: invalid literal constant: pool needs to be closer
{standard input}:346: Error: invalid literal constant: pool needs to be closer
{standard input}:347: Error: invalid literal constant: pool needs to be closer
{standard input}:352: Error: invalid literal constant: pool needs to be closer
{standard input}:353: Error: invalid literal constant: pool needs to be closer
{standard input}:358: Error: invalid literal constant: pool needs to be closer
{standard input}:359: Error: invalid literal constant: pool needs to be closer
{standard input}:364: Error: invalid literal constant: pool needs to be closer
{standard input}:365: Error: invalid literal constant: pool needs to be closer
{standard input}:371: Error: invalid literal constant: pool needs to be closer
{standard input}:377: Error: invalid literal constant: pool needs to be closer
```

There are a lot of these same casting warnings, but this one is the only one with an actual error from the assembler compiler.

hhromic, 2018-05-28:
After thinking a bit about this, it might be related that I'm building for armv6 here and might be an upstream bug. Similar to https://github.com/WebPlatformForEmbedded/WPEWebKit/issues/445 that so far didn't get any attention.
However this time it's not compiling instead of just warning.

wouterlucas, 2018-05-28:
ARMv6 might be a separate issue, but the Westeros build doesn't work with 2.20 WPE WebKit. I've been running with the EGLFS one until this is resolved.

hhromic, 2018-05-29:
Yes I can confirm now that for RPI3 it also doesn't build with Westeros.
Then we can leave the issue open to remind us that this is an upstream blocker.
I will rename the issue accordingly too.

gil0109, 2018-06-14:
Sorry, I am new to Yocto but do you think it would ok to update the /meta-wpe/recipes-graphics/westeros/westeros.inc file and change the value of the WESTEROS_SCREV
from:  "319b26391a64a5695568eac5ff9ebf294dafee68"
to: "${AUTOREV}"

wouterlucas, 2018-06-14:
Setting to autorev is certainly a possibility, Westeros has been pretty stable in the past. We could to that for other parts of the solution to instead of keeping bumps. 

The only downside is however that you may run into situations where the build is temporarily broken and no one did a validation. I'm okay with that, but not sure if everyone else would :)

gil0109, 2018-06-14:
Thanks, I will update on my local machine just to get the Sumo branch working until it's resolved.  Maybe we should have a bleading edge branch .. haha.

kraj, 2018-06-14:
No to AUTOREVs in general please. It will break builds where policy it to disallow fetches from outside firewall and you are at the behest of westeros guys doing right thing otherwise you are broken. Do you want to give that control to packages ?

hhromic, 2018-06-14:
Hi guys, I would advocate **against** automatically following the master HEAD of wpewebkit and other "complex" softwares, e.g. westeros.
In my opinion these are kind of fragile and is best to bump their revisions only on known working states.

wouterlucas, 2018-06-14:
Definitely agree on release branches, we should carefully control the releases. However we technically could for the master branch? Or also against that @kraj @hhromic ?

kraj, 2018-06-14:
All I care is that we lock the revision and when we move to new version then we test it before making the SRCREV bump

hhromic, 2018-06-15:
I also support keeping the system like it is: with known revisions and periodic bumps only after testing them. Otherwise the stability will get out of hands :)

kraj, 2018-06-15:
if you want to do CI for say wpe, then you can smartly override the revision
```
SRCREV_pn-<recipename> = "${AUTOREV}"
```
in local.conf on a autobuilder or local build machine but lets not hardcode it.

wouterlucas, 2018-06-15:
OK - we'll keep it the way it is and please use Khem's suggestion if you want to locally be more on the bleeding edge.

kraj, 2018-06-22:
@hhromic can you try using thumb2 ISA ? you can set this in ```local.conf```
```
ARM_INSTRUCTION_SET = "thumb"
```

if this works then we need a patch for ```recipes-wpe/wpewebkit/wpewebkit_2.20.bb```

but there is no common override for arm1176jzfshf so we need to figure out what would be most suitable way, Probably if you post your OVERRIDES list for rpi0-wifi then we can figure out more.

hhromic, 2018-06-25:
@kraj, setting `ARM_INSTRUCTION_SET` to `thumb` in `loca.conf` doesn't seem to do anything.
The Bitbake environment doesn't show `thumb` in `TUNE_FEATURES`:
```
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
```
And the build of wpewebkit still fails. Maybe for recent Yocto setting the ISA to thumb is done differently? I didn't seem to find anything about it :/

hhromic, 2018-07-02:
@kraj, looks like the Thumb2 ISA is not supported on RPI0 hence why Yocto doesn't enable it:
See: <https://www.raspberrypi.org/forums/viewtopic.php?t=209461>

kraj, 2018-07-02:
pi0 is based on BCM2835 SOC which is based on ARM1176JZ(F)-S processor which should be supporting thumb2.

hhromic, 2018-07-03:
I see. Unfortunately I don't have any confidence/experience on altering tuning in Yocto or any other build system :(. Perhaps you can shed some light for some tests.
However, after checking, looks like this is something that would be need to be done at the `meta-raspberrypi` layer level in `conf/machine/include/tune-arm1176jzf-s.inc` more than here in `meta-wpe`.
Also I have been reading that armv6 supports thumb but in turn potentially makes the vpu slower. I haven't found clear documentation around this beyond forums (e.g. opinions).
Shall we raise an issue in meta-raspberrypi to be discussed with the people there?

kraj, 2018-07-03:
can you try with forcing

```
ARM_INSTRUCTION_SET = "arm"
```
in WPE recipe ?

and separately upload one compiler cmdline for anyone source file. I want to see what options are in
play. Its possible that we can workaround this

hhromic, 2018-07-04:
hi @kraj , here are compiler lines per situation,  all for the `raspberrypi0-wifi` machine.

Leaving the WPE recipe as-is:
```
/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++  -DBUILDING_JavaScriptCore -DBUILDING_WITH_CMAKE=1 -DBUILDING_WPE__=1 -DDATA_DIR=\"share\" -DGETTEXT_PACKAGE=\"WPE\" -DHAVE_CONFIG_H=1 -DMETROLOGICAL=1 -IDerivedSources/ForwardingHeaders -isystem /home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/recipe-sysroot/usr/include/glib-2.0 -isystem /home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/recipe-sysroot/usr/lib/glib-2.0/include -I. -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/API -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/assembler -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/b3 -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/b3/air -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/bindings -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/builtins -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/bytecode -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/bytecompiler -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/dfg -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/disassembler -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/disassembler/ARM64 -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/disassembler/ARMv7 -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/disassembler/udis86 -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/domjit -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/ftl -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/heap -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/debugger -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/inspector -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/inspector/agents -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/inspector/augmentable -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/inspector/remote -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/interpreter -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/jit -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/llint -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/parser -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/profiler -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/runtime -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/tools -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/wasm -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/wasm/js -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/yarr -IDerivedSources/JavaScriptCore -IDerivedSources/JavaScriptCore/inspector -IDerivedSources/JavaScriptCore/runtime -IDerivedSources/JavaScriptCore/yarr -I/home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d [... cut ...] MakeFiles/JavaScriptCore.dir/llint/LowLevelInterpreter.cpp.o -MF Source/JavaScriptCore/CMakeFiles/JavaScriptCore.dir/llint/LowLevelInterpreter.cpp.o.d -o Source/JavaScriptCore/CMakeFiles/JavaScriptCore.dir/llint/LowLevelInterpreter.cpp.o -c /home/yocto/wpe/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/wpewebkit/2.20+gitAUTOINC+d3c3f5df97-r2/git/Source/JavaScriptCore/llint/LowLevelInterpreter.cpp
```

I compared the three outputs and they are exactly the same (no compiler flags/arguments change).

kraj, 2018-07-05:
@hhromic yes I somehow overlooked the fact that 1176 is not direct descendent of 1156, so yes it does not support thumb2. Using thumb1 while possible, I would not recommend it since its not used as much and hence not a well maintained backend in gcc. So actually setting **ARM_INSTRUCTION_SET** to thumb is not a good option since that would make it build with thumb1 if it worked

original error you are seeing is due to datapool being emitted too far from where they are accessed so it does seem to be either gcc bug where its not accounting for the distance ( should be less than 4K in case of LDR )

One workaround could be to generate per function ELF section which would force it to emit current pool after every function.

Try adding 
```
-ffunction-sections -fdata-sections
```
via cflags/cxxflags and see if it helps.

hhromic, 2018-07-05:
@kraj, I parsed the cmdline above and highlighted the important flags to be more easily readable:
```
-fdata-sections
-fdiagnostics-color=always
-feliminate-unused-debug-types
-ffunction-sections
-finline-limit=90
-fno-exceptions
-fno-rtti
-fno-strict-aliasing
-fPIC
-fsection-anchors
-fvisibility=hidden
-fvisibility-inlines-hidden
-march=armv6
-mfloat-abi=hard
-mfpu=vfp
-MD
-MF
-MT
-mtune=arm1176jzf-s
-O2
-std=c++14
-Wall
-Wcast-align
-Wextra
-Wformat-security
-Wmissing-format-attribute
-Wno-expansion-to-defined
-Wno-maybe-uninitialized
-Wno-noexcept-type
-Wpointer-arith
-Wundef
-Wwrite-strings
```
And as you can see, the flags you suggest are already there during compilation :( Any other ideas?

kraj, 2018-07-05:
ugh. Can we try with -Os instead of -O2 in wpe recipe ?
```
SELECTED_OPTIMIZATION_remove = "-O2"
SELECTED_OPTIMIZATION_append = " -Os"
```

hhromic, 2018-07-05:
@kraj, unfortunately that didn't work either, exact same asm error.
The problem file is:
https://github.com/WebPlatformForEmbedded/WPEWebKit/blob/master/Source/JavaScriptCore/llint/LowLevelInterpreter.cpp
and it seems to rely on a lot of assembler code. Perhaps this is a programming problem more than a GCC/compilation issue.
I tried using RPI3 as machine and the **compilation succeeds**, but with the following messages:
```
{standard input}: Assembler messages:
{standard input}:52: use of r13 is deprecated
{standard input}:198: use of r13 is deprecated
```
Apparently the issue is armv6-only. Do you think we should migrate the issue to the WPEWebKit repository or do you want to give it another try here?

kraj, 2018-07-05:
actually JIT asm is not working arm mode and only thumb2 ISA is where it works. I think we should be able to disable it when we dont have thumb2 available. Please add the below line to wpe recipe

```
EXTRA_OECMAKE_append_armv6 = " -DENABLE_JIT=OFF "
```

hopefully this should fix it for you.

hhromic, 2018-07-06:
finally we are having some good news by disabling the JIT for armv6 :)
now wpewebkit compiles! ..but doesn't install :( The error is:
```
UnifiedSource69.cpp:(.text._ZN3JSC25uniqueSecurityOriginTokenEv+0x20): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
```
I found a similar issue here https://github.com/PurpleI2P/i2pd/issues/897 that suggests adding `-latomic` to the linker. What do you think? Probably we'll need to add libatomic as build dependency. I will try to test this somehow. Open to your suggestions anyway @kraj !

@wouterlucas too sad to read that, my main motivation to work with meta-wpe is that it was the only browser that worked decently in the RPI0 for a personal project. I understand the RPI0/1 is not a powerful machine but supporting it was a very big plus for me :(  I will open an issue in WPEWebKit depending on if we give-up here, hopefully they care to keep it working :)

kraj, 2018-07-06:
linking to libatomic is right approach for ARM  <= V6 since atomic compiler intrinsics are not available there. However if you can help understand where the problem is we can surgically apply the link fix. If there are more than one place then we might need to link unconditionally.

kraj, 2018-07-06:
Can you try this patch 
```patch
diff --git a/Source/JavaScriptCore/CMakeLists.txt b/Source/JavaScriptCore/CMakeLists.txt
index 26e3b5388d7..849b9220c82 100644
--- a/Source/JavaScriptCore/CMakeLists.txt
+++ b/Source/JavaScriptCore/CMakeLists.txt
@@ -115,16 +115,16 @@ set(JavaScriptCore_LIBRARIES
     ${ICU_I18N_LIBRARIES}
     ${LLVM_LIBRARIES}
 )
-
-# Since r228149, on MIPS we need to link with -latomic, because
-# __atomic_fetch_add_8 is not available as a compiler intrinsic. It is
-# available on other platforms (including 32-bit Arm), so the link with
-# libatomic is only neede on MIPS.
-if (WTF_CPU_MIPS)
-    list(APPEND JavaScriptCore_LIBRARIES
-        -latomic
-    )
+# Check if libatomic is needed in order to use std::atomic, and add
+# it to the list of JavaScriptCore libraries.
+file(WRITE ${CMAKE_BINARY_DIR}/test_atomic.cpp
+     "#include <atomic>\n"
+     "int main() { std::atomic<int64_t> i(0); i++; return 0; }\n")
+try_compile(ATOMIC_BUILD_SUCCEEDED ${CMAKE_BINARY_DIR} ${CMAKE_BINARY_DIR}/test_atomic.cpp)
+if (NOT ATOMIC_BUILD_SUCCEEDED)
+    list(APPEND JavaScriptCore_LIBRARIES -latomic)
 endif ()
+file(REMOVE ${CMAKE_BINARY_DIR}/test_atomic.cpp)

 set(JavaScriptCore_SCRIPTS_SOURCES_DIR "${JAVASCRIPTCORE_DIR}/Scripts")
```

hhromic, 2018-07-06:
A-ha!  after investigating and setting libatomic here unconditionally (as a test):
https://github.com/WebPlatformForEmbedded/WPEWebKit/blob/b67bfec6fea168e3f9dd6e71d84c8c2f53610d0e/Source/JavaScriptCore/CMakeLists.txt#L119
the compilation/installation now all succeeds!
Turns out not only MIPS requires explicit `-latomic` but apparently armv6 too (as @kraj just mentioned).

I now need to know the correct defines to use in the Cmake conditional alongside MIPS. Any suggestion @kraj? I was thinking on conditioning to `WTF_ARM_ARCH_VERSION <= 6` but not sure how this works in Cmake syntax (not familiar with it). Something like:
```
if (WTF_CPU_MIPS OR WTF_CPU_ARM AND WTF_ARM_ARCH_VERSION <= 6)
  ....(set -latomic)
```
I can then send a PR to the WPEWebKit repo and hopefully be merged soon. Alternatively we could try doing this fix in the recipe here as a patch, but I strongly believe this should be in the upstream cmakefile of WPEWebKit. Any thoughts?

hhromic, 2018-07-06:
Hehe a bit of a race condition here commenting =). I was reaching the same conclusion.

Interesting patch, it actually tries to compile an atomic no matter the architecture and set `-latomic` accordingly.  However I feel it a bit low-level (creating a source file, compile it etc) and hackish. Isn't it better/cleaner to use an architecture-define as is done for the MIPS conditional?

Also, actually that test is performed by cmake but somehow doesn't affect the usage of libatomic:
```
-- Performing Test ATOMIC_INT64_IS_BUILTIN
-- Performing Test ATOMIC_INT64_IS_BUILTIN - Failed
-- Performing Test ATOMIC_INT64_REQUIRES_LIBATOMIC
-- Performing Test ATOMIC_INT64_REQUIRES_LIBATOMIC - Success
```
Maybe we can condition to `ATOMIC_INT64_REQUIRES_LIBATOMIC` ?

kraj, 2018-07-06:
No, the patch is correct way of detecting feature support for atomics, this does not depend upon architecture but on compiler being able to support it. And different compiler can provide different kind of support e.g. these atomics may not existing on i586 for clang but might for gcc and so on. You approach will end up with workarounds on workarounds.

hhromic, 2018-07-06:
I'm testing the patch you suggest now, probably will work as it addresses the actual problem.
What do you suggest me to proceed? send a PR to WPEWebKit? any chance to accelerate the merge there?

kraj, 2018-07-06:
Prepare a patch for meta-wpe and also for wpewebkit separately. we can live with meta-wpe patch till it gets merged into wpewebkit

commit 9272374e (2018-07-07): Merge pull request #219 from hhromic/fix-armv6-build

wpewebkit: fix armv6 build (fixes #210)
closed 2018-07-07 by wouterlucas