WebPlatformForEmbedded/meta-wpe #212: Error

gil0109, 2018-06-15:
Trying to Build with meta-wpe (Rocko) which I need to do due to my Mender addon's.

Getting this error:

ERROR: Command Error: 'quilt --quiltrc /home/gillani/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/20170728+gitAUTOINC+46e07d52d8-r1/recipe-sysroot-native/etc/quiltrc push' exited $
Applying patch 0001-mse-Print-MediaTime-consistently-using-s.patch
patching file Source/WebCore/platform/graphics/gstreamer/mse/MediaPlayerPrivateGStreamerMSE.cpp
Hunk #1 FAILED at 207.
1 out of 1 hunk FAILED -- rejects in file Source/WebCore/platform/graphics/gstreamer/mse/MediaPlayerPrivateGStreamerMSE.cpp

For now, I have Remarked out:  SRC_URI += "file://0001-mse-Print-MediaTime-consistently-using-s.patch"  in the file: wpewebkit_20170728.bb

Any suggestions?

kraj, 2018-06-15:
you might have to forward port those patches

gil0109, 2018-06-15:
And now I will learn how to do that.. thanks ;)

wouterlucas, 2018-06-15:
That patch has been upstreamed, seems like I forgot to remove it. Try again with https://github.com/WebPlatformForEmbedded/meta-wpe/commit/a3bbafbb8ac122c11f29c0d02c077d8e1123abc7

gil0109, 2018-06-15:
Great thanks, updated the repo (which updated based on your commit).  Building now.. wish me luck.. it failed at 98% so I am hoping it's my last issue.  I will report the results so you can close this thread.

The only other thing I had to manually fix was this missing patch:
https://github.com/WebPlatformForEmbedded/meta-wpe/pull/202/files#diff-4a3a7801d8bfe07e71e467c54217a2a8

Can you upstream this to the Rocko branch?

gil0109, 2018-06-15:
@wouterlucas worked like a charm, thank you.  Closing issue.

wouterlucas, 2018-06-18:
@gil0109 Dynamic layers isn't support on Rocko AFAIK, so that commit you referenced wont help. What's the other thing you are running into?

gil0109, 2018-06-18:
After the build seemed successfully completed.  The image boots with no power to usb or networking.  4 Raspberry icons and that's it.  Maybe the layer fix caused this?

Honestly, I can not get any of the branches ( Master,Sumo or Rocko ) to build without error.
wpe-westeros-image or  wpe-eglfs-image with an RPI3

Any suggestions of which works right now?

My Ultimate goals is to build a Digital Signage image (to replace my raspbian stretch image) with the following:
1. Webkit with GPU Acceleration
2. Fullscreen launch with URL based on a sites files listing mac/url/timezone
3. Read only (currently I use Overlay)
4. Add Mender Layer for maintenance and patching
5. UFW to lock it down more
6. Change default user id and password (for security reasons)

wouterlucas, 2018-06-18:
Hmm, unfortunately I cant check right now as I'm traveling. But can you try reverting https://github.com/WebPlatformForEmbedded/meta-wpe/commit/b4fb4f2e1226f953cc2d8848b3307de1ab15a73c ?

That's the only 'major' thing that changed to the composition of the images. Albeit that worked for me on older versions, maybe newer branches are behaving differently.

gil0109, 2018-06-18:
@wouterlucas OK, Tried with master and sumo and fails with same issue with from: https://github.com/WebPlatformForEmbedded/meta-wpe/issues/210.

Now trying the following.. 

Build Configuration:
BB_VERSION           = "1.39.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5+snapshot-20180618"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "sumo:0beae3326a89060d353bf177f5acb10e12421793"
meta-oe
meta-python
meta-multimedia
meta-networking      = "sumo:52904e753532d095b14972fa03a25e7e24696159"
meta-raspberrypi     = "sumo:b2da4618b0ac2ad7dbc26387cad4c03796f8a06e"
meta-wpe             = "b4fb4f2:91284a15d6927ad1e2e6d41fb222b4f36bf1aa4b"

Fails with same issue with from: https://github.com/WebPlatformForEmbedded/meta-wpe/issues/210

Will try  bitbake wpe-eglfs-image again with this in my local.conf:

MACHINE = "raspberrypi3"
#WPE Changes
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
WPE_BACKEND = "rpi libinput"
WPE_PLATFORM = "egl"
NETFLIX_BACKEND = "rpi"
DISTRO_FEATURES_remove = "wayland"
GCCVERSION = "8.%"

gil0109, 2018-06-18:
This worked but I am unsure how to launch the browser, any ideas?

wouterlucas, 2018-06-19:
@gil0109 is that an error while logging in or an error while trying to run the command?

Because with the latest version there is no WPELauncher anymore, however reverting that commit might have messed up the packaging.

gil0109, 2018-06-19:
@wouterlucas Sorry for being so new at this. 

That's the error when booting.  There is no logon.  I get to the shell with no password.  How do I launch a URL from the shell?

wouterlucas, 2018-06-19:
No worries :) 

So you can login, but you cannot launch WPELauncher after reverting that commit locally correct? 
Is there anything on screen? Because if I "theoretically" follow what you did you should have a build with WPEFramework. Which autostarts and lauches a welcome screen that displays the device's IP.

If you can see that welcome screen, take your normal PC's browser and navigate to the IP shown on screen. That should give you access to the browser and be able to launch a URL.

However if that welcome screen is not there, check if the `WPEFramework` process is running.

gil0109, 2018-06-19:
Interesting.  
I see the WPEFramework in /usr/bin.
No ifconfig so I can't see what IP it has.

Do you think it's worth rebuilding using master or sumo for everything?  Maybe that will fix the issue?
I am assuming the image handles DHCP.

gil0109, 2018-06-19:
Also, another question.  Does that page show up if you use Image without Wayland (as westeros fails to build)?

WPE_BACKEND = "rpi libinput"
WPE_PLATFORM = "egl"
NETFLIX_BACKEND = "rpi"
DISTRO_FEATURES_remove = "wayland"

wouterlucas, 2018-06-19:
No ifconfig is a bit odd and yes that should work just fine with the settings you've noted above. Though only the `DISTRO_FEATURES_remove` should be required. 

I'm trying to see what I can do using qemu while on the go, but it might take me a while to be of help. Best case I have some results by tomorrow in order to try to reproduce.

gil0109, 2018-06-19:
@wouterlucas Thank you so much for your help.  I look forward to hearing how your tests go.

I will also rebuild using these settings exactly from: https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/Raspberry-PI.  And see if that makes any difference.

Should take 5 hours to rebuild ;)

gil0109, 2018-06-19:
@wouterlucas  Built with no errors using the instructions provided for RP3.  I still have the same issue.  
 I am hoping you see something I am doing wrong.

Also, unable to run ps command.  Nothing in the /boot directory either.

Build Configuration:
BB_VERSION           = "1.39.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5+snapshot-20180619"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "HEAD:53dfa673d78216f852a47bdb48392ee213e3e1cd"
meta-oe
meta-python
meta-multimedia
meta-networking      = "HEAD:52904e753532d095b14972fa03a25e7e24696159"
meta-raspberrypi     = "HEAD:b2da4618b0ac2ad7dbc26387cad4c03796f8a06e"
meta-wpe             = "HEAD:05f55254ecec44bc2c4eb8eb0fc1389b763d6fc2"

BBLayers.conf:
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/gillani/rpi-yocto/poky/meta \
  /home/gillani/rpi-yocto/poky/meta-poky \
  /home/gillani/rpi-yocto/poky/meta-yocto-bsp \
  /home/gillani/rpi-yocto/meta-openembedded/meta-oe \
  /home/gillani/rpi-yocto/meta-openembedded/meta-python \
  /home/gillani/rpi-yocto/meta-openembedded/meta-multimedia \
  /home/gillani/rpi-yocto/meta-openembedded/meta-networking \
  /home/gillani/rpi-yocto/meta-raspberrypi \
  /home/gillani/rpi-yocto/meta-wpe \
  "
local.conf added these lines:
#MACHINE ??= "qemux86"
MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
GCCVERSION = "8.%"
WPE_BACKEND = "rpi libinput"
DISTRO_FEATURES_remove = "wayland"
PREFERRED_VERSION_linux-raspberrypi = "4.9%"

Hopefully you can see the output when running image on pi below:

[image]

gil0109, 2018-06-19:
@wouterlucas 

Not sure if this helps but I bitbake'd core-image-minimal.  That booted up with network.  Got to a login screen fine and networking.

wouterlucas, 2018-06-19:
Yeah we should include/require that so it should work the same. But maybe that inclusion/dependency isn't properly created anymore. I got my qemuarm image built and trying to boot it in QEMU so I can see what's going on. I'm traveling so wont get time to do this until later tonight.

wouterlucas, 2018-06-20:
Hmmm okay, so the qemu route wasn't very helpful in order to get more insight. I can build wpe-eglfs-image (minus WebKit, as that wont work under QEMU) and run it just fine. I get a prompt, login works and all the required bits and pieces are there to run the image including WPEFramework and dependencies.

Guess it must be something RPI specific that changed in meta-wpe and triggers this.

wouterlucas, 2018-06-20:
Just a stab in the dark, but can you locally remove this file and rebuild: https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/dynamic-layers/raspberrypi/recipes-core/base-files/base-files_%25.bbappend

That's the only RPI specific pieces in this layer, maybe the mounts are changed and that breaks the bootup.

gil0109, 2018-06-20:
Will do and I will report back.  PS the meta raspberry core image works fine too.

gil0109, 2018-06-21:
Yes, I chalk up the Kernel panic to a corrupt image.  I tried again but the same issue still occures.  I will give the include a try and report back.  Thanks

gil0109, 2018-06-21:
That didn't work either.  Same result. Any other suggestions?

Note: the wpe-image also fails.

Build was fine with the expected warnings:
WARNING: Duplicate inclusion for /home/gillani/rpi-yocto/poky/meta/recipes-core/images/core-image-                          minimal.bb in /home/gillani/rpi-yocto/meta-wpe/recipes-core/images/wpe-image.bb
WARNING: Duplicate inclusion for /home/gillani/rpi-yocto/poky/meta/recipes-core/images/core-image-                          minimal.bb in /home/gillani/rpi-yocto/meta-wpe/recipes-core/images/wpe-image.bb
WARNING: Duplicate inclusion for /home/gillani/rpi-yocto/poky/meta/recipes-core/images/core-image-                          minimal.bb in /home/gillani/rpi-yocto/meta-wpe/recipes-core/images/wpe-image.bb
WARNING: Duplicate inclusion for /home/gillani/rpi-yocto/poky/meta/recipes-core/images/core-image-                          minimal.bb in /home/gillani/rpi-yocto/meta-wpe/recipes-core/images/wpe-image.bb
WARNING: Duplicate inclusion for /home/gillani/rpi-yocto/poky/meta/recipes-core/images/core-image-minimal.bb in /home/gillani/rpi-yocto/meta-wpe/recipes-core/images/wpe-image.bb
WARNING: Duplicate inclusion for /home/gillani/rpi-yocto/poky/meta/recipes-core/images/core-image-minimal.bb in /home/gillani/rpi-yocto/meta-wpe/recipes-core/images/wpe-image.bb
NOTE: Tasks Summary: Attempted 3935 tasks of which 3926 didn't need to be rerun and all succeeded.

wouterlucas, 2018-06-21:
The duplicate is because https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-core/images/wpe-image.bb#L7 is still there, but that shouldn't hurt.

If that provide the same results I'm out of options that I can theoretically think of in terms of recent changes. It just doesn't make sense to me that other machines boot fine while RPI does not. The only other option is going back in history and try to find the offending change, I can try that once I get back home and have access to my setup. I remember running a RPI master setup about a month ago (late May?), the changes from June are mostly cherry-picks from other Yocto branches tested on other machines.

gil0109, 2018-06-21:
@wouterlucas I will try a couple more things.  I will add the rpi-basic-image.bb to the wpe-eglfs-image.bb.

And, I am going to try with removal of all the settings:
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
GCCVERSION = "8.%"
WPE_BACKEND = "rpi libinput"
DISTRO_FEATURES_remove = "wayland"
PREFERRED_VERSION_linux-raspberrypi = "4.9%"

And I look forward to your results.  Thanks.

gil0109, 2018-06-21:
@wouterlucas 

None of the above worked.  Do you think I have a corrupted build system?  Should I clear it and rebuild?
Also, I assumed I could use Ubuntu Server 16.04 to build the images.  Is that ok?

gil0109, 2018-06-21:
@wouterlucas it seems this issue was discovered in issue #209 .  Should we merge the two issues?  Sorry I didn't catch that.

gil0109, 2018-06-23:
@wouterlucas I believe there is something in this patch that is causing the boot issue.  Can you take a look at this commit? 9cd5a76bcf3928560a189396e57b88e502c194e5

wouterlucas, 2018-06-26:
@gil0109 Thanks for your extensive debugging, it was indeed that wpebackend-rdk image install line that messed things up. 

I'm unsure how, but I've moved the RPI specific to an RDEPENDS in the wpewebkit packagegroup and now it seems to come up for me (after also disabling bluetooth, thats broken on master). 

If all goes well you should see a WPE Hello screen that starts up automatically. That screen normally also displays the IP for the box, but for some reason its stuck at "Hello". I'll debug that later. Navigate with a browser of choice to the IP of the device and you should see a little control panel that allows you to interact with it. To change the URL go to the `WebKit Browser` tab and enter any URL you wish to try out.

Let me know if that works and looking forward to your feedback!

BatslyAdams, 2018-06-26:
It looks like I'm a bit late but I wanted to say I was able to successfully build last night by removing the '_rpi' suffix from IMAGE_INSTALL (`IMAGE_INSTALL_rpi += "wpebackend-rdk" -> IMAGE_INSTALL += "wpebackend-rdk"`). This resulted in a much larger image that would make it to the prompt and I could then ssh in, but still no display / service.

Caught the gs error by poking through the logs and moved `libWPEFrameworkBluetooth.so` to disable the plugin and I was up and running!

Out of curiosity, what did you both use in your local.conf settings for the eglfs build? I followed the wiki but I wanted to keep in line what what you all are building.

Love this project / appreciate the help @wouterlucas / @gil0109!

wouterlucas, 2018-06-26:
Please take the latest commit @BatslyAdams https://github.com/WebPlatformForEmbedded/meta-wpe/commit/2adec1aad29fd82e4d5ea42cb03a4846ff2b4493 I've disabled bluetooth for now and moved the wpebackend-rdk to an rdepends with https://github.com/WebPlatformForEmbedded/meta-wpe/commit/3f43d4b44144d7fad87b6bc969dbc08489024912

There is an API mismatch with the newer version of glib dbus, I'll raise it with the team and until that lets disable bluetooth on newer versions of Yocto.

wouterlucas, 2018-06-26:
And the only thing I have in my local conf is:
```
DISTRO_FEATURES_remove = "wayland"
```

gil0109, 2018-06-26:
@wouterlucas Should I have keep the x11 and opengl entries  as well? 

MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_remove = "wayland"

wouterlucas, 2018-06-26:
Shoot yes, that is needed. Sorry I assume that part as "static" of my configuration as I never touch it. My entire local.conf is:

```
MACHINE = "raspberrypi2"
IMAGE_FEATURES += "tools-debug"
#IMAGE_FEATURES += "tools-tweaks"
#IMAGE_FEATURES += "dbg-pkgs"
# rootfs for debugging
#IMAGE_GEN_DEBUGFS = "1"
#IMAGE_FSTYPES_DEBUGFS = "tar.gz"
#IMAGE_INSTALL_append = " gdbserver"
# explicitly disable x11 and enable opengl
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
ENABLE_UART = "1"
HOSTTOOLS_NONFATAL_append = " ssh"
# Enlarge boot partition to 160MiB
BOOT_SPACE = "163840"
CONF_VERSION = "1"

# enable packages
PACKAGE_CLASSES ?= "package_ipk"
#PACKAGE_CLASSES ?= "package_ipk"
PACKAGE_FEED_URIS = "http://192.168.11.100:8088/"
PACKAGE_FEED_BASE_PATHS = "ipk"
PACKAGE_FEED_ARCHS = "all cortexa7hf-neon-vfpv4 cortexa7t2hf-neon-vfpv4 raspberrypi2"

#DISTRO_FEATURES_remove = "wayland"
#DISTRO_FEATURES_remove = "wpeframework"
```

And for my RPI config I have some more WiFi / Bluetooth stuff in there but that's irrelevant. 

Also I like adding meta-96boards for a tool that expands the SD card on first boot (https://github.com/96boards/96boards-tools). That would add:
```
IMAGE_INSTALL_append += " 96boards-tools"
```

gil0109, 2018-06-26:
@wouterlucas Interesting that you have remarked out the wayland entry.

wouterlucas, 2018-06-26:
Typo, wanted to mark out the wpeframework. Guess edits don’t follow through on email :-)

gil0109, 2018-06-26:
@wouterlucas So I built it and it shows a hello screen.  I believe my DHCP takes a bit of time to give an IP address.. is there a simple change I can make to have the system wait for the IP before booting to the next part and setup the WPEFramework?

Also, is there an easy way to breakout of the "Hello" screen to a command line?

wouterlucas, 2018-06-26:
That should be fine, you can enable wpeframework distro feature and that way WPEFramework will take control of the network. In that mode the browser is started way early and the events will be processed as the system gets its IP. However it is a little more experimental then having the standard tools to get the network. 

There is no way to break out of the hello screen, you're looking at the browser that is loaded. Easiest is to either navigate your browser to the IP of the box or ssh into it. I'll look into why we're not reporting the IP on the screen, that will help if you do not know the IP before hand. 

You can of course disable startup of WPEFramework by commenting out the sysinit or systemd in the WPEFramework recipe, if you want to prevent WPEFramework from starting automatically. Though running an IP scanner or checking the router is probably as easy (I use Angry IP scanner and its great, http://angryip.org/).

gil0109, 2018-06-26:
@wouterlucas Thank you, I look forward for the IP addition.  It's hard as I don't manage the router and we are not allowed to scan the network in our location (without being blocked and in trouble).

A couple of questions if you don't mind.  Let me know if there is a better place to ask :)

1. My goal is to have the device launch a specified URL (for Digital Signage purpose) instead of me connecting to the device to push out the URL.  Is this easy?  What would the command be? WPEFramwork <url> ?  
2. I noticed your image was 170k larger than the one I created with Igalia.  Any idea why?  

Thanks

wouterlucas, 2018-06-26:
Certainly!

You can update the URL at build time or locally, to change it at build time set this variable to something else:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-wpe/wpeframework/wpeframework-plugins_git.bb#L20

Or you can change it on your target machine by editing:
`/etc/WPEFramework/plugins/WebKitBrowser.json`

You should get a JSON that looks like:
{
 "locator":"libWPEFrameworkWebKitBrowser.so",
 "classname":"WebKitBrowser",
 "precondition":[
  "Graphics",
  "WebSource"
 ],
 "autostart":true,
 "configuration":{
  "url":"file:///var/www/index.html",
  "useragent":"Mozilla/5.0 (Macintosh, Intel Mac OS X 10_11_4) AppleWebKit/602.1.28+ (KHTML, like Gecko) Version/9.1 Safari/601.5.17",
  "injectedbundle":"libWPEInjectedBundle.so",
  "transparent":false,
  "compositor":"noaa",
  "inspector":"0.0.0.0:9998",
  "fps":true,
  "cursor":false,
  "touch":false,
  "msebuffers":"audio:2m,video:15m,text:1m",
  "memoryprofile":"128m",
  "memorypressure":"databaseprocess:50m,networkprocess:100m,webprocess:300m,rpcprocess:50m",
  "mediadiskcache":false,
  "diskcache":0,
  "xhrcache":false
 }
}
```

Update the url value and you should be set (restart WPEFramework obviously).

For 2. 170k? Measuring what, the rpi-sdimg or something else?

gil0109, 2018-06-26:
@wouterlucas For 2:  Its the rpi-sdimg.

BatslyAdams, 2018-06-26:
Hi! Let me know if I should split this into a different issue but I thought it belonged here since we're all currently on the same build. After building with the local.conf posted earlier on master I'm receiving the following error/crash on URLs I've been setting up for testing (specifically ones with video or maybe webaudio).

`OMX-Cannot open OpenMAX registry file /home/root/WebKitBrowser//.omxregister
OMX-A Component loader constructor fails. Exiting`

This crashes WPEFramework, so to try and troubleshoot I ran `/usr/bin/omxregister-bellagio` and copied the resultant `/home/root/.omxregister` to `/home/root/WebKitBrowser/.omxregister` and now some pages load, but I'm still unable to render video.

`OMX-Component not found with current ST static component loader.`

I'm also get this following error on new page loads:
`Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.
Cannot create EGL WPE context: EGL_SUCCESS`
but this seems to happen on every page, not sure if this is normal behavior but I thought I'd mention it just in case.

gil0109, 2018-06-26:
@BatslyAdams Please try this local.conf entries:

MACHINE = "raspberrypi3"
# For WPE local.conf changes
MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
WPE_BACKEND_rpi = "rpi"
WPE_PLATFORM = "egl"
DISTRO_FEATURES_remove = "wayland"
# END WPE

And report your results.

BatslyAdams, 2018-06-26:
Same error:
`OMX-Cannot open OpenMAX registry file /home/root/WebKitBrowser//.omxregister
OMX-A Component loader constructor fails. Exiting
ERROR: WebProcess crashed: exiting ...`
Test URL:
https://html5-video-crash.netlify.com/

wouterlucas, 2018-06-26:
On the size, I looked at this a while ago and our image is slightly bigger because of a few things:
* We include the kernel-modules (https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-core/images/wpe-image.bb#L17) which adds about 25MB uncompressed
* We include package management (https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-core/images/wpe-image.bb#L14), which adds a little. But I use this a lot, I do not flash my RPIs all the time. Instead I rely on OPKG or DEB to provide me with updated packages at runtime with new builds
* We include ssh, which is helpful when testing/debugging.
* The igalia layer only comes with a small wrapper the launched WPEWebKit, we provide an entire Framework with plugins for a lot more functionality (webserver, webproxy, DIALServer, WiFi support, etc). This adds a little more space
* I tend to run with `IMAGE_FEATURES += "tools-debug"` for debug tools. 
* And I usually set my RPI SD image to be around 160MB, after that I rely on 96board tools to expand it consume the entire SD card. Which in my case varies from 4GB to 32GB cards (not expanding will take a long time on 32GB :) ).

If I turn the above off and reduce WPEFramework to a minimum I can get to a pretty small image. But since you're running on an RPI2 or 3, there is no real need to do so. If you are running with a device with severe memory restrictions best is to make a project specific image configuration that reduces the feature set to a minimum that you would require.

wouterlucas, 2018-06-26:
@BatslyAdams best is to track this in a separate issue. Please include your build config, local config and machine you are running it on.

gil0109, 2018-06-26:
@wouterlucas Thanks for the explanation.  My plan is to add meta-mender to manage 85 of these devices and patching.  So the smaller the image the better. 

I changed the URL as you suggested in the json, but when I start WPEFramework again, it takes me to the Hello page.

Suggestions?

gil0109, 2018-06-26:
@wouterlucas Thanks for the explanation.  I loaded my digital signage up but the portion that displays an MP4 fails to load.

Is there log files to find out why this is the case?

wouterlucas, 2018-06-26:
I've added the Hello index.html page fix too. As for MP4 playback, not sure would need to check. 

Note that on master you are defaulting to WPE 2.20, there might be issues with that. If playback is of major concern I'd suggest switching to rocko as that supports wpe-20170728. Though I'll gladly poke around with master and see if we're missing any gstreamer patches/switches.

gil0109, 2018-06-26:
Thanks.  I will give that a shot.  I think we can happily close this issue, if you agree.  Again, thanks so much for your time on this.  I appreciate working and solving problems together.

wouterlucas, 2018-06-26:
Yep, closing it off. And likewise, thanks for your due diligence!

closed 2018-06-26 by wouterlucas

cchilumbu, 2018-07-04:
I have followed this thread with interest. 

After building 'wpe-eglfs-image', browser is not launched after boot, even though WEP processes are running. 

Do I have to have Firefox or Chromium installed as well, perhaps via meta-browser?

How can I launch browser from command line?

Thank you

cchilumbu, 2018-07-04:
Thank you for your reply Karim,

It certainly does help, and I am slowly getting the concepts.

Is it necessary to specify this in 'conf/local.conf':
DISTRO_FEATURES_remove = "wayland"

I ask because the same statement appears in the recipe 'wpe-eglfs-image.bb':
DISTRO_FEATURES_remove = "wayland"

If I specify this in local.conf, my build fails. 

All my layer branches are pyro, and I am working on an x86 platform.

My complete local.conf is shown below:
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_remove = "x11"
# DISTRO_FEATURES_remove = "wayland"
WPE_PLATFORM = "egl"
WPE_BACKEND = "mesa libinput"
ENABLE_UART = "1"
HOSTTOOLS_NONFATAL_append = " ssh"
CONF_VERSION = "1"

I am not sure if I WPE_BACKEND is correct or not as specified.

Thank you in advance.

kraj, 2018-07-04:
DISTRO_FEATURES is a global config namespace scope variable so it can be only used in config metadata like local.conf, machine config and distro config etc. Using it in recipe namespace as say in wpe-eglfs-image.bb is not going to work and therefore is not correct use of it.

You should describe the error in detail that you see when you remove wayland from distro features

Secondly, WPE_BACKEND depends upon the GL backend being used so when you want EGLFS and no x11 or wayland then it should be set appropriately as well.

e.g. for rpi when not using wayland you want to set it as below

```
WPE_BACKEND = "rpi"
```

gil0109, 2018-07-04:
@cchilumbu I am sorry, I have only tested on Raspberry Pi 3 and Sumo.  Not sure what you would need to do to get it working on x86.

cchilumbu, 2018-07-04:
Thank you guys,

Let me get the build error I am getting when I remove wayland in local.conf:

Just to clarify, would it be correct to set as show below:
WPE_PLATFORM = "eglfs"
WPE_BACKEND = "eglfs"

Thanks, and I will report back the build error...

cchilumbu, 2018-07-04:
Here is my build failure:

Checking for module 'gstreamer-1.0 >= 1.2.3'
| --   Found gstreamer-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-base-1.0 >= 1.2.3'
| --   Found gstreamer-base-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-app-1.0 >= 1.2.3'
| --   Found gstreamer-app-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-audio-1.0 >= 1.2.3'
| --   Found gstreamer-audio-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-fft-1.0 >= 1.2.3'
| --   Found gstreamer-fft-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-gl-1.0 >= 1.2.3'
| --
| -- Checking for module 'gstreamer-mpegts-1.0>=1.4.0'
| --   Found gstreamer-mpegts-1.0, version 1.10.4
| -- Checking for module 'gstreamer-pbutils-1.0 >= 1.2.3'
| --   Found gstreamer-pbutils-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-tag-1.0 >= 1.2.3'
| --   Found gstreamer-tag-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-video-1.0 >= 1.2.3'
| --   Found gstreamer-video-1.0 , version 1.10.4
| -- Found GStreamer: GSTREAMER_INCLUDE_DIRS;GSTREAMER_LIBRARIES;GSTREAMER_VERSION;GSTREAMER_BASE_INCLUDE_DIRS;GSTREAMER_BASE_LIBRARIES;GSTREAMER_APP_INCLUDE_DIRS;GSTREAMER_APP_LIBRARIES;GSTREAMER_PBUTILS_INCLUDE_DIRS;GSTREAMER_PBUTILS_LIBRARIES;GSTREAMER_VIDEO_INCLUDE_DIRS;GSTREAMER_VIDEO_LIBRARIES;GSTREAMER_MPEGTS_INCLUDE_DIRS;GSTREAMER_MPEGTS_LIBRARIES;GSTREAMER_TAG_INCLUDE_DIRS;GSTREAMER_TAG_LIBRARIES;GSTREAMER_GL_INCLUDE_DIRS;GSTREAMER_GL_LIBRARIES;GSTREAMER_AUDIO_INCLUDE_DIRS;GSTREAMER_AUDIO_LIBRARIES;GSTREAMER_FFT_INCLUDE_DIRS;GSTREAMER_FFT_LIBRARIES (Required is at least version "1.2.3")
| CMake Error at Source/cmake/GStreamerChecks.cmake:41 (message):
|   GStreamerGL is needed for USE_GSTREAMER_GL.
| Call Stack (most recent call first):
|   Source/cmake/OptionsWPE.cmake:209 (include)
|   Source/cmake/WebKitCommon.cmake:59 (include)
|   CMakeLists.txt:150 (include)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/wpewebkit/20170728+gitAUTOINC+d2f97f9992-r1/build/CMakeFiles/CMakeOutput.log".
| See also "/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/wpewebkit/20170728+gitAUTOINC+d2f97f9992-r1/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/wpewebkit/20170728+gitAUTOINC+d2f97f9992-r1/temp/log.do_configure.421)
ERROR: Task (/home/cchilumbu/projects/yocto-dr0081/sources/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_20170728.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4239 tasks of which 4238 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/cchilumbu/projects/yocto-dr0081/sources/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_20170728.bb:do_configure

wouterlucas, 2018-07-05:
I've never built it for X86 using this repo, but it should technically be possible. You'd want to set the `WPE_BACKEND = "wayland"` and use the weston compositor. Best is to use the `core-image-weston` image and include the wpeframework / wpewebkit packagegroups from there.

I don't think we'd need any specific platform in WPE_PLATFORM, try making it empty. Or we may need to adjust the recipe a bit to avoid a platform being selected at all in `wpewebkit`.

[5 more replies]