WebPlatformForEmbedded/meta-wpe #215: (sumo build break on RPI) wpewebkit_2.20.bb recipe in sumo branch depends on gstreamer 1.14.0

ghuser0, 2018-06-19:
Hello,

I am experiencing a build error when trying to compile wpewebkit (via meta-wpe) in sumo for the Raspberry Pi 3.

I think the problem is that the sumo branch of meta-wpe includes this commit "wpewebkit: bump and adjust rdepends for Gstreamer 1.14.0":
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/8fb6dff9ede52e48350d36d2ae51085acf56f54a

Whereas the sumo branch of poky includes gstreamer 1.12.4:
http://git.yoctoproject.org/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer?h=sumo

and the recipe gstreamer1.0-plugins-base_1.12.4.bb does not include opengl:
http://git.yoctoproject.org/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base.inc?h=sumo

(opengl at this version is included in gstreamer1.0-plugins-ugly_1.12.4.bb):
http://git.yoctoproject.org/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad.inc?h=sumo

There seem to be lots of other commits in the sumo branch reverting some of the upgrade to 1.14.0:
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/654e302fd022ab5c515cbf0024102f6f345a7c4b
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/e68ad28c8f290fe2ed0dbe8218d1fdda69a10b89
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/f0bc8b346fd60a685abfb0bcddc71df6636fe119
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/34485b1c13fd30c028b125dbe8dad2f99ae678d4
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/20b9bfc1829b35f641e8b13476f8f38834708097
https://github.com/WebPlatformForEmbedded/meta-wpe/commit/2f174c315c6e213126ea523394f0f1dbc30acdee

but not this one and the dependency on gstreamer1.0-plugins-base-opengl remains:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/119ee294cb9a7135d537d4dad13cd5b0ddf8a8f6/recipes-wpe/wpewebkit/wpewebkit_2.20.bb#L176

The configuration I'm using in my distro.conf to bring in WPE is:
# Setup WPE including Wayland/Westeros
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl egl wayland wpeframework"
# Temporarily use different westeros revision while meta-wpe doesn't build with what it uses
WESTEROS_SRCREV = "8ec4db488273a81720934bebe0ab27ab511cc249"
# Temporarily use different wpewebkit revision while meta-wpe doesn't build with what it uses
SRCREV_pn-wpewebkit = "f2de8d0bbba9f6cf14d3ade0057c6cebfdadec35"
BASE_URI_pn-wpewebkit = "git://github.com/WebPlatformForEmbedded/WPEWebKit.git;protocol=git;branch=stable"

During my build, I see the following warnings:
WARNING: wpewebkit-2.20+gitAUTOINC+f2de8d0bbb-r1 do_package_qa: QA Issue: wpewebkit rdepends on gstreamer1.0-plugins-base-opengl, but it isn't a build dependency? [build-deps]
WARNING: wpewebkit-2.20+gitAUTOINC+f2de8d0bbb-r1 do_package_qa: QA Issue: wpewebkit rdepends on gstreamer1.0-plugins-base-mpg123, but it isn't a build dependency? [build-deps]

And then I get the error:
The following packages have unmet dependencies:
 packagegroup-wpeframework : Depends: wpeframework-plugins but it is not going to be installed
 packagegroup-wpewebkit : Depends: wpewebkit but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

(this is the error with DEB style packages - I get another error with RPM style packages)

I am going to try and do some patching later today or tomorrow.

wouterlucas, 2018-06-20:
Ugh, okay. This is due to my merges from master into sumo, I must have overlooked the Gstreamer bumps. 

Once I get some time I'll circle back to sumo and revert the Gstreamer related bumps from master into sumo.

wouterlucas, 2018-06-20:
@ghuser0 please try again with latest revert on wpewebkit, this should put the RDEPS back to Gstreamer 1.12.x.

ghuser0, 2018-06-20:
Thanks very much for having a look at this.  Pulling and rebuilding I get the same error I wrote above when I tried to patch it out myself in my own layer:

"ERROR: Nothing RPROVIDES 'gstreamer1.0-plugins-ugly-mpg123' (but /home/user/poky/meta-wpe-recipes-wpe/wpewebkit/wpewebkit_2.20.bb RDEPENDS on or otherwise requires it)

ERROR: Required build target 'my-image' has not buildable providers.
Missing or unbuildable dependency chain was: ['my-image', 'packagegroup-wpewebkit', 'wpebackend', 'wpebackend-rdk', 'wpewebkit', 'gstreamer1.0-plugins-ugly-mpg123']

But again, that doesn't make sense to me because mpg123 should be in the ugly package and should be configured to build by default in that package's PACKAGECONFIG.

I'm going to try to blow out my tmp, cache, and sstate-cache and try again, so I'll comment back when that's had a chance to run.

wouterlucas, 2018-06-20:
That's odd. As far as I can see poky provides mpg123 as part of Gstreamer1.0-plugins-ugly:
https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly.inc?h=sumo#n22

We enable it through a bbappend:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/sumo/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly_1.12.4.bbappend#L8

And depend on it (now correctly for Sumo):
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/sumo/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly_1.12.4.bbappend#L8

Even prior to my wpewebkit revert your local build should have that enabled and available. Unless you have other changes locally?

ghuser0, 2018-06-20:
I agree it doesn't make sense to me!  And I don't even know if you need to enable it through bbappend (although probably helps) since it's enabled by default:
http://git.yoctoproject.org/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly.inc?h=sumo#n12

I don't see anything in my layer, image, distro, or local config that would cause this but I'm continuing to look on my end....

ghuser0, 2018-06-20:
Hi, getting gstreamer1.0-plugins-ugly built for me required setting
LICENSE_FLAGS_WHITELIST = "commercial_gstreamer1.0-plugins-ugly" in my local.conf.
It's building and running now - thank you!

It didn't seem to require it, but I was also wondering if "gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly" should be a part of the mediasource, video, and webaudio PACKAGECONFIG variables, since they will all include RDEPS on all of those packages via RDEPS_MEDIASOURCE, RDEPS_VIDEO, and RDEPS_WEBAUDIO.

Thanks again for the help.

wouterlucas, 2018-06-20:
Great! Glad it worked out. Thats an interesting change though, I would have expect this value to work accordingly in the layer.conf:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/sumo/conf/layer.conf#L33

They are set as RDEPS in the PACKAGECONFIG variables, e.g. video rdeps ${RDEPS_VIDEO}:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/sumo/recipes-wpe/wpewebkit/wpewebkit_2.20.bb#L68

Which translates to:
https://github.com/WebPlatformForEmbedded/meta-wpe/blob/sumo/recipes-wpe/wpewebkit/wpewebkit_2.20.bb#L135-L139

Plus it has the complete packages itself in the build dependency. Closing this one.

closed 2018-06-20 by wouterlucas

ghuser0, 2018-06-20:
Okay, thanks!