WebPlatformForEmbedded/meta-wpe #217: Missing WPE Launcher

cchilumbu, 2018-07-04:
Hello,

After building 'wpe-eglfs-image', browser is not launched after boot, even though WEP processes are running.

My conf/local.conf looks like:
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_remove = "x11"
WPE_PLATFORM = "egl

All my layers are branch 'pyro'. Target is x86 up-board.

I do not have WPE Launcher anywhere on my target, what do I need to do to get it into my rootfs?

Do I have to have Firefox or Chromium installed as well, perhaps via meta-browser?

How can I launch browser from command line?

Thank you

psivasubramanian, 2018-07-04:
Hi @cchilumbu 
With latest mete-wpe revision, wpeframework.service is supposed to start WPE browser in the startup. can you check "systemctl status wpeframework -l" or "journctl -u wpeframework"?

cchilumbu, 2018-07-04:
I am now trying to install systemctl onto my target....

However, when I run 'top' command on the target, I can see :
/usr/bin/WPEFramework -b     

I can also see WPEProcess 

In my /var/log/messages, I can see repeated messages:
Startup: Activated plugin [WebKitBrowser]
Shutdown: Deactivated plugin [WebKitBrowser]

A few seconds later, I get a segmentation fault:
WPEProcess segfault at 0  ....  error 4 in libWPEBackend.so.1

All my layer branches are pyro, and I am working on an x86 platform.

My complete local.conf is shown below:
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_remove = "wayland"

WPE_PLATFORM = "egl"
WPE_BACKEND = "mesa libinput"
ENABLE_UART = "1"
HOSTTOOLS_NONFATAL_append = " ssh"
CONF_VERSION = "1"

Any ideas please?

Thank you

cchilumbu, 2018-07-04:
Build failure is shown below:

hecking for module 'gstreamer-1.0 >= 1.2.3'
| --   Found gstreamer-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-base-1.0 >= 1.2.3'
| --   Found gstreamer-base-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-app-1.0 >= 1.2.3'
| --   Found gstreamer-app-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-audio-1.0 >= 1.2.3'
| --   Found gstreamer-audio-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-fft-1.0 >= 1.2.3'
| --   Found gstreamer-fft-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-gl-1.0 >= 1.2.3'
| --
| -- Checking for module 'gstreamer-mpegts-1.0>=1.4.0'
| --   Found gstreamer-mpegts-1.0, version 1.10.4
| -- Checking for module 'gstreamer-pbutils-1.0 >= 1.2.3'
| --   Found gstreamer-pbutils-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-tag-1.0 >= 1.2.3'
| --   Found gstreamer-tag-1.0 , version 1.10.4
| -- Checking for module 'gstreamer-video-1.0 >= 1.2.3'
| --   Found gstreamer-video-1.0 , version 1.10.4
| -- Found GStreamer: GSTREAMER_INCLUDE_DIRS;GSTREAMER_LIBRARIES;GSTREAMER_VERSION;GSTREAMER_BASE_INCLUDE_DIRS;GSTREAMER_BASE_LIBRARIES;GSTREAMER_APP_INCLUDE_DIRS;GSTREAMER_APP_LIBRARIES;GSTREAMER_PBUTILS_INCLUDE_DIRS;GSTREAMER_PBUTILS_LIBRARIES;GSTREAMER_VIDEO_INCLUDE_DIRS;GSTREAMER_VIDEO_LIBRARIES;GSTREAMER_MPEGTS_INCLUDE_DIRS;GSTREAMER_MPEGTS_LIBRARIES;GSTREAMER_TAG_INCLUDE_DIRS;GSTREAMER_TAG_LIBRARIES;GSTREAMER_GL_INCLUDE_DIRS;GSTREAMER_GL_LIBRARIES;GSTREAMER_AUDIO_INCLUDE_DIRS;GSTREAMER_AUDIO_LIBRARIES;GSTREAMER_FFT_INCLUDE_DIRS;GSTREAMER_FFT_LIBRARIES (Required is at least version "1.2.3")
| CMake Error at Source/cmake/GStreamerChecks.cmake:41 (message):
|   GStreamerGL is needed for USE_GSTREAMER_GL.
| Call Stack (most recent call first):
|   Source/cmake/OptionsWPE.cmake:209 (include)
|   Source/cmake/WebKitCommon.cmake:59 (include)
|   CMakeLists.txt:150 (include)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/wpewebkit/20170728+gitAUTOINC+d2f97f9992-r1/build/CMakeFiles/CMakeOutput.log".
| See also "/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/wpewebkit/20170728+gitAUTOINC+d2f97f9992-r1/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/wpewebkit/20170728+gitAUTOINC+d2f97f9992-r1/temp/log.do_configure.421)
ERROR: Task (/home/cchilumbu/projects/yocto-dr0081/sources/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_20170728.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4239 tasks of which 4238 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/cchilumbu/projects/yocto-dr0081/sources/meta-wpe/recipes-wpe/wpewebkit/wpewebkit_20170728.bb:do_configure

samouds, 2018-07-06:
I have a similar problem, when I launch the WPEFramework I get the default page ( https://github.com/WebPlatformForEmbedded/meta-wpe/blob/13d683c52102080d8fde9d05fe2145eea62b3bc7/recipes-wpe/wpeframework/files/index.html ) with the animated squares, how to display another page ( youtube ) than the default one ?

pwielders, 2018-07-06:
2 possibilities. 
You can configure it to start with another start URL. Defined in the WebKitBrowser plugin configuration, to be found @ /etc/WPEFramework/plugins/WebKitBrower.json.
You can change it by navigating with a browser to the WPEFramework UI pages, accessible from http:<IP address of your device>. Here you will find in the left hand pane a section called WebKitBrowser, if you select it, you get the p[ossibility to set a custom URL or pick one from the drop down box.

samouds, 2018-07-06:
thanks @pwielders. But I'm running a headless RPi and I were using WPELauncher in a bash script to start with a specific url : WPELauncher http://www.youtube.com

Is there an equivalent command for WPEFramework since the WPELauncher was removed ?

wouterlucas, 2018-07-07:
You can also configure the URL you need at build time by setting:
```
WEBKITBROWSER_STARTURL  = "http://www.google.com"
```

In your local.conf, create a new image and you should be good to go. This will at build time set the file the Pierre mentioned above. 

For headless it should even be easier, just navigate to the IP of your PI with a browser from your desktop and you should get a control panel that allows you to set the URL remotely. The Hello page is created for you to be able to figure out what the IP of the RPI is.

kraj, 2018-07-07:
@wouterlucas this should be documented in README.md

wouterlucas, 2018-07-07:
Agree Khem, it's more a matter of time then willingness or seeing the need to update the docs.

closed 2018-07-07 by wouterlucas