WebPlatformForEmbedded/meta-wpe #224: Document how to get touch input for WPE on Raspberrypi (wpe-eglfs-image)

DWagenk, 2018-07-12:
Hi,
I've been looking around the WPE repos for ducumentation on how to get touch input for my system running.

Hardware:
- Raspberrpi-cm3
- Custom DPI-TFT
- Touch Controller via I2C

Software:
- Linux based on poky distro with meta-wpe layer -> building wpe-eglfs-image
- custom layer (configuring DPI-TFT and Touch driver)

I've been stuck having the touch driver working with libinput, but not getting any reaction to touch events in WPEWebKitBrowser.

```
# libinput debug-events
-event0   DEVICE_ADDED     XYZ-TouchScreen     seat0 default group1  cap:t calib
 event0   TOUCH_DOWN        +1.05s      0 (0) 50.52/47.98 (243.00/154.00mm)
 event0   TOUCH_FRAME       +1.05s
 event0   TOUCH_UP          +1.07s
 event0   TOUCH_FRAME       +1.07s
 event0   TOUCH_DOWN        +1.43s      0 (0) 74.01/35.83 (356.00/115.00mm)
 event0   TOUCH_FRAME       +1.43s
 event0   TOUCH_UP          +1.56s
 event0   TOUCH_FRAME       +1.56s
 event0   TOUCH_DOWN        +1.93s      0 (0) 14.55/52.02 (70.00/167.00mm)
 event0   TOUCH_FRAME       +1.93s
 event0   TOUCH_UP          +1.97s
 event0   TOUCH_FRAME       +1.97s
```

In WPEBackend-rdk/src/input/Libinput/LibinputServer.cpp I found that having USE_VIRTUAL_KEYBOARD=ON (PACKAGECONFIG: virtualinput) set changes a lot of how the input handling works. virtualinput was Part of the default PACKAGECONFIG flags.
Now I removed it and touch-input for WPEWebKitBrowser is working.

Relevant parts of my local.conf:
```
WPE_BACKEND = "rpi libinput"
PACKAGECONFIG_pn-wpebackend-rdk = "${WPE_BACKEND" #to remove virtualinput from PACKAGECONFIG
WPE_PLATFORM = "egl touch"
NETFLIX_BACKEND = "rpi"
DISTRO_FEATURES_remove = "wayland"
```

I found that the env variable WPE_BCMRPI_TOUCH must be set for bcm-rpi view-backend to support touch input, so if I add `export WPE_BCMRPI_TOUCH=true` to /etc/init.d/wpeframework script before it starts /usr/bin/WPEFramework touch input finally works.

All this feels more like a hack. Especially since I need an OnScreenKeyboard for input, which probably is exactly what PACKAGECONFIG flag virtualinput is meant for, right?

So I'd be very glad to get an explanation, of wich flags to set to have touch input via libinput work in a less hacky way.
Anyone here to help?
Daniel

wouterlucas, 2018-07-12:
You can set touch to true in `/etc/WPEFramework/plugins/WebKitBrowser.json` to enable that. You do not need to set the export if you.

I was unaware the virtualinput had that effect. You would only need virtual input if you want to forward keys from the webapi towards the device.

I added some more details on the wiki for this, hopefully its useful to you:
https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/WPEFramework#configuration

adamtorgerson, 2018-10-24:
I also had to remove virtualinput from wpebackend-rdk (on the sumo branch) to get touch input working. Seems to work fine after that.

closed 2018-11-02 by wouterlucas

elcritch, 2019-07-02:
I just spent a few hours tracking down why `cog` wasn't finding input devices. It's on a buildroot on an RPi3 with the official touchscreen and a custom init system. Adding `WPE_BCMRPI_TOUCH=true` didn't work and `libinput list-devices` didn't show anything despite `/dev/input/event0` existing. 

Finally it turned out that `udevd` needs to be started, followed by `/usr/bin/udevadm trigger` then starting `cog` with `WPE_BCMRPI_TOUCH=1` set works. Thought I'd post that here in case others run into similar issues on buildroot systems without an init system.

d21d3q, 2019-09-30:
> You can set touch to true in `/etc/WPEFramework/plugins/WebKitBrowser.json` to enable that. You do not need to set the export if you do.
Is it possible to have it set to true by default?

wouterlucas, 2019-10-01:
@d21d3q I don't mind adding that as _rpi specific install, since this will only be used on RPI devices.

Though probably the more elegant solution would be to extend wpeframework-plugins cmake and handle it through a package config toggle (which in turn should be RPI specific as well).