WebPlatformForEmbedded/meta-wpe #225: Q: why settings in local.conf are not in image recipes?

hhromic, 2018-07-12:
Hello, I have been wondering for a while why some "required" variables in `local.conf`, as stated in the Wiki for Raspberry Pi, are not included in the respective image recipes?

Take the case of Westeros builds for RPI for example, where it is required that:
````
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl"
````
Why these are not set in the image recipe `wpe-westeros-image.bb`? Actually, the image recipe includes a check for the `opengl` package:
```
REQUIRED_DISTRO_FEATURES = "wayland opengl"
```
So why not just add the required feature automatically instead of just checking?

hhromic, 2018-07-12:
Likewise for the EGLFS image recipe, it is requested in the Wiki to use this in `local.conf`:
```
WPE_BACKEND = "rpi libinput"
WPE_PLATFORM = "egl"
NETFLIX_BACKEND = "rpi"
DISTRO_FEATURES_remove = "wayland"
```
Should all that be in the image recipe for EGLFS too automatically?

wouterlucas, 2018-07-12:
I tried that a long time ago and they weren't picked up, hence I just sticked to `local.conf`. If they do work from the image recipes I'm all for moving them there and make starting out with this project as easily as possible. 

Did you try that?

hhromic, 2018-07-12:
I see, no I haven't tried but I can experiment and provide feedback. As you know I'm also all in for the best out-of-the-box experience. Will report back when I have a moment to try. Thanks for the answer!

hhromic, 2018-07-12:
I did now some quick tests and seems to be working fine in recent Yocto :)
I tested some changes to try to cover both image types, see below.

Changes in `wpe-eglfs-image.bb` (for the **EGLFS** image), which ensure `wayland` is removed: 
```patch
-CONFLICT_DISTRO_FEATURES = "wayland x11"
+DISTRO_FEATURES_remove = "wayland"

-REQUIRED_DISTRO_FEATURES = "opengl"
+WPE_PLATFORM = "egl"
+WPE_BACKEND_rpi = "rpi libinput"
+NETFLIX_BACKEND_rpi = "rpi"
```
Changes in `wpe-westeros-image.bb` (for the **Westeros** image), which ensure `wayland` is present:
```patch
-REQUIRED_DISTRO_FEATURES = "wayland opengl"
+DISTRO_FEATURES_append = " wayland"
```
Changes in `wpe-image.inc` (common for **both** images), which ensure `opengl` is present and `x11` is removed:
```patch
+
+DISTRO_FEATURES_remove = "x11"
+DISTRO_FEATURES_append = " opengl"
```
All these according with the documented requirements and the current recipes. You will note that I un-did the recent `REQUIRED_DISTRO_FEATURES` and `CONFLICT_DISTRO_FEATURES` changes from @kraj, as these are unnecessary when setting the features explicitly as I'm doing.

I checked the output environment for both image types. Here is the **EGLFS** image environment:
```
# $NETFLIX_BACKEND
#   override[rpi]:set /home/yocto/meta-wpe/recipes-core/images/wpe-eglfs-image.bb:12
#     "rpi"
NETFLIX_BACKEND="rpi"
# $WPE_BACKEND
#   override[rpi]:set /home/yocto/meta-wpe/recipes-core/images/wpe-eglfs-image.bb:11
#     "rpi libinput"
WPE_BACKEND="rpi libinput"
#
# $WPE_BACKEND_rpi
#   set /home/yocto/meta-wpe/recipes-core/images/wpe-eglfs-image.bb:11
#     "rpi libinput"
WPE_BACKEND_rpi="rpi libinput"
#
# $WPE_PLATFORM
#   set /home/yocto/meta-wpe/recipes-core/images/wpe-eglfs-image.bb:10
#     "egl"
WPE_PLATFORM="egl"
# $DISTRO_FEATURES [9 operations]
#   _append /home/yocto/wpe/conf/local.conf:248
#     " systemd"
#   set? /home/yocto/poky/meta-poky/conf/distro/poky.conf:22
#     "${DISTRO_FEATURES_DEFAULT} ${DISTRO_FEATURES_LIBC} ${POKY_DEFAULT_DISTRO_FEATURES}"
#   set? /home/yocto/poky/meta/conf/distro/include/default-distrovars.inc:22
#     "${DISTRO_FEATURES_DEFAULT} ${DISTRO_FEATURES_LIBC}"
#   set /home/yocto/poky/meta/conf/documentation.conf:138
#     [doc] "The features enabled for the distribution."
#   set? /home/yocto/poky/meta/conf/bitbake.conf:802
#     ""
#   _remove /home/yocto/meta-wpe/recipes-core/images/wpe-image.inc:10
#     "x11"
#   _append /home/yocto/meta-wpe/recipes-core/images/wpe-image.inc:11
#     " opengl"
#   _remove /home/yocto/meta-wpe/recipes-core/images/wpe-eglfs-image.bb:8
#     "wayland"
#   append utils.py:116 [features_backfill]
#     " pulseaudio bluez5 gobject-introspection-data ldconfig"
# pre-expansion value:
#   "${DISTRO_FEATURES_DEFAULT} ${DISTRO_FEATURES_LIBC} ${POKY_DEFAULT_DISTRO_FEATURES} systemd opengl pulseaudio bluez5 gobject-introspection-data ldconfig"
DISTRO_FEATURES="acl alsa argp bluetooth ext2 irda largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc ipv4 ipv6 libc-backtrace libc-big-macros libc-bsd libc-cxx-tests libc-catgets libc-charsets libc-crypt libc-crypt-ufc libc-db-aliases libc-envz libc-fcvt libc-fmtmsg libc-fstab libc-ftraverse libc-getlogin libc-idn libc-inet-anl libc-libm libc-locales libc-locale-code libc-memusage libc-nsswitch libc-rcmd libc-rtld-debug libc-spawn libc-streams libc-utmp libc-utmpx libc-wordexp libc-posix-clang-wchar libc-posix-regexp libc-posix-regexp-glibc libc-posix-wchar-io largefile opengl ptest multiarch vulkan systemd opengl pulseaudio bluez5 gobject-introspection-data ldconfig"
```
And for the **Westeros** image environment:
```
# $DISTRO_FEATURES [9 operations]
#   _append /home/yocto/wpe/conf/local.conf:248
#     " systemd"
#   set? /home/yocto/poky/meta-poky/conf/distro/poky.conf:22
#     "${DISTRO_FEATURES_DEFAULT} ${DISTRO_FEATURES_LIBC} ${POKY_DEFAULT_DISTRO_FEATURES}"
#   set? /home/yocto/poky/meta/conf/distro/include/default-distrovars.inc:22
#     "${DISTRO_FEATURES_DEFAULT} ${DISTRO_FEATURES_LIBC}"
#   set /home/yocto/poky/meta/conf/documentation.conf:138
#     [doc] "The features enabled for the distribution."
#   set? /home/yocto/poky/meta/conf/bitbake.conf:802
#     ""
#   _remove /home/yocto/meta-wpe/recipes-core/images/wpe-image.inc:10
#     "x11"
#   _append /home/yocto/meta-wpe/recipes-core/images/wpe-image.inc:11
#     " opengl"
#   _append /home/yocto/meta-wpe/recipes-core/images/wpe-westeros-image.bb:11
#     " wayland"
#   append utils.py:116 [features_backfill]
#     " pulseaudio bluez5 gobject-introspection-data ldconfig"
# pre-expansion value:
#   "${DISTRO_FEATURES_DEFAULT} ${DISTRO_FEATURES_LIBC} ${POKY_DEFAULT_DISTRO_FEATURES} systemd opengl wayland pulseaudio bluez5 gobject-introspection-data ldconfig"
DISTRO_FEATURES="acl alsa argp bluetooth ext2 irda largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc ipv4 ipv6 libc-backtrace libc-big-macros libc-bsd libc-cxx-tests libc-catgets libc-charsets libc-crypt libc-crypt-ufc libc-db-aliases libc-envz libc-fcvt libc-fmtmsg libc-fstab libc-ftraverse libc-getlogin libc-idn libc-inet-anl libc-libm libc-locales libc-locale-code libc-memusage libc-nsswitch libc-rcmd libc-rtld-debug libc-spawn libc-streams libc-utmp libc-utmpx libc-wordexp libc-posix-clang-wchar libc-posix-regexp libc-posix-regexp-glibc libc-posix-wchar-io largefile opengl ptest multiarch wayland vulkan systemd opengl wayland pulseaudio bluez5 gobject-introspection-data ldconfig"
```

As you can see, the variables seem to be set correctly from the image recipes. What do you think @wouterlucas , @kraj ?  If you like these changes I can prepare the PR. Also open for input on improvements.

hhromic, 2018-07-12:
One more quick question. For the case of `WPE_BACKEND_rpi = "rpi libinput"`,  is `libinput` required only for `rpi` or for any machine type?

wouterlucas, 2018-07-13:
I dont think you need to set the WPE BACKEND anymore for RPI with the current state of wpebackend-rdk. 

And libinput should be included for any target that will do EGLFS. For any Wayland based (Westeros, Weston or WPEFramework Compositor) backend they do not require libinput.

kraj, 2018-07-13:
@hhromic what are you trying to achieve here ? FWIW your change is not correct. variables in local.conf and others are configuration metadata which applies globally, this data can not be and should not be edited per recipe and even if you do, it wont be effective. REQUIRE_DISTRO_FEATURE anf CONFLICT_DISTRO_FEATURES are in recipe namespace data. Therefore it will use the global data to ensure that we have correct distro features to build the image correctly, note that it just checks and does not modify.

hhromic, 2018-07-13:
@kraj, short read: I'm trying to clean-up the layer and improve end-user friendliness.

The long read: I understand the scope of metadata in `local.conf` versus the recipes, however these variables I'm trying to incorporate in the recipes are not really user configurable as far as I understand.
In the ideal world, an end-user should be able to:
1. clone the layer,
2. add the necessary dependencies,
3. create a minimum `local.conf` file with the desired `MACHINE` value, and then
4. just run `bitbake wpe-eglfs-image` or `bitbake wpe-westeros-image` to get a base standard and functional image.

However, currently, the user is asked to set variables in `local.conf` such as `WPE_BACKEND` or `DISTRO_FEATURES` depending on the desired image, that in my opinion should be automatically set in the respective image recipe (eglfs, westeros). If the user will have always to set e.g. `WPE_BAKEND = "egl"` when building `wpe-eglfs-image`, why not add that statically in the image recipe? Is not really optional or changeable for that particular image.

Another example, `opengl` is required no matter what for eglfs and westeros images (according to current instructions), then why not set this statically in their respective image recipes? The values of these variables are not meant to be user editable/changeable. Are they?

That is my main motivation with this issue/question. All this being said, I don't believe the issue is invalid, we are perhaps not understanding each other :) What do you think?

kraj, 2018-07-13:
DISTRO_FEATURES have wider scope than images. So they can not be manipulated inside a recipe since images are in recipe namespace. One can argue if x11 and wayland should be distro features or not but they have impact on multiple recipes how they are built based on these being in distro. 

using EGLFS or wayland are going to result in change in several packages based upon whats used
therefore its a combination of a global knob whatever you want to call it

you are viewing it from image point of view, unfortunately thats not the right vantage.

wouterlucas, 2018-07-16:
The only way to make it easier for "standard" RPI builds with Poky/OE-core is to provide a little bootstrapper script and redirect new users to that. Similar to the one we have for the buildbot, but then provide the user with an option to select Wayland or EGLFS before initing/creating the configuration.

That still makes it controllable and less hack-ish then editing a local.conf file whilst still keeping the ability to integrate the wpe images into different setups and layers. How about that?

hhromic, 2018-07-16:
Ummm. In that case I would still prefer to encourage the end-users to understand what they are doing with their `local.conf` files more than providing a script that obfuscates the inner workings of the build. Therefore I think that engineering the recipes to be smarter is the way to go, like @kraj did with the REQUIRE_DISTRO_FEATURES and CONFLICT_DISTRO_FEATURES assertions. These tell the users what is wrong and helps them learn and understand better their builds.
Another good example of good engineering towards this is the auto-selection of WPE_BACKEND and WPE_PLATFORM in the WPE recipe based on the distro features, so the user doesn't have to set them manually (but still advanced users can override it). We should totally aim on improving these techniques.

My original attempt was to find an elegant solution based on standard Yocto practices to pre-configure the distro features, however I realised now that it is not the way to think about it :)

hhromic, 2018-07-17:
Sorry to bother again with this guys. I noticed that the [instructions for RPI](https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/Raspberry-PI) for Westeros builds require the `x11` distro feature to be removed:
```
Remove x11 from Distro features since we use wayland+westeros
DISTRO_FEATURES_remove = "x11"
Add opengl to Distro Features
DISTRO_FEATURES_append = " opengl"
```
However the `wpe-westeros-image.bb` image recipe doesn't check for the x11 conflict, only that the required `wayland opengl` features are present:
```
REQUIRED_DISTRO_FEATURES = "wayland opengl"
```
In `wpe-eglfs-image.bb`, the recipe does handly the "conflict" and "required" properly:
```
CONFLICT_DISTRO_FEATURES = "wayland x11"
REQUIRED_DISTRO_FEATURES = "opengl"
```
@kraj , should the westeros recipe also have a `CONFLICT_DISTRO_FEATURES = "x11"` line to be consistent?
Also, `opengl` seems to be a common required feature, should it be in the base `wpe-image.bb` file instead?
I can prepare a PR with these refinements if you agree with these. Thanks for the patience :)

hhromic, 2018-07-17:
Ah, and is `x11` actually conflicting in all cases? which means also should go to `wpe-image.bb`?
I ask all these questions because I don't know how you build for other machines than RPI =)

wouterlucas, 2018-07-17:
Yep! It is for the most common use case for meta-wpe, we tend to run full screen EGL OR using Wayland (which in turn mostly does full screen EGL too). Neither of the common configs use X11 =)

kraj, 2018-07-17:
its ok to check for x11 disabled.

hhromic, 2018-07-19:
Thanks both for the help and explanations. Learning a lot.
I'm going to close this for now. Cheers!

closed 2018-07-19 by hhromic