WebPlatformForEmbedded/meta-wpe #231: WPE browser displaying content in smaller resolution with Westeros

psivasubramanian, 2018-07-31:
Issue:
WPE browser (WPEFramework) displaying content in smaller resolution (not fullscreen) with Westeros backend. This is observed on iMX8 and Rpi3 boards with latest meta-wpe:morty revisions like 
iMX8 - https://github.com/WebPlatformForEmbedded/meta-wpe/commit/064a42453c5e343c4e409616157eb958d9090f08 
Rpi3 - https://github.com/WebPlatformForEmbedded/meta-wpe/commit/7b3ff1e0daf8e540b8dbeea0751c8d346618bfc5

Last recent working setup reference:
This revision https://github.com/WebPlatformForEmbedded/meta-wpe/commit/9edaccbf7d51a8099f501abf8c09e0c73d1f911f is the last recent working WPE we have on HiKey platform where this issue is not observed.

Root cause:
WPE browser displays the content with the default WebCreation resolution 800x600 only as widget::resize() is not happening as IPC message "DrawingArea:UpdateBackingStoreState" is not at all received by WebProcess::didReceiveMessage(). The DrawingArea:UpdateBackingStoreState() call flow is responsible for resizing the WebPage object with full 1280x720 resolution as  

sizeForVisibleContent 249 width 800 height 600
Web process message 'WebPage:LoadRequest'
Web process message 'WebPage:SetActivityState'
**Web process message 'DrawingArea:UpdateBackingStoreState'**
WebPage::setSize 1397
WebPage::setSize 1401
sizeForVisibleContent 249 width 1280 height 720

But this is not happening on both the revisions i.e. May 2018 & Jul 2018 we tested recently.

FYI @kraj.
@pavannandyala observed this on Rpi3 as mentioned above.

pavannandyala, 2018-08-06:
Anyone validated wpeframework  on RPI platform (morty branch)

wouterlucas, 2018-08-06:
I did recently, had to add a patch to fix userland dependency with libepoxy. But that has been resolved, right now I can bring up WPEFramework on Morty, graphics works but the gstreamer pipeline complains about omx errors. 

I did not have any more time to dive into the omx errors with morty on RPI.

psivasubramanian, 2018-08-14:
I can confirm this issue on Rpi3 also with Morty based wpe-westeros-image. Attached the photos of the same as below

[image]

[image]

Even the key events are not going to WPE, but only to terminal same as iMX8.

TomMettam, 2018-09-01:
Confirmed also.

wouterlucas, 2018-09-01:
You're probably better off trying the eglfs image instead.

wouterlucas, 2019-07-26:
By default Westeros will create a 720p window, see comment: https://github.com/WebPlatformForEmbedded/meta-wpe/issues/172#issuecomment-354137423

However since meta-raspberrypi moved the default graphics output to 1080p I've created a rather blunt patch to set Westeros to 1080p for meta-wpe:
https://github.com/WebPlatformForEmbedded/meta-wpe/pull/325/commits/ff1e85d15ab06d0ec7861f2a0850c4d589f2cd67

Please review pull #325 and if OK that should resolve the smaller window issue.

wouterlucas, 2019-07-29:
Merged in, should be OK now.

closed 2019-07-29 by wouterlucas