WebPlatformForEmbedded/meta-wpe #237: WPEWebkit crash when playing any media

TomMettam, 2018-09-01:
Following the build instructions at https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/Raspberry-PI and using "bitbake wpe-eglfs-image"

WebGL works perfectly fine, HelloRacer performs well and was stable when left overnight.

However, when any media is played (video or audio), WPEWebkit exits. Manually attempting to play media with gstreamer results in:

`Setting pipeline to PAUSED ...
OMX-Cannot open OpenMAX registry file /home/root/.omxregister
OMX-A Component loader constructor fails. Exiting
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 7153' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
`

wouterlucas, 2018-09-01:
I think that needs to be fixed at: 
https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-multimedia/gstreamer/gstreamer1.0-omx_%25.bbappend

Pinging @kraj

wouterlucas, 2018-10-11:
Issue moved to meta-raspberry, pull request is open here: https://github.com/agherzan/meta-raspberrypi/pull/313

closed 2018-10-11 by wouterlucas