WebPlatformForEmbedded/meta-wpe #244: waveshare touch screen on raspberry pi NOT working

outbackdingo, 2018-10-12:
ok quick question im looking for a document of something that says i can use the waveshare touchscreen with WPE in kiosk mode ... i enabled the touch and cursor according to the document, however it didnt seem to work as expected... any help appreciated

takeitfurhter, 2018-11-01:
If you follow the general Waveshare instructions  you should be fine. Add the below to config.txt and make sure you have the ads7846 overlay in your image

dtparam=i2c_arm=on
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900
hdmi_drive=1
hdmi_force_hotplug=1

outbackdingo, 2018-11-02:
Ook i must be missing something because this doesnt help, any other pointers? 
and ive edited WebKitBrowser ffor touch and cursor to true...
heres my local.conf

MACHINE = "raspberrypi3"

GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
EXTRA_IMAGE_FEATURES = " debug-tweaks"
IMAGE_FEATURES_append = " ssh-server-dropbear"
IMAGE_INSTALL_append = " kernel-image kernel-devicetree python3-pip bluez5 i2c-tools python-smbus bridge-utils hostapd dhcp-server iptables wpa-supplicant mosquitto rrdtool"
KERNEL_DEVICETREE_append = "overlays/ads7846-overlay.dtb"

ENABLE_UART = "1"
ENABLE_DWC2 = "1"
ENABLE_I2C = "1"

ERROR_QA_remove = "license-checksum"
WARN_QA_append = " license-checksum"
LICENSE_FLAGS_WHITELIST = "commercial restricted"

DISTRO_FEATURES_remove = " x11"
DISTRO_FEATURES_append = " opengl wayland systemd python3-pip bluez5 bluetooth wifi"
# DISTRO_FEATURES_append = " opengl systemd python3-pip"
PREFERRED_PROVIDER_virtual/libgles2 = "userland"
PREFERRED_PROVIDER_virtual/egl = "userland"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""

WPE_BACKEND = "rpi libinput"

#to remove virtualinput from PACKAGECONFIG
PACKAGECONFIG_pn-wpebackend-rdk = "${WPE_BACKEND}"

WPE_PLATFORM = "westeros touch"
NETFLIX_BACKEND = "rpi"
WEBKITBROWSER_STARTURL ?= "http://localhost:8180/"
WPEFRAMEWORK_PLUGIN_WEBSERVER_PORT ?= "8180"

take-it-further, 2018-11-02:
I forgot to mention that you also need SPI enabled as that's what the touch sensor is connected to and something like

RPI_EXTRA_CONFIG = " max_usb_current=1 \
hdmi_group=2 \
hdmi_mode=87 \
hdmi_cvt 800 480 60 6 0 0 0 \
hdmi_drive=1 \
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=15 \
    "

in your local.conf

This is what we're using with the 5" model anyway

outbackdingo, 2018-11-02:
im confused, this touch screen is connected to USB hub.... spi enabled? and my config.txt looks like this
https://www.waveshare.com/wiki/10.1inch_HDMI_LCD_(H)_(with_case)

dtparam=i2c1=on
dtparam=i2c_arm=on
# Enable UART
enable_uart=1

avoid_warnings=1
dtoverlay=i2c-rtc,pcf8523
enable_uart=1
framebuffer_height=480
framebuffer_width=800
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
max_usb_current=1

dtoverlay=ads7846-overlay,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900

take-it-further, 2018-11-02:
Ah. I thought you were referring to the 5 inch version https://www.waveshare.com/wiki/5inch_HDMI_LCD

I've never used the 10 inch but it's clearly very different to the 5". You'd have to look at what USB touch controller it uses.

closed 2018-12-03 by wouterlucas