WebPlatformForEmbedded/meta-wpe #250: WPEFramework activation of few plugins are blocked

skanagaraj1, 2018-11-09:
Hi Everyone,

While starting the wpeframework.service using the systemctl start command I am seeing the following error message under journalctl log. Could you please help me to find out what could have cause this error?

[PluginServer.cpp:336]: Startup: Activated plugin [Controller]:[Controller]
[PluginServer.cpp:586]: Startup: Activation of plugin [Bluetooth]:[Bluetooth] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [Dictionary]:[Dictionary] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [WebKitBrowser]:[UX] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [WifiControl]:[WifiControl] blocked
[PluginServer.cpp:586]: Startup: Activation of plugin [WebKitBrowser]:[YouTube] blocked
[PluginHost.cpp:403]: Startup: WPEFramework actively listening.
[PluginServer.cpp:336]: Startup: Activated plugin [LocationSync]:[LocationSync]
[PluginServer.cpp:279]: Startup: Activation of plugin [TimeSync]:[TimeSync], postponed, preconditions have not b.
[PluginServer.cpp:336]: Startup: Activated plugin [TraceControl]:[Tracing]
[PluginServer.cpp:336]: Startup: Activated plugin [Snapshot]:[Snapshot]
[PluginServer.cpp:336]: Startup: Activated plugin [DeviceInfo]:[DeviceInfo]
[PluginServer.cpp:336]: Startup: Activated plugin [Monitor]:[Monitor]
[PluginServer.cpp:336]: Startup: Activated plugin [RemoteControl]:[RemoteControl]
WPEFramework[22158]: open failed
WPEFramework[22158]: 18:02:19:760 06 Assert 124!../../RefTarget_ZRC_MSO_GP712_
kill[22201]: kill: you need to specify whom to kill
systemd[1]: wpeframework.service: Control process exited, code=exited status=1

skanagaraj1, 2018-11-09:
The plugins are blocked because their conguration file has autostart as false. I am not sure why WPEFramework[22158]: open failed here

Monitor.json: "autostart":true,
LocationSync.json: "autostart":true,
YouTube.json: "autostart":false,
Bluetooth.json: "autostart":false,
WebKitBrowser.json: "autostart":true,
UX.json: "autostart":false,
WifiControl.json: "autostart":false,
TimeSync.json: "autostart":true,
Dictionary.json: "autostart":false,

wouterlucas, 2018-11-09:
@skanagaraj1 looks like the Remote Control has the RF4CE module enabled but its pointing to an .so it can't find.

There are a few ways from here:
* Disable meta-qorvo, GreenPeak or any other RF4CE recipes, that should make sure the remote control cmake doesn't find any GP packageconfig and that option won't be built. That should atleast bring up the stack just fine
* Point to the right GP module that you have on your system by setting https://github.com/WebPlatformForEmbedded/meta-wpe/blob/morty/recipes-wpe/wpeframework/wpeframework-plugins_git.bb#L36 and any other related RF4CE modules

skanagaraj1, 2018-11-10:
Thanks that worked. I am able to open the url of the device with my browser and go to webkitbrowser and set the url and get the current fps ,memory consumption and other information. 
Could you please tell me how can I open the browser via wpewebkitbrowser to stream videos

skanagaraj1, 2018-11-12:
WPE browser is not started during start up and wpe-launcher has been removed from the build as shown below. 
CORE_IMAGE_BASE_INSTALL_remove = "wpe-launcher wpelauncher"
Could you please tell me how to launch the browser to stream videos?

skanagaraj1, 2018-11-12:
I able to stream the videos now. Thanks for your help

wouterlucas, 2018-11-12:
You're welcome! :)

closed 2018-11-12 by wouterlucas