WebPlatformForEmbedded/meta-wpe #256: RDP or VNC support

mcz, 2018-12-05:
Hello,

Do you guys have any suggestion, or is there any way we can connect to the WPE browser and remote control it? What do you guys suggest?

Thanks,
Peter

wouterlucas, 2018-12-05:
Hi Peter,

There are various ways to control WPE remotely:
- keys can be sent remotely using the remote control api, see wpeframework ui for an example. Any key you press while the UI is in focus will already be automatically forwarded 
- URL can be remotely set and queried, including current FPS
- on RPI and Broadcom screen captures can be requested through the Snapshot plugin to see what’s going on

It’s not exactly VNC but remote control nonetheless. Additionally you can SSH into the device (or use webshell through the UI) and manage the device through a console.

mcz, 2018-12-06:
Thanks so much @wouterlucas . These sound a bit advanced :)
If we have SSH access to the device, how we can see whats going on on the screen? 
Can we use web inspector to see whats happening in the browser and maybe remote control it?

SJLC, 2018-12-19:
@mcz as far as I can tell, Weston RDP has no support whatsoever for client authentication
https://github.com/wayland-project/weston/blob/master/libweston/compositor-rdp.c
so in most scenarios it would be an unacceptable security risk

closed 2019-07-25 by wouterlucas