WebPlatformForEmbedded/meta-wpe #257: enable debug in wpeframework getting build issue

pavannandyala, 2018-12-05:
PACKAGECONFIG_append = "debug" in wpeframework getting build issue 

Error
******************************************************************************************************************************
  -mfloat-abi=hard -mcpu=cortex-a7  --sysroot=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0=/usr/src/debug/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0 -fdebug-prefix-map=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3=  -fvisibility-inlines-hidden  -march=armv7ve -marm -mfpu=neon-vfpv4  -mfloat-abi=hard -mcpu=cortex-a7  --sysroot=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed CMakeFiles/RPCProcess.dir/Process.cpp.o  -o WPEProcess-1.0.0000000 -rdynamic ../protocols/libWPEFrameworkProtocols.so.1.0.0000000 ../core/libWPEFrameworkCore.so.1.0.0000000 ../tracing/libWPEFrameworkTracing.so.1.0.0000000 ../cryptalgo/libWPEFrameworkCryptalgo.so.1.0.0000000 -Wl,-rpath,/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/protocols:/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/core:/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/tracing:/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/cryptalgo:
| CMakeFiles/RPCProcess.dir/Process.cpp.o: In function `WPEFramework::Core::IPCMessageType<2u, WPEFramework::RPC::Data::Input, WPEFramework::RPC::Data::Output>::IResponse()':
| /usr/src/debug/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/git/Source/core/Proxy.h:286: undefined reference to `non-virtual thunk to WPEFramework::Core::IPCMessageType<2u, WPEFramework::RPC::Data::Input, WPEFramework::RPC::Data::Output>::RawSerializedType<WPEFramework::RPC::Data::Output, 5u>::AddRef() const'
| CMakeFiles/RPCProcess.dir/Process.cpp.o: In function `WPEFramework::Core::IPCMessageType<2u, WPEFramework::RPC::Data::Input, WPEFramework::RPC::Data::Output>::IParameters()':
| /usr/src/debug/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/git/Source/core/Proxy.h:286: undefined reference to `non-virtual thunk to WPEFramework::Core::IPCMessageType<2u, WPEFramework::RPC::Data::Input, WPEFramework::RPC::Data::Output>::RawSerializedType<WPEFramework::RPC::Data::Input, 4u>::AddRef() const'
| collect2: error: ld returned 1 exit status
| make[2]: *** [Source/WPEProcess/WPEProcess-1.0.0000000] Error 1
| make[2]: Leaving directory `/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build'
| make[1]: *** [Source/WPEProcess/CMakeFiles/RPCProcess.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
| In file included from /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3/usr/include/c++/6.4.0/vector:63:0,
|                  from /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/git/Source/ocdm/../core/core.h:6,
|                  from /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/git/Source/ocdm/Module.h:8,
|                  from /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/git/Source/ocdm/open_cdm.cpp:18:
| /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3/usr/include/c++/6.4.0/bits/stl_uninitialized.h: In static member function 'static _ForwardIterator std::__uninitialized_copy<_TrivialValueTypes>::__uninit_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = std::move_iterator<WPEFramework::Core::ThreadPoolType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info, 1u, 4u>::ThreadUnitType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info>*>; _ForwardIterator = WPEFramework::Core::ThreadPoolType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info, 1u, 4u>::ThreadUnitType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info>*; bool _TrivialValueTypes = false]':
| /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3/usr/include/c++/6.4.0/bits/stl_uninitialized.h:68:9: note: parameter passing for argument of type 'std::move_iterator<WPEFramework::Core::ThreadPoolType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info, 1u, 4u>::ThreadUnitType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info>*>' will change in GCC 7.1
|          __uninit_copy(_InputIterator __first, _InputIterator __last,
|          ^~~~~~~~~~~~~
| /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3/usr/include/c++/6.4.0/bits/stl_uninitialized.h:68:9: note: parameter passing for argument of type 'std::move_iterator<WPEFramework::Core::ThreadPoolType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info, 1u, 4u>::ThreadUnitType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info>*>' will change in GCC 7.1
| /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3/usr/include/c++/6.4.0/bits/stl_uninitialized.h: In constructor 'WPEFramework::RPC::InvokeServerType<MESSAGESLOTS, THREADPOOLCOUNT>::InvokeServerType(uint32_t) [with unsigned int MESSAGESLOTS = 4u; short unsigned int THREADPOOLCOUNT = 1u]':
| /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3/usr/include/c++/6.4.0/bits/stl_uninitialized.h:126:41: note: parameter passing for argument of type 'std::move_iterator<WPEFramework::Core::ThreadPoolType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info, 1u, 4u>::ThreadUnitType<WPEFramework::RPC::InvokeServerType<4u, 1u>::Info>*>' will change in GCC 7.1
|   __uninit_copy(__first, __last, __result);
|                                          ^
| [ 88%] Linking CXX shared library libocdm.so
| cd /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/ocdm && /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_link_script CMakeFiles/ocdm.dir/link.txt --verbose=1
| /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++  -fPIC  -march=armv7ve -marm -mfpu=neon-vfpv4  -mfloat-abi=hard -mcpu=cortex-a7  --sysroot=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0=/usr/src/debug/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0 -fdebug-prefix-map=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3=  -fvisibility-inlines-hidden  -march=armv7ve -marm -mfpu=neon-vfpv4  -mfloat-abi=hard -mcpu=cortex-a7  --sysroot=/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/raspberrypi3 -g  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -shared -Wl,-soname,libocdm.so.1 -o libocdm.so.1.0.0000000 CMakeFiles/ocdm.dir/open_cdm.cpp.o CMakeFiles/ocdm.dir/ProxyStubs.cpp.o ../protocols/libWPEFrameworkProtocols.so.1.0.0000000 ../core/libWPEFrameworkCore.so.1.0.0000000 ../tracing/libWPEFrameworkTracing.so.1.0.0000000 ../cryptalgo/libWPEFrameworkCryptalgo.so.1.0.0000000 -Wl,-rpath,/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/protocols:/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7h [... cut ...]
| cd /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build/Source/ocdm && /mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_symlink_library libocdm.so.1.0.0000000 libocdm.so.1 libocdm.so
| make[2]: Leaving directory `/mnt/home/pnandyala/pawan-workspace/METRO/build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+96c97a98d4-r0/build'

psivasubramanian, 2018-12-12:
@pavannandyala can you please try this https://github.com/WebPlatformForEmbedded/meta-wpe/pull/261 if logging helps in your case.

pavannandyala, 2018-12-12:
@psivasubramanian this logging is only helping for wpewebkit, not for wpeframework

pavannandyala, 2018-12-13:
not able to observe build issue in recent images so closing this

wouterlucas, 2018-12-13:
As discussed in FD, likely related to cmake version in Morty. Does not happen on newer CMAKE version 3.7.2 and higher.

Likely best solution is to pull down the Pyro cmake (https://github.com/openembedded/openembedded-core/tree/pyro/meta/recipes-devtools/cmake) and stick it in meta-wpe morty branch for compatibility

wouterlucas, 2018-12-14:
I've tried the cmake backport from Pyro, but that did not seem to matter. Might be a toolchain related change on the RDKV side of things. Will need to look for other solutions on this.

wouterlucas, 2019-07-25:
This is solved, closing stale issue.

closed 2019-07-25 by wouterlucas