WebPlatformForEmbedded/meta-wpe #271: Cursor Does Not Move

fellowgeek, 2018-12-23:
I am running the most recent build on Rocko branch, when I enable the mouse cursor for a plugin, the cursor shows up but does not move.

Thanks

wouterlucas, 2018-12-26:
Rocko is quite outdated, can you try with master? I'd be willing to sync up Rocko but before I do would be nice to known if the issue is present on master.

fellowgeek, 2018-12-28:
I tried to build from master but seems something is broken and it won't start to build.

fellowgeek, 2019-01-15:
I build the master using the YOE build and I confirm the issue exists.

Thanks

wouterlucas, 2019-01-15:
Just to get an idea of your setup. Can you provide some more info:
* Are you running fullscreen EGL or Westeros image?
* You enabled the cursor boolean in WebKitBrowser.json?
* I'm assuming you're using an RPI3? 

I'll try the same on my end to cross validate.

malouf-erfan, 2019-01-15:
I tried both egg and Westeros image, and cursor is enabled in the WebKitBrowser.json file, cursor does show up but does not move.

I tested  on RPI3, RPI3B+, abd Pi-Zero-W

miikeat, 2019-04-14:
Is this fixed in the meantime?
sounds kind of related to #224 #238 and both claim to be fixed.

blakelloyd-coderus, 2020-04-23:
I think I'm hitting this same issue, the cursor appears, but I can't get it to move. What is strange is if I tap on the screen it does do a click on the screen. This is on warrior running fullscreen EGL with the cursor enabled on a RPi3

blakelloyd-coderus, 2020-05-28:
Anybody have any ideas for this? As the touchscreen works fine in the full sato build.