WebPlatformForEmbedded/meta-wpe #278: Build failure: EGL_DISPMANX_WINDOW_T undefined

shtig, 2019-01-21:
While trying to build following the instructions from "quick start" section of the wiki, running into the issue:
```
FAILED: CMakeFiles/WPEBackend-rdk.dir/src/bcm-rpi/renderer-backend.cpp.o 
/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot-native/usr/bin/aarch64-yoe-linux/aarch64-yoe-linux-g++  -DBACKEND_BCM_RPI=1 -DKEY
_INPUT_HANDLING_VIRTUAL=1 -DWPEBackend_rdk_EXPORTS -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot/usr/include/WPEFramework/vir
tualinput -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/git/src/input -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-r
dk/0.1-r0/git/src/util -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot/usr/include/gio-unix-2.0 -I/home/build/yoe-distro/build/
tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot/usr/include/glib-2.0 -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sy
sroot/usr/lib/glib-2.0/include -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot/usr/include/wpe-0.2 -I/home/build/yoe-distro/bui
ld/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot/usr/include/interface/vmcs_host/linux -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-
rdk/0.1-r0/recipe-sysroot/usr/include/interface/vcos/pthreads -I/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot/usr/include -fsta
ck-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/
0.1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0=/usr/src/d
ebug/wpebackend-rdk/0.1-r0 -fdebug-prefix-map=/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot= -fdebug-prefix-map=/home/build/yoe
-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wf
ormat-security -Werror=format-security  --sysroot=/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/recipe-sysroot -std=c++1y -fno-exceptions -fno
-strict-aliasing -fno-rtti -DNDEBUG -fPIC -MD -MT CMakeFiles/WPEBackend-rdk.dir/src/bcm-rpi/renderer-backend.cpp.o -MF CMakeFiles/WPEBackend-rdk.dir/src/bcm-rpi/renderer-back
end.cpp.o.d -o CMakeFiles/WPEBackend-rdk.dir/src/bcm-rpi/renderer-backend.cpp.o -c /home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/git/src/bcm-r
pi/renderer-backend.cpp
/home/build/yoe-distro/build/tmp/work/aarch64-yoe-linux/wpebackend-rdk/0.1-r0/git/src/bcm-rpi/renderer-backend.cpp:50:5: error: 'EGL_DISPMANX_WINDOW_T' does not name a type
     EGL_DISPMANX_WINDOW_T nativeWindow { 0, };
     ^~~~~~~~~~~~~~~~~~~~~
```
and then there is a bunch of other errors for missing definitions.

miikeat, 2019-04-14:
Out of curiosity I tried, probably same as you, a build for raspberrypi3-**64** but I would guess 64bit builds are not supported.

kraj, 2019-04-14:
Yes as of now the work for 64bit support is in progress especially for imx8 platform. Rpi3 is a tricky case since we have to use Mesa and vc4graphics in 64bit mode the h/w accelerated playback won’t yet work we are working on that issue too

wouterlucas, 2019-07-25:
You need to disable vc4graphics for the EGLFS backend to work. If you want to include vc4graphics than you need to go the Wayland/Westeros route.

closed 2019-07-25 by wouterlucas

marhoy, 2019-09-29:
@wouterlucas Sorry about the stupid question, but how do I "disable vc4graphics".

I'm trying to build wpewebkit for raspberrypi3 with the latest wpebackend-rdk and latest meta-raspberrypi, and I'm getting:

```
build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpebackend-rdk/gitAUTOINC-r0/git/src/bcm-rpi/renderer-backend.cpp:50:5: error: 'EGL_DISPMANX_WINDOW_T' does not name a type
|      EGL_DISPMANX_WINDOW_T nativeWindow { 0, };
```

psivasubramanian, 2019-09-30:
@marhoy just remove "vc4graphics" from MACHINE_FEATURES in meta-raspberrypi/conf/machine/include/rpi-base.inc.

marhoy, 2019-09-30:
@psivasubramanian Thanks!

Instead of fiddling with the meta-raspberrypi, I placed this in my local.conf
```
MACHINE_FEATURES_remove = "vc4graphics"
```
...and it worked! I'm now able to build the latest version.

Thanks :)

psivasubramanian, 2019-10-01:
@marhoy You are welcome!

limbouser, 2020-03-11:
How would the local.conf of a build that needs vc4_graphics look like? ie, the Westeros or Wayland route. I can't get the WPEWebKit to load when I tried some configurations.