WebPlatformForEmbedded/meta-wpe #281: Build failure: bluetooth/mgmt.h

TomMettam, 2019-02-17:
It seems recent changes have broken the build:

<pre>/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+fe68b1113c-r1/git/Bluetooth/Bluetooth.h:8:10: fatal error: bluetooth/mgmt.h: No such file or directory
|  #include &lt;bluetooth/mgmt.h&gt;
|           ^~~~~~~~~~~~~~~~~~
| compilation terminated.
</pre>

bramoosterhuis, 2019-02-19:
This is a known issue, you can either remove Bluetooth distro feature or switch off Bluetooth in the wpeframework-plugins_git.bb manually by deleting this line:https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-wpe/wpeframework/wpeframework-plugins_git.bb#L36

TomMettam, 2019-02-19:
Is there another issue ticket for the problem?

wouterlucas, 2019-02-20:
Temporarily disabled bluetooth on wpeframework-plugins for this. That header does not get installed in Yocto and we'll need to write a bbappend for that.

closed 2019-02-20 by wouterlucas