WebPlatformForEmbedded/meta-wpe #284: imx

mikeDonahoe, 2019-02-20:
Originally built everything from meta-webkit but I need support for Widevine OCDM.  I had 
same issue as https://github.com/WebPlatformForEmbedded/meta-wpe/issues/169
attempt to pull in X11 headers.  I augmented this patch https://github.com/WebPlatformForEmbedded/meta-wpe/blob/master/recipes-wpe/wpewebkit/files/0001-Define-MESA_EGL_NO_X11_HEADERS-when-not-using-GLX.patch to include <wayland-egl.h>.  It builds but I'm having issue starting WPEFramework.  Attempted fdo and rdk wayland backend and they have the same issue.

fsl-imx-wayland
IMAGE_INSTALL_append = " wpebackend-fdo"
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " opengl wayland opencdm widevine"
PACKAGECONFIG_append_pn-wpebackend-rdk = " wayland"
WPE_BACKEND="wayland"
WPE_PLATFORM="egl"

[       1842 us] WPEFramework
[       1977 us] Starting time: Wed, 20 Feb 2019 21:29:32 GMT
[       2013 us] SystemId:      OE2IYLkAAB
[       2033 us] Tree ref:      6a91c8e99d7fe16ef95255b5fb4e8aeff1654690
[       2049 us] Build ref:     6a91c8e99d7fe16ef95255b5fb4e8aeff1654690
[       2067 us] Version:       1.0.0000000
[       3395 us] Interface [lo], fully functional
[       3433 us] Forcing the network to IPv4 only.
[      17931 us] Accessor: http://192.168.1.175:80
[      17968 us] Interface IP: 0.0.0.0
[      38829 us] WPEFramework actively listening.
[      39191 us] EVENT: Platform
[      40761 us] EVENT: Network
[      40940 us] EVENT: Identifier: OE2IYLkAAB
[      41099 us] EVENT: Graphics
[      41181 us] EVENT: Provisioning
[      41206 us] EVENT: Streaming
[      41277 us] Activated plugin [Controller]:[Controller]
[      41588 us] Activation of plugin [Bluetooth]:[Bluetooth] blocked
[      41957 us] Activation of plugin [WebKitBrowser]:[UX] blocked
[      41999 us] Activation of plugin [WebKitBrowser]:[YouTube] blocked
[      43969 us] Activated plugin [DeviceInfo]:[DeviceInfo]
[      43995 us] Activated plugin [Dictionary]:[Dictionary]
[      47901 us] Activated plugin [Monitor]:[Monitor]
[      48250 us] Activated plugin [LocationSync]:[LocationSync]
[      55084 us] Activated plugin [RemoteControl]:[RemoteControl]
[      69217 us] Activation of plugin [TimeSync]:[TimeSync], postponed, preconditions have not been met, yet.
[      70795 us] Activated plugin [TraceControl]:[TraceControl]
[      88124 us] Activated plugin [TraceControl]:[Tracing]
[     172814 us] Activated plugin [OCDM]:[OCDM]
[     297843 us] Activation of plugin [WebKitBrowser]:[WebKitBrowser], postponed, preconditions have not been met, yet.
[     318954 us] Activation of plugin [WifiControl]:[WifiControl], failed. Error [class definitions does not exist]
[     401643 us] EVENT: WebSource
[     405721 us] Activated plugin [WebServer]:[WebServer]
[     408675 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
WPEView: failed to create backend
~ repeated over and over

mikeDonahoe, 2019-02-21:
=======================================================
    glmark2 2017.07
=======================================================
    OpenGL Information
    GL_VENDOR:     Vivante Corporation
    GL_RENDERER:   Vivante GC2000
    GL_VERSION:    OpenGL ES 3.0 V6.2.4.p2.163672
=======================================================
[build] use-vbo=false: FPS: 403 FrameTime: 2.481 ms
[build] use-vbo=true: FPS: 650 FrameTime: 1.538 ms
[texture] texture-filter=nearest: FPS: 366 FrameTime: 2.732 ms
[texture] texture-filter=linear: FPS: 350 FrameTime: 2.857 ms
[texture] texture-filter=mipmap: FPS: 339 FrameTime: 2.950 ms
[shading] shading=gouraud: FPS: 405 FrameTime: 2.469 ms
[shading] shading=blinn-phong-inf: FPS: 400 FrameTime: 2.500 ms
[shading] shading=phong: FPS: 296 FrameTime: 3.378 ms
[shading] shading=cel: FPS: 245 FrameTime: 4.082 ms
[bump] bump-render=high-poly: FPS: 289 FrameTime: 3.460 ms
[bump] bump-render=normals: FPS: 552 FrameTime: 1.812 ms
[bump] bump-render=height: FPS: 441 FrameTime: 2.268 ms
libpng warning: iCCP: known incorrect sRGB profile
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 134 FrameTime: 7.463 ms
libpng warning: iCCP: known incorrect sRGB profile
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 60 FrameTime: 16.667 ms
[pulsar] light=false:quads=5:texture=false: FPS: 305 FrameTime: 3.279 ms
libpng warning: iCCP: known incorrect sRGB profile
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 45 FrameTime: 22.222 ms
libpng warning: iCCP: known incorrect sRGB profile
[desktop] effect=shadow:windows=4: FPS: 127 FrameTime: 7.874 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 64 FrameTime: 15.625 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 64 FrameTime: 15.625 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 80 FrameTime: 12.500 ms
[ideas] speed=duration: FPS: 137 FrameTime: 7.299 ms
[jellyfish] <default>: FPS: 128 FrameTime: 7.812 ms
[terrain] <default>: FPS: 0 FrameTime: inf ms
[shadow] <default>: FPS: 138 FrameTime: 7.246 ms
[refract] <default>: FPS: 23 FrameTime: 43.478 ms
[conditionals] fragment-steps=0:vertex-steps=0:^C FPS: 397 FrameTime: 2.519 ms
=======================================================
                                  glmark2 Score: 247 
=======================================================

mikeDonahoe, 2019-02-21:
<pre>
-     L1 [5934]: "Spawning a new process: 5934."
-     L1 [5934]: "Spawning a new plugin WebKitImplementation."
-     L1 [5934]: "Created an internal factory communication channel, on /tmp/communicator as Client"
-     L1 [5934]: "Loading ProxyStubs from /usr/lib/wpeframework/proxystubs/"
-     L1 [5934]: "Interface Aquired. 0x1bbd1d0."
-     L1 [5525]: " A remote socket hooked up to an external factory communication channels as Client"
-     L1 [5934]: "Invoking the Announce message to the server. 178"
-     L1 [5934]: "Todo Announce message handled. 182"
-     L1 [5525]: "Remote Process 5934, has announced itself."
-     L1 [5934]: "Process up and running: 5934."
-     L1 [5934]: "No handler defined to handle the incoming frames. [3]"
-     L1 [5525]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5934]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5525]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5934]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5934]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5934]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5934]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5525]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5934]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5525]: "No handler defined to handle the incoming frames. [5]"
-     L1 [5525]: "No handler defined to handle the incoming frames. [5]"
[    8256446 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
WPEView: failed to create backend
WPENetworkProcess: /usr/src/debug/wpeframework/3.0+gitAUTOINC+6a91c8e99d-r0/git/Source/core/Singleton.cpp:21: WPEFramework::Core::Singleton::SingletonList::~SingletonList(): Assertion `m_Singletons.empty() == true' failed.
</pre>

wouterlucas, 2019-02-21:
it seems that WPEWebKit can't initialize the view/backend. Does this work with the meta-webkit launchers like Cog or similar? Because if that doesn't work its probably easier to debug the graphics part first without all the extras WPEFramework brings.

If it does work with Cog and not with WPEFramework, likely the backend isnt properly setup. I've never gotten around to try FDO in combination with WPEFramework. Likely the recipes will need further adjustment for that.

mikeDonahoe, 2019-02-21:
New to WPE so please forgive me.
In order to build cog I see it is looking for 'wpe-webkit-0.1' which I don't have.

pkg-config --libs wpe-webkit
-lWPEWebKit -lwpe-0.2

Do I need to not build from wpewebkit_20170728.bb?

When I built from meta-webkit cog worked fine.

mikeDonahoe, 2019-02-22:
I was able to run cog and that worked with wpebackend-fdo.  But then found out that wpebackend-fdo does not have _wpe_view_backend_interface so libwpe returns NULL for wpe_view_backend_create() .  Switched back to wpebackend-rdk and I'm now able to render pages.  The original issue I had with rdk was having PACKAGECONFIG_append_pn-wpebackend-rdk = " wayland" instead of PACKAGECONFIG_append_pn-wpebackend-rdk = " wayland-egl"

closed 2019-02-22 by mikeDonahoe