WebPlatformForEmbedded/meta-wpe #290: Latest You Build From Master Does not Load Websites

fellowgeek, 2019-03-21:
Hello,

I compiled the latest YOE build from master for Raspberry Pi3, it does display the built in website but fails to load anything from network, I tried https://www.google.com for example.

I see the WPEFramework and network processes but no graphical output only terminal.

I verified, network works fine and I can ping sites over internet.

kraj, 2019-03-24:
Can you check if you have ca-certificates in image if not then it could be the problem.

malouf-erfan, 2019-03-25:
Hi @kraj I added the  ca-certificates package to the image and still HTTPS not working, I compiled from master about 1 month ago and it was working, I wonder if something related to https has changed

wouterlucas, 2019-07-25:
Works fine with latest master, can you retry?

closed 2019-07-29 by wouterlucas