WebPlatformForEmbedded/meta-wpe #291: Widevine Netflix

mikeDonahoe, 2019-03-25:
I was wondering if anyone has had any luck getting Netflix to work.  Currently, have Amazon Prime and other Widevine protected streams working on the iMX6.  However, Netflix always displays an error.  Is it possible to playback Netflix content via WPEFramework or is the closed plugin mentioned in https://github.com/WebPlatformForEmbedded/meta-wpe/issues/164 required?

mikeDonahoe, 2019-03-25:
Error that is displayed.

[image]

petegriffin, 2019-04-02:
Have you tried updating the user agent that WPE is sending to report a newer version of Chrome (e.g. v73)?

mikeDonahoe, 2019-04-03:
Yes, I did.  I actually took the user agent string from my Laptop, which is able to play Netflix.  I'm guessing Netflix is checking for something in addition to the user agent.

closed 2019-07-02 by mikeDonahoe

cchsu23, 2020-07-01:
@mikeDonahoe  Hi I has the same problem with you, do you have solution?