WebPlatformForEmbedded/meta-wpe #298: rk3399 recipes

StuartIanNaylor, 2019-05-22:
Just a request
I have tried the raspberry pi recipe and the fixed no scalable window thing is no that great.

I noticed collabora have been doing much with mali and wondered if more "rk3399" specific recipes could be included.
RockPi & RockPro64 ?

kraj, 2019-05-22:
sure if there are specific needs for WPE then this is the place, otherwise BSP layers are appropriate place.

StuartIanNaylor, 2019-05-23:
It is really BSP but just thought I would ask if there is a Rockpi BSP & Panfrost support?

wouterlucas, 2019-07-25:
There is no direct support for RockPi, best is to see if you can get Wayland/Westeros to work or see if you can go the wpebackend-fdo route over at meta-webkit :)

closed 2019-07-25 by wouterlucas