WebPlatformForEmbedded/meta-wpe #300: RP3 vanilla build

shacal, 2019-06-11:
Unable to get vanilla WPE to work with current settings

conf/local.conf
```
DISTRO_FEATURES_remove_rpi = "x11"
DISTRO_FEATURES_remove_rpi = "wayland"
DISTRO_FEATURES_append = " opengl"

WPE_BACKEND_rpi = "rpi libinput"
WPE_PLATFORM = "egl"
NETFLIX_BACKEND = "rpi"
```

Error
```
root@raspberrypi3:~# /usr/bin/WPEFramework
[       1664 us] WPEFramework
[       1855 us] Starting time: Tue, 11 Jun 2019 23:07:23 GMT
[       1935 us] SystemId:      OEuCfrQLya
[       2005 us] Tree ref:      6f3b75efe9af35e84675a0c2cb420173abb8d584
[       2074 us] Build ref:     6f3b75efe9af35e84675a0c2cb420173abb8d584
[       2144 us] Version:       1.0.
[       4059 us] Interface [lo], fully functional
[       4220 us] Forcing the network to IPv4 only.
[       7690 us] Accessor: http://10.0.2.79:80
[       7940 us] Interface IP: 0.0.0.0
[      16854 us] WPEFramework actively listening.
[      17492 us] EVENT: Platform
[      17817 us] EVENT: Security
[      17964 us] EVENT: Network
[      18053 us] EVENT: Identifier: OEuCfrQLya
[      18827 us] EVENT: Graphics
[      18971 us] EVENT: Internet [127.0.0.1]
[      19059 us] EVENT: TimeZone: , Country: , Region: , City: 
[      19183 us] EVENT: Time: Tue, 11 Jun 2019 23:07:23 GMT
[      19261 us] EVENT: Provisioning
[      19430 us] EVENT: Decryption
[      19602 us] EVENT: WebSource
[      19774 us] EVENT: Streaming
[      20239 us] Activated plugin [Controller]:[Controller]
[      20459 us] Activation of plugin [WebKitBrowser]:[Apps] blocked
[      20839 us] Activation of plugin [WebKitBrowser]:[UX] blocked
[      20945 us] Activation of plugin [WebKitBrowser]:[YouTube] blocked
[      25430 us] Activated plugin [LocationSync]:[LocationSync]
[      27954 us] Activated plugin [Dictionary]:[Dictionary]
[      33639 us] Activated plugin [DeviceInfo]:[DeviceInfo]
[      33945 us] Activated plugin [Monitor]:[Monitor]
[      34120 us] Interface [wlan0] activated, no IP associated
[      34253 us] Interface [eth0] activated, DHCP request issued
[      35195 us] DNS functionality updated [/etc/resolv.conf]
[      35582 us] Activated plugin [NetworkControl]:[NetworkControl]
[      48371 us] Activated plugin [Snapshot]:[Snapshot]
[      49372 us] Activated plugin [TimeSync]:[TimeSync]
[      50869 us] Activated plugin [RemoteControl]:[RemoteControl]
[     204010 us] Activated plugin [TraceControl]:[TraceControl]
[     219144 us] Activated plugin [WifiControl]:[WifiControl]
[     268858 us] WebSource is not defined as External !!
[     268943 us] Activated plugin [WebServer]:[WebServer]
[     319580 us] EVENT: Internet [193.184.32.182]
[     320180 us] EVENT: TimeZone: EET-2EEST,M3.5.0/3,M10.5.0/4, Country: FI, Region: , City: 
[     423763 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
[LibinputServer] Initialization of virtual keyboard and linux input system succeeded.
Couldn't open "libEGL.so.1": "libEGL.so.1": cannot open shared object file: No such file or directory
[     437134 us] CRASH: WebProcess crashed, exiting...
```

Something needs to be updated somewhere? :D

psivasubramanian, 2019-06-19:
If you give the full path of graphics libraries as https://github.com/WebPlatformForEmbedded/meta-wpe/blob/96459565f847fd7e4ddc53b548344144fca7a5d0/recipes-graphics/libepoxy/libepoxy_%25.bbappend#L5-L9, it will resolve.

shacal, 2019-06-27:
Waiting this: #295

shacal, 2019-06-27:
And /usr/lib/ prefix did not resolve my vanilla tutorial build.
Crashes WepProcess...
Forced shutdown WebKitBrowser by rason: Failure

petegriffin, 2019-06-28:
@shacal You could try removing the libepoxy.bbappend entirely from meta-wpe layer. This resolved the issue for me (see strace output here https://github.com/WebPlatformForEmbedded/meta-wpe/issues/301)

wouterlucas, 2019-06-28:
Is this still relevant after https://github.com/WebPlatformForEmbedded/meta-wpe/pull/295 ?

closed 2019-07-20 by wouterlucas