WebPlatformForEmbedded/meta-wpe #301: libepoxy_%.bbappend causing libEGL.so library loading problems

petegriffin, 2019-06-12:
The libepoxy bbappend in meta-wpe causes libEGL.so to not load on my iMX8M platform. Writing a small test program to use libepoxy library and then running strace shows

openat(AT_FDCWD, "/lib/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0@\20\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=14296, ...}) = 0
mmap(NULL, 77920, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xffffa45ca000
mprotect(0xffffa45cd000, 61440, PROT_NONE) = 0
mmap(0xffffa45dc000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0xffffa45dc000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xffffa48ad000
mprotect(0xffffa473e000, 12288, PROT_READ) = 0
mprotect(0xffffa45dc000, 4096, PROT_READ) = 0
mprotect(0xffffa487c000, 28672, PROT_READ) = 0
mprotect(0x410000, 4096, PROT_READ)     = 0
mprotect(0xffffa48b9000, 4096, PROT_READ) = 0
munmap(0xffffa48b1000, 23537)           = 0
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
brk(NULL)                               = 0x37daf000
brk(0x37dd0000)                         = 0x37dd0000
write(1, "Hello libepoxy world\n", 21Hello libepoxy world
)  = 21
openat(AT_FDCWD, "\"libEGL.so.1\"", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "Couldn't open \"libEGL.s"..., 120Couldn't open "libEGL.so.1": "libEGL.so.1": cannot open shared object file: No such file or directory
) = 120
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [], 8) = 0
getpid()                                = 29798
gettid()                                = 29798
tgkill(29798, 29798, SIGABRT)           = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=29798, si_uid=1000} ---
+++ killed by SIGABRT +++
Aborted

By removing this bbappend libepoxy, my simple libepoxy test application and WPEFramework all start OK. So I'm wondering how this patch works for you guys? 

In case it is useful my little test libepoxy app is here https://github.com/petegriffin/egltest

TomMettam, 2019-06-16:
I received this issue with a fresh build with WPEWebkit.

After removing the bbappend above, i now get:

`Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.
Cannot create EGL WPE context: EGL_SUCCESS`

TomMettam, 2019-06-16:
(And with your test app @petegriffin I get: `Hello libepoxy world
Couldn't open libEGL.so.1: /usr/lib/libEGL.so.1: undefined symbol: glPointSizePointerOES
Aborted
`

TomMettam, 2019-06-16:
Added -lGLESv2 and all is good now though.

`./libepoxyhello
Hello libepoxy world
EGL_TRUE: everything all good`

shacal, 2019-06-17:
Maybe same issue that latest build has on my side
#300

kraj, 2019-06-17:
that bbappend is doing something thats SOC specific, I think broadcom layer would be right place to make this change so probably it should be moved out from meta-wpe, however this also means if other machines would need to address loading the EGL libs for libepoxy, since these libraries may have dependencies on some private libraries or public ones as seen in comment from @TomMettam above for rpi.

TomMettam, 2019-06-22:
@kraj Any idea how i can resolve my issue? It's preventing me from using this at the moment. I guess the wrong libEGL is being referenced?

kraj, 2019-06-22:
I think there is another pull request pending for enabling vc4graphics which has fix for it

psivasubramanian, 2019-06-24:
@TomMettam Could you please try this fix - https://github.com/WebPlatformForEmbedded/meta-wpe/blob/3e32f424ddb059aff0abd8dd912e2946f7351aa7/recipes-graphics/libepoxy/libepoxy_%25.bbappend (Line 5-8) which @kraj mentioned.

petegriffin, 2019-06-28:
@TomMettam apologies I've only just seen all the comments on this issue :-( The way I work around this issue is to remove the bbappend entirely. If you have the simple egltest working with libepoxy, then you shouldn't have any issues with wpewebkit.

petegriffin, 2019-07-02:
Closing as #307 fixes this issue

closed 2019-07-02 by petegriffin