WebPlatformForEmbedded/meta-wpe #306: WPEFramework-plugins build issues when ${WPE_COMPOSITOR} is enabled

Khan3033, 2019-07-01:
GCC_VERSION = 9.1.0
branch = master
${WPE_COMPOSOTOR} is enabled to resolve below issues:
1) Browser window is rendered in small resolution
2) Key events are not working
Enabled WPEFramework backend in wpebackend-rdk and Compositor plugin in WPEFramework Plugins. Please find below errors which we have got after enabling ${WPE_COMPOSITOR} 

/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h:440:18: error: expected unqualified-id before numeric constant
  440 |             bool Success = true;
      |                  ^~~~~~~
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h: In static member function 'static uint32_t WPEFramework::Core::NumberType<TYPE, SIGNED, BASETYPE>::Convert(const wchar_t*, uint32_t, NUMBER&, NumberBase, const TemplateIntToType<0>&)':
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h:548:18: error: expected unqualified-id before numeric constant
  548 |             bool Success = true;
      |                  ^~~~~~~
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h: In static member function 'static uint32_t WPEFramework::Core::NumberType<TYPE, SIGNED, BASETYPE>::Convert(const char*, uint32_t, NUMBER&, NumberBase, const TemplateIntToType<1>&)':
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h:620:18: error: expected unqualified-id before numeric constant
  620 |             bool Success = true;
      |                  ^~~~~~~
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h: In static member function 'static uint32_t WPEFramework::Core::NumberType<TYPE, SIGNED, BASETYPE>::Convert(const char*, uint32_t, NUMBER&, NumberBase, const TemplateIntToType<0>&)':
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/Number.h:728:18: error: expected unqualified-id before numeric constant
  728 |             bool Success = true;
      |                  ^~~~~~~
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/JSON.h: At global scope:
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/JSON.h:801:38: error: expected unqualified-id before numeric constant
  801 |             static constexpr uint8_t None = 0x00;
      |                                      ^~~~
/home/balajiv/riyaz-workspace/yoe_19_JUNE/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework-plugins/3.0+gitAUTOINC+7ece0856af-r1/recipe-sysroot/usr/include/WPEFramework/core/JSON.h:920:39: error: expected unqualified-id before numeric constant
  920 |             static constexpr uint32_t None = 0x00000000;

psivasubramanian, 2019-07-01:
Hi @pwielders 
Any idea here, something to do with gcc version?

wouterlucas, 2019-07-26:
@Khan3033 @psivasubramanian can you check if this is still an issue with latest master?
For RPI3 and latest OE-core I do not see the same issue.

psivasubramanian, 2019-07-26:
Hi @wouterlucas, no it's resolved with https://github.com/WebPlatformForEmbedded/ThunderNanoServices/commit/ec4e3c3f5295778409265c2f6e2b2b5a3ea8ad00. Maybe @Khan3033 can close this?

wouterlucas, 2019-07-26:
I'll close it, thanks for letting me know @psivasubramanian

closed 2019-07-26 by wouterlucas