WebPlatformForEmbedded/meta-wpe #315: Writing an X11 backend for WPE

peoro, 2019-07-09:
For starters, I'm not sure if this is the right place for my question. I apologize if it's not.

I'd like to start playing around with WPE and would love to get it running on my desktop computer. I'm using X11 rather than Wayland (Nvidia :( ), and from what I understood there's no X11 launcher/backend available.

I'd be happy to try and write an X11 backend: it might be useful (at least to me) and it seems like a good way to familiarize with WPE.

Before I waste my time I'd like to know if what I'm planning to do makes sense, and whether I understood correctly the WPE architecture:

 - [WPEWebkit](https://github.com/WebPlatformForEmbedded/WPEWebKit) is the port of WebKit meant to be used with the WPE infrastructure. The main difference with the other ports of WebKit is that it renders webpages using EGL.
- [libwpe](https://github.com/WebPlatformForEmbedded/libwpe) offers the interface (e.g. the header files) to use WPEWebKit as a library.
- The various backends (FDO, RDK, Mesa) create the EGL context necessary for WPEWebkit and run it.
- The launchers (cog, dyz, dinghy?) create a surface the backend can draw on and handle input, sending it to the backend.

If I understood everything correctly, then I'm wondering why the separation between backend and launcher is so important.  

The simplest way to develop an X11 client seems to be writing a single binary that works both as a launcher and as a backend. It could use Xlib/XCB (or even some toolkits, like Qt) to create an X11 window, an EGL context for it and process input. It would use the various functions/structs exposed in libwpe to istantiate WPEWebKit and pass it an EGL context and input events.

Am I on the right track?

kraj, 2019-07-09:
I think you are on right track, however, wpe repo might be better to discuss this proposal, since meta-wpe is a yocto layer primarily deals with build side of things.

peoro, 2019-07-09:
Oops, sorry!

Would you be able to transfer this issue, or should I close it and open a new one? And by `wpe repo` you mean `libwpe`?

kraj, 2019-07-09:
open a new one at maybe https://github.com/WebPlatformForEmbedded/WPEWebKit

closed 2019-07-20 by wouterlucas