WebPlatformForEmbedded/meta-wpe #320: WebKitBrowser does not start

TomMettam, 2019-07-19:
Using a vanilla build with master, WebKitBrowser refuses to start.

[     429046 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
[     721496 us] Deactivated plugin [WebKitBrowser]:[WebKitBrowser]
[     931315 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
[    1195794 us] Deactivated plugin [WebKitBrowser]:[WebKitBrowser]
[    1405704 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
[    1706825 us] Deactivated plugin [WebKitBrowser]:[WebKitBrowser]
[    1914587 us] Activated plugin [WebKitBrowser]:[WebKitBrowser]
[    2187098 us] Deactivated plugin [WebKitBrowser]:[WebKitBrowser]

There's no other output in syslog.

wouterlucas, 2019-07-20:
Hey Tom, 

It seems the wpebackend-rdk is misconfigured through the recipe for the EGLFS case.
Set `WPE_BACKEND` to `rpi` and should will start.

I've fixed the issue in the recipe but it comes a long a whole update of srvrecvs and more, so will take me some time to push out.

TomMettam, 2019-07-20:
@wouterlucas  Thanks for your response. Unfortunately, no dice. This is the entirety of my local.conf (I tried to cut everything back to just what information appears on the front page of this repo):

```
DISTRO_FEATURES_remove_rpi = "x11"
DISTRO_FEATURES_remove_rpi = "wayland"

WPE_BACKEND = "rpi"
```

Is anything else required? `WPE_PLATFORM` maybe?

wouterlucas, 2019-07-20:
No that should be alright. Platform might be needed to enable EGL for video playback. But I saw wpebackend-rdk did not have a backend set by default for RPI EGLFS. 

I’ll check it as part of all the updates. Will probably take a day or more to get that all sorted.

TomMettam, 2019-07-22:
Thanks @wouterlucas, I appreciate your help! I've tried a raw image with meta-webkit and cog which also encounters egl issues. I'm desperately trying to get something recent running, but failing.

wouterlucas, 2019-07-22:
Try adding https://github.com/WebPlatformForEmbedded/meta-wpe/pull/321

With that at least I got graphics again, will run a few more tests and see what else is needed.

wouterlucas, 2019-07-25:
Merged in, should be fine now. Also try  https://github.com/WebPlatformForEmbedded/meta-wpe/pull/323 for the latest Gstreamer 1.16 bumps.

With that 1080p graphics + video works fine for me on RPI3 EGLFS.

closed 2019-07-29 by wouterlucas